lancaster-university/codal-microbit-v2 #272: I2C redirect twice freezes the micro:bit

microbit-carlos, 2023-02-13:
With this example CODAL programme with the latest code in main, the second `uBit.i2c.redirect()` freezes the micro:bit (the "hi" message never scrolls):

```cpp
#include "MicroBit.h"

MicroBit uBit;

int main() {
    uBit.init();

    uBit.i2c.redirect(uBit.io.P0, uBit.io.P1);
    uBit.i2c.redirect(uBit.io.P1, uBit.io.P0);

    while (true) {
        uBit.display.scroll("hi");
        uBit.sleep(1000);
    }
}
```

Most interestingly, if BLE is enabled (just using the codal.ble.json file instead of codal.json) it throws a 020 out-of-memory error.

finneyj, 2023-02-13:
Thanks @microbit-carlos. Strange - I definitely tested that one! must be a regression... I'll look at it.

finneyj, 2023-03-07:
Thanks @microbit-carlos 

I've now investigated this. The title of this issue is a bit misleading. In general case, the redirect() operation did operate correctly multiple times. However, there is an edge case where the redirect() code was getting itself into an infinitely deep recursive loop in conditions where a pin was being redirected to a peripheral to which it was already attached.

i.e. the following code would be fine, but the example above would not:

```
uBit.i2c.redirect(uBit.io.P0, uBit.io.P1);
uBit.i2c.redirect(uBit.io.P2 uBit.io.P3);  
uBit.i2c.redirect(uBit.io.P0, uBit.io.P1);
uBit.i2c.redirect(uBit.io.P2 uBit.io.P3);  
```

I've tracked this down to the NRF52Pin::disconnect() function.

It's now patched in master to address this issue:
https://github.com/lancaster-university/codal-core/commit/ca19454c022d5d27ee6810900b91c50e0dcbaf28
https://github.com/lancaster-university/codal-nrf52/commit/5fa1d16d6de6bd2848ae776e3039ce6bcad4e3e1

microbit-carlos, 2023-05-17:
Damien has tested this and confirmed it's fixed 🎉
- https://github.com/microbit-foundation/micropython-microbit-v2/issues/112#issuecomment-1548794240

closed 2023-05-17 by microbit-carlos