lancaster-university/codal-microbit-v2 #280: Microbit v2 + FreeRTOS

crisgplj, 2023-02-24:
Hello, I was interested in using the microbit v2 but I have a problem.

I manage to compile in eclipse https://github.com/lancaster-university/microbit-v2-samples

but when trying to merge FreeRTOS https://www.freertos.org/

I get nothing.

Any ideas?

Thank you!!!

microbit-carlos, 2023-02-27:
Hi @crisgplj,

Did you have a specific use for adding FreeRTOS? The micro:bit CODAL HAL uses fibers to achieve cooperative concurrency already without the need to add an RTOS.

You can find more info about fibers and concurrency in these docs:
- https://lancaster-university.github.io/microbit-docs/concepts/
- https://lancaster-university.github.io/microbit-docs/advanced/
- https://makecode.microbit.org/device/reactive

crisgplj, 2023-03-02:
ello! I try to port several old projects on the microbit so I use FreeRtos.

I wanted to tell you something else, what flags or processing symbols would I need to compile with g++?

With cmake everything works but I need the project to be portable

microbit-carlos, 2023-03-03:
CMake has a generator for Eclipse, so you can use that to export the project to eclipse:
https://cmake.org/cmake/help/latest/manual/cmake-generators.7.html
https://cmake.org/cmake/help/latest/generator/Eclipse%20CDT4.html

microbit-carlos, 2023-11-03:
I'll close this as resolved, but feel free to reopen if you have any further questions, thanks!

closed 2023-11-03 by microbit-carlos