lancaster-university/codal-microbit-v2 #283: Sound level is zero when loud and quiet event handlers are registered

martinwork, 2023-03-02:
In test deployment https://makecode.microbit.org/app/d4edbad2afea7659ac6cb7bc231bf71183ee861a-980f0ccdaf with CODAL 0.2.49, including loud and quiet handlers seems to cause sound level readings to be stuck a zero. After deleting the handlers, non-zero levels are back.

MakeCode example: https://makecode.microbit.org/_9m42M89hJMXb
In MakeCode v5, this example works as expected, flashing LEDs on the left for loud & quiet events, and flickering the LED on the right for any level above 0.

C++ test to follow

martinwork, 2023-03-02:
The C++ equivalent seems to work fine.

The MakeCode example seems to work in beta, with CODAL 0.2.40. The levels are higher in CODAL 0.2.40, so the level LED (4,0) is flickering most of the time. Strangely, the LED seems to be on more after a RESET than immediately after WebUSB flashing.

```
#include "MicroBit.h"

MicroBit uBit;

////////////////////////////////////////////////////////////////

#define MICROPHONE_MIN 52.0f
#define MICROPHONE_MAX 120.0f

codal::LevelDetectorSPL *getMicrophoneLevel() {
    return uBit.audio.levelSPL;
}

int soundLevel() {
    int micValue = getMicrophoneLevel()->getValue();
    int scaled = max(MICROPHONE_MIN, min(micValue, MICROPHONE_MAX)) - MICROPHONE_MIN;
    scaled = min(0xff, scaled * 0xff / (MICROPHONE_MAX - MICROPHONE_MIN));
    return scaled;
}

////////////////////////////////////////////////////////////////

void flicker( int x, int y)
{
    uBit.display.image.setPixelValue( x, y, uBit.display.image.getPixelValue(x, y) ? 0 : 255);
    uBit.sleep(100);
    uBit.display.image.setPixelValue( x, y, uBit.display.image.getPixelValue(x, y) ? 0 : 255);
}

////////////////////////////////////////////////////////////////

void forever()
{
    while (true)
    {
        if ( soundLevel() > 0)
        {
            flicker( 4, 0);
        }
        uBit.sleep(20);
    }
}

void onLoud( MicroBitEvent)
{
    flicker( 0, 0);
}

void onQuiet(MicroBitEvent)
{
    flicker( 0, 4);
}

int main()
{
    // Initialise the micro:bit runtime.
    uBit.init();
    
    uBit.messageBus.listen( DEVICE_ID_MICROPHONE, LEVEL_THRESHOLD_HIGH, onLoud);
    uBit.messageBus.listen( DEVICE_ID_MICROPHONE, LEVEL_THRESHOLD_LOW, onQuiet);

    create_fiber( forever);

    release_fiber();
}

{
    "target": {
        "name": "codal-microbit-v2",
        "url": "https://github.com/lancaster-university/codal-microbit-v2",
        "branch": "master",
        "type": "git",
        "test_ignore": true,
        "dev": true
    },
    "config":{
        "DEVICE_DMESG": 0,
        "DMESG_SERIAL_DEBUG": 0,
        "CODAL_DEBUG": 0,
        "MICROBIT_BLE_ENABLED" : 0,
        "MICROBIT_BLE_PAIRING_MODE": 1,
        "CONFIG_MICROBIT_ERASE_USER_DATA_ON_REFLASH": 1
    }
}

```

martinwork, 2023-03-16:
@JohnVidler I think this C++ sample seemed to work for me because I used a microbit_v2_samples build with DMESG enabled. Without DMESG it only works after a reset. The MakeCode sample doesn't work after reset. I think using uBit.serial rather than DMESG doesn't make the C++ work.

martinwork, 2023-03-16:
@JohnVidler It seems LowPassFilter::applyEffect is generating zeros

martinwork, 2023-03-16:
@JohnVidler Using the debug code below, all the float calculations seem to all be zero immediately after download. The numbers are the same, but the calculations fail.

```
Add to CodalDmesg.h (just as a convenient place):

#include "Serial.h"
extern codal::Serial *target_serial;

Add to CodalDmesg.cpp:

codal::Serial *target_serial = NULL;

void LowPassFilter::applyEffect(ManagedBuffer inputBuffer, ManagedBuffer outputBuffer, int format)
{
    int bytesPerSample = DATASTREAM_FORMAT_BYTES_PER_SAMPLE(format);
    int sampleCount = inputBuffer.length() / bytesPerSample;
    uint8_t *in = inputBuffer.getBytes();
    uint8_t *out = outputBuffer.getBytes();

    for( int i=0; i<sampleCount; i++)
    {
        float value = StreamNormalizer::readSample[format]( in );
        if ( target_serial)
          target_serial->send( "value " + ManagedString( (int) (value * 1000)) + "\n");
        if ( target_serial)
          target_serial->send( "lpf_value " + ManagedString( (int) (lpf_value * 1000)) + "\n");
        if ( target_serial)
          target_serial->send( "lpf_beta " + ManagedString( (int) (lpf_beta * 1000)) + "\n");

        float c1 = lpf_value - (float)value;
        float c2 = lpf_beta * c1;
        float c3 = lpf_value - c2;

        if ( target_serial)
          target_serial->send( "c1 " + ManagedString( (int) (c1 * 1000)) + "\n");
        if ( target_serial)
          target_serial->send( "c2 " + ManagedString( (int) (c2 * 1000)) + "\n");
        if ( target_serial)
          target_serial->send( "c3 " + ManagedString( (int) (c3 * 1000)) + "\n");

        lpf_value = lpf_value - (lpf_beta * (lpf_value - (float)value));
        if ( target_serial)
          target_serial->send( "new " + ManagedString( (int) (lpf_value * 1000)) + "\n");
        StreamNormalizer::writeSample[format]( out, (int)lpf_value );

        in += bytesPerSample; 
        out += bytesPerSample; 
    }
}

```

microbit-carlos, 2023-05-17:
@martinwork this should be fixed in the latest v0.2.52 tag, could you retest and close if it's all done?
Thanks!

martinwork, 2023-05-18:
@microbit-carlos Seems to be OK now - tested C++ sample, not MakeCode.

closed 2023-06-16 by microbit-carlos