lancaster-university/codal-microbit-v2 #285: Missing UART baud rates

microbit-carlos, 2023-03-09:
Related to:
- https://github.com/lancaster-university/codal-nrf52/issues/38

There are more baud rates in MakeCode that are compatible with the nRF52, but are not included in CODAL:

So we need to add `1200`, `2400`, `4800`, `14400`, `19200`, and `28800`.

https://github.com/lancaster-university/codal-nrf52/blob/5fa1d16d6de6bd2848ae776e3039ce6bcad4e3e1/source/NRF52Serial.cpp#L197-L207

[image]

[image]

[image]

whaleygeek, 2025-01-10:
@microbit-carlos Also 250,000 is vital for DMX, can we add that to the list to expose as well please? The chip does support it. 

In fact, I don't really understand why all of those supported baud rates aren't just exposed all the way from top of the stack to right down to the chip? - why don't we just pass the integer from the app all the way down to CODAL, rather than having multiple methods of setting the baud rate at different levels?

I *might* have to drop to MicroPython and use Carlos's magic register write trick to set the correct baud rate constant directly in the chip register to get this project to work. That's always a bit brittle to chip variants and system clock speed changes though.

Carlos suggested sub-classing and exposing a new setBaudRate, but that drags me into a tool-setup universe that I don't really want to get in to for such a small project.

microbit-carlos, 2025-01-10:
> @microbit-carlos Also 250,000 is vital for DMX, can we add that to the list to expose as well please? The chip does support it.

Yes, we can add that as well 👍

> In fact, I don't really understand why all of those supported baud rates aren't just exposed all the way from top of the stack to right down to the chip? - why don't we just pass the integer from [...]

Makecode and MicroPython pass the baud as an integer to the `NRF52Serial::setBaudrate()` method, which sets the register value via `nrf_uarte_baudrate_set()`:
https://github.com/lancaster-university/codal-nrf52/blob/5fa1d16d6de6bd2848ae776e3039ce6bcad4e3e1/source/NRF52Serial.cpp#L209
https://github.com/NordicSemiconductor/nrfx/blob/85c444ee0b76272d8a074b82845f04e92b3253c4/hal/nrf_uarte.h#L622

But we cannot just put a baud number directly in the register, we need to use a lookup table to write the value specified in the datasheet and shown in the screenshot above. So this lookup table needs to be in the HAL, as different chips could have different values.

> I _might_ have to drop to MicroPython and use Carlos's magic register write trick to set the correct baud rate constant directly in the chip register to get this project to work. That's always a bit brittle to chip variants and system clock speed changes though.

Yes, this should work as well, and V1 or V2 could be detected in Python code to provide a different register value for V1 and V2 compatibility.
You could also do direct register access in a MakeCode extension, but it does require the extra steps to create an extension (unfortunately the MakeCode editor Javascript view doesn't let us create .cpp files, so it needs to be created locally).
Perhaps we should create a simple MakeCode extension with raw memory operations for this kind of use-case (only for advance users, so no need to send it via approval process).

> Carlos suggested sub-classing and exposing a new setBaudRate, but that drags me into a tool-setup universe that I don't really want to get in to for such a small project.

That's fair, the nice thing about MakeCode is that once you have an extension with C++ code you can do it all via the online editor, but that first step to create the extension does provide initial friction.