lancaster-university/codal-microbit-v2 #286: Compass recalibration fails about half the time

martinwork, 2023-03-10:
Arising from ticket  https://support.microbit.org/helpdesk/tickets/61000 (private)

After recalibration, by pressing A is the samples below, headings are limited to a small range. I see only East and North.

The original report used Python clearCalibration() and let heading() trigger calibration, with headings getting stuck on West.

It seems to go wrong every second recalibration.

MakeCode: https://makecode.microbit.org/_4u2C5qex3VLf
C++: [compass.zip](https://github.com/lancaster-university/codal-microbit-v2/files/10941293/compass.zip)

martinwork, 2023-03-26:
Some trace in MicroBitCompassCalibrator::calibrateUX shows alternate calibrations have very different centres.
The first one is a good calibration.
```
CompassCalibration cal = calibrate(data, samples);
DMESG( "cal centre %d %d %d scale %d %d %d radius %d", cal.centre.x, cal.centre.y, cal.centre.z, cal.scale.x,  cal.scale.y,  cal.scale.z, cal.radius);
compass.setCalibration(cal);

cal centre 26080 -17448 470 scale 1024 1024 1024 radius 48422
cal centre 2848 -1314 -3056 scale 1024 1024 1024 radius 48390
cal centre 24650 -15434 -200 scale 1024 1024 1024 radius 48723
cal centre 4390 -3044 -3780 scale 1024 1024 1024 radius 48713
cal centre 22852 -14082 912 scale 1024 1024 1024 radius 47044
cal centre 4202 -5206 -3510 scale 1024 1024 1024 radius 46693
```

martinwork, 2023-03-26:
The cal.scale values are always 1024, This is the same in the DAL.
https://github.com/lancaster-university/codal-microbit-v2/blob/50eef908efdb099d8d7397329a84bb1695c77fc0/source/MicroBitCompassCalibrator.cpp#L188

max has been redefined to convert to int
https://github.com/lancaster-university/codal-core/blob/350a794a6f43d8e91b4762f110bad675169dc908/inc/core/CodalCompat.h#L68

martinwork, 2023-03-26:
With the above problem fixed V2 still fails every other recalibration. V1 doesn't fail. Maybe the raw data is wrong?

martinwork, 2023-03-26:
I think the V1/V2 difference is that Compass::clearCalibration doesn't clear the calbration
https://github.com/lancaster-university/codal-core/blob/350a794a6f43d8e91b4762f110bad675169dc908/source/driver-models/Compass.cpp#L224

whereas MicroBitCompass::clearCalibration() does
https://github.com/lancaster-university/microbit-dal/blob/602153e9199c28c08fd2561ce7b43bf64e9e7394/source/drivers/MicroBitCompass.cpp#L285

sampleENU stores calibrated XYZ readings,
https://github.com/lancaster-university/codal-core/blob/master/source/driver-models/Compass.cpp#L317

and these get returned in compass.getSample(RAW)
https://github.com/lancaster-university/codal-core/blob/350a794a6f43d8e91b4762f110bad675169dc908/source/driver-models/Compass.cpp#L336
https://github.com/lancaster-university/codal-core/blob/master/source/types/CoordinateSystem.cpp#L72

I'll test tomorrow.

martinwork, 2023-03-27:
Created PR https://github.com/lancaster-university/codal-core/pull/159
New C++ test sample dumps calibration to serial: [compass.zip](https://github.com/lancaster-university/codal-microbit-v2/files/11080465/compass.zip)

martinwork, 2023-03-27:
Created PR https://github.com/lancaster-university/codal-microbit-v2/pull/290 to fix calibration scale.xyz

martinwork, 2023-07-13:
@JohnVidler This fix for this issue is in https://github.com/lancaster-university/codal-core/pull/159

JohnVidler, 2023-07-27:
Merged in @martinwork ‘s PR to fix this issue.

closed 2023-07-27 by JohnVidler