lancaster-university/codal-microbit-v2 #295: Problems loading components directly

crisgplj, 2023-03-31:
Whe i use from the oficial page

#include "MicroBit.h"

MicroBitDisplay display;

int main()
{
    while(1)
        display.scroll(":)");
}

Compile say:   error: no matching function for call to 'codal::MicroBitDisplay::MicroBitDisplay()'

Any idea?

I cant use the instance Microbit only components

If you try too 

MicroBitI2C i2c = MicroBitI2C(I2C_SDA0, I2C_SCL0);
MicroBitAccelerometer accelerometer = MicroBitAccelerometer(i2c);

compile: undefined reference to `MicroBitAccelerometer::coordinateSpace'

Thanks!!

microbit-carlos, 2023-04-05:
Hi @crisgplj,

You can see the available MicrobitDisplay constructor in:
https://github.com/lancaster-university/codal-microbit-v2/blob/be35cc1d13697a783e08658a94e3af2b6c1afc4e/inc/MicroBitDisplay.h#L53

And you can see how we initialise MicroBitAcceleromter in:
https://github.com/lancaster-university/codal-microbit-v2/blob/be35cc1d13697a783e08658a94e3af2b6c1afc4e/model/MicroBit.cpp#L89

crisgplj, 2023-04-05:
Hi! And this example is for other version? I do the same and dont work
https://lancaster-university.github.io/microbit-docs/advanced/

Thanks!

microbit-carlos, 2023-04-05:
Yes, those docs are for micro:bit V1, which uses the DAL (V2 uses DAL's evolution CODAL).
We are looking at updating the documentation, so some of the advance stuff like using individual components might not quite work for V2.
For that kind of thing looking at how the `uBit` object is created is the best way to figure out how to initialise the individual components.

crisgplj, 2023-04-06:
Okay, thanks!

microbit-carlos, 2023-04-06:
I'll close this as resolve if that's okay? Feel free to reopen if you have any questions 👍

closed 2023-04-06 by microbit-carlos

crisgplj, 2023-04-15:
Hi, I know this is out of your context but you all know what I'm doing wrong.

I am interacting with the engine of the maqueen but it does not work

I know that in awesome microbit, is microbit:maqueen for arduino but I need to work at a low level

I am using the function

//0x10 motor direction
uBit.i2c.write(0x10, data, size of (data), false);

send data as a uint8_t data[3]={0};

//Indicate which motor
where data[0]=0
//Address
date[1]=0
//Speed
data[2]=255

Thanks!!!