lancaster-university/codal-microbit-v2 #301: `uBit.audio.levelSPL->getValue()` takes around 40ms to get valid values

microbit-carlos, 2023-04-06:
With a similar programme than https://github.com/lancaster-university/codal-microbit-v2/issues/300, when it works correctly it still takes 40 ms from `uBit.audio.enable()` before `uBit.audio.levelSPL->getValue()` starts returning good values.

Tweaking that specific programme, so that it prints a levelSPL value every 5ms (instead of 10, although the ticker is configured at 4ms). Fort the first 40-ish ms, about the first half of values are `0`, and the second half are value that is clearly larger than reality (tested in a relatively quiet room), not always the same "high" value, but definitely higher than expected. After that we start seeing reasonable values

```
0
0
0
0
153
153
153
63   <-- From here the values look reasonable
63
58
58
58
53
```

```
0
0
0
0
129
129
129
66  <-- From here the values look reasonable
66
56
56
56
68
```

microbit-carlos, 2023-04-20:
First time the sound level is read we can block until we have enough samples to produce a valid value.

microbit-carlos, 2023-05-17:
This is still an issue, which probably explains why the quiet and loud events are still raised on startup.

With the 0.2.52 tag I'm getting this output:
```
0
0
122
122
75
75
73
73
...
```

JohnVidler, 2023-07-04:
On `master` the levelSPL object will now block (deschedule) unitl at least 2 valid ADC buffers have been seen, this seems to eliminate this issue, but can also be tuned with a config parameter.

Pending further testing, but I think this is now fixed.

microbit-carlos, 2023-07-06:
I'm not seeing much difference here between tag v0.2.55 and the latest commits in the main branches.

Slightly modified the example to reduce the sleep time in the loop and also print the loud and quiet events:

```cpp
#include "MicroBit.h"
#include "CodalDmesg.h"

MicroBit uBit;

void onLoud(MicroBitEvent) {
    DMESG("LOUD");
}

void onQuiet(MicroBitEvent) {
    DMESG("QUIET");
}

int main() {
    uBit.init();

    uBit.messageBus.listen(DEVICE_ID_SYSTEM_LEVEL_DETECTOR, LEVEL_THRESHOLD_HIGH, onLoud);
    uBit.messageBus.listen(DEVICE_ID_SYSTEM_LEVEL_DETECTOR, LEVEL_THRESHOLD_LOW, onQuiet);

    uBit.audio.enable();
    uBit.audio.levelSPL->setUnit(LEVEL_DETECTOR_SPL_8BIT);
    for (int i = 0; i < 100; i++) {
        DMESG("[%d] %d", uBit.systemTime(), (int)uBit.audio.levelSPL->getValue());
        uBit.sleep(1);
    }

    while (true);
}
```

With v0.2.55:

```
relocate vtor to 0x0 -> 0x20002200 0x20002200
[160] 0
[164] 0
[168] 0
[172] 0
[176] 0
[180] 0
[184] 123
LOUD
[188] 123
[192] 123
[196] 123
[200] 123
[204] 123
QUIET
[208] 68
[212] 68
[216] 68
[220] 68
[224] 68
[228] 68
[232] 61
[236] 61
[240] 61
[244] 61
```

And with the latest commits:

```
relocate vtor to 0x0 -> 0x20002200 0x20002200
[160] 0
[164] 0
[168] 0
[172] 0
[176] 0
[180] 0
[184] 126
LOUD
[188] 126
[192] 126
[196] 126
[200] 126
[204] 126
QUIET
[208] 66
[212] 66
[216] 66
[220] 66
[224] 66
```

microbit-carlos, 2023-07-10:
My last test was checking out the hashes from the target-lock.json file, so I had to add the `"dev": true` entry to `codal.json` to pull the latest commits from all dependencies.

With that done I am getting the expected results 🎉 
```
relocate vtor to 0x0 -> 0x20002200 0x20002200
[160] 70
QUIET
[232] 70
[236] 66
[256] 66
[260] 66
[280] 66
[284] 66
[300] 66
[304] 66
[324] 66
```

```
$ python build.py -s

***/Users/microbit-carlos/workspace/mbef/codal/microbit-v2-samples/libraries/codal-core
Branch: master, Nearest Tag: streamracefix-v0 (c24d50c3025a2ab6202c34aded96824d3ea07777)
## master...origin/master

***/Users/microbit-carlos/workspace/mbef/codal/microbit-v2-samples/libraries/codal-nrf52
Branch: master, Nearest Tag: ~none~ (84b1aeec870addcfb6027a924dbfe83f1f10c6db)
## master...origin/master

***/Users/microbit-carlos/workspace/mbef/codal/microbit-v2-samples/libraries/codal-microbit-nrf5sdk
Branch: master, Nearest Tag: ~none~ (ef4662e13875a7b03e7296d7ac24a2b4d231f323)
## master...origin/master

***/Users/microbit-carlos/workspace/mbef/codal/microbit-v2-samples/libraries/codal-microbit-v2
Branch: master, Nearest Tag: v0.2.55 (02a85fa86a00ac204cbd2a4caacd5a964e1cf26e)
## master...origin/master

***/Users/microbit-carlos/workspace/mbef/codal/microbit-v2-samples
Branch: master, Nearest Tag: v0.2.11 (f4aaca3a9ff55dfd79c9ec41024dbbc0bec4de04)
## master...origin/master
 M codal.json
 M source/main.cpp
```

microbit-carlos, 2023-07-10:
@JohnVidler one thing I've noticed is that the first value is returned as early as before, and the delay seems to be introduced between the first and second value:

Before:
```
relocate vtor to 0x0 -> 0x20002200 0x20002200
[160] 0
[164] 0
[168] 0
[172] 0
[176] 0
[180] 0
[184] 126
LOUD
[188] 126
[192] 126
[196] 126
[200] 126
[204] 126
QUIET
[208] 66
```

I've also removed the LOUD/QUIET events to ensure they don't add an additional delay.

Current (both measurements after a flash, which takes a bit longer to get the first value, not sure why):

```
relocate vtor to 0x0 -> 0x20002200 0x20002200
[160] 70
[232] 70
[236] 66
```
```
relocate vtor to 0x0 -> 0x20002200 0x20002200
[160] 66
[232] 66
[236] 66
```

Current (timings from reset by pressing the reset button):
```
relocate vtor to 0x0 -> 0x20002200 0x20002200
[46] 66
[118] 66
[122] 66
```

```
relocate vtor to 0x0 -> 0x20002200 0x20002200
[46] 192
[118] 192
[122] 194
```

The both situations the new delay is also 72 milliseconds, is that more or less the expected delay?
Before the fix we were getting valid data after around 48ms.

microbit-carlos, 2023-07-10:
Okay, just noticed the reason the delay appears to be between the 1st and 2nd call to `uBit.audio.levelSPL->getValue()` it's because of my code, not related to CODAL
(`DMESG("[%d] %d", uBit.systemTime(), (int)uBit.audio.levelSPL->getValue());`).

However, I guess the question still stands if we can reduce the delayed time a little bit from the current 70ms to something closer to the 48ms from the initial report, or maybe it's not forth the effort?

microbit-carlos, 2023-07-14:
@JohnVidler will do a final update to reduce the startup delay and this will be tagged for v0.2.56.

JohnVidler, 2023-07-17:
I've re-measured the timings with a very minor tweak to the logic here, such that we allow the level detector to unblock any waiting fibers on the _second_ buffer, rather than the _third_, and we get the following:

```
// BOOT
0       level = 81    <-- Time printed BEFORE the getLevel() call, so initially zero
97      level = 81    <-- 97ms later we've got a value and looped around (cold start delay)
101     level = 75
117     level = 75
121     level = 71
0       level = 75    <-- Release and press the sample button, resets the timer
1       level = 66    <-- 1ms later we get a result (hot start minimum time, the MIC was still 'hot' here)
5       level = 113
25      level = 113
29      level = 104
49      level = 104
53      level = 100
69      level = 100
0       level = 101  <-- Release and press the sample button, resets the timer
5       level = 101  <-- 5ms later (ish) we get a result (warm start, the mic was 'off' but the ADC was running)
9       level = 102
21      level = 102
25      level = 98
45      level = 98
49      level = 102
```

This seems repeatable on my test kit, so we're looking at something like 97ms on a 'cold start' (no adc, mic off), <1ms for a 'hot start' (mic on, adc on), and 5ms for a 'warm start' (mic off, adc on).

microbit-carlos, 2023-07-17:
That's a bit worse than I was measuring in my test code, could you run this program with your branches and compare results?

```cpp
#include "MicroBit.h"
#include "CodalDmesg.h"

MicroBit uBit;

int main() {
    uBit.init();

    uBit.audio.levelSPL->setUnit(LEVEL_DETECTOR_SPL_8BIT);
    uBit.audio.enable();

    DMESG("[%d]", uBit.systemTime());
    for (int i = 0; i < 100; i++) {
        int audioLevel = (int)uBit.audio.levelSPL->getValue();
        DMESG("[%d] %d", uBit.systemTime(), audioLevel);
        uBit.sleep(1);
    }

    while (true);
}
```

With this I get 70 ms:

```
relocate vtor to 0x0 -> 0x20002200 0x20002200
[46]
[116] 162
```

JohnVidler, 2023-07-24:
This should now be fixed and pushed to live in PXT - I'm closing this for now, but do re-open if this is a problem again.

closed 2023-07-24 by JohnVidler

microbit-carlos, 2023-07-24:
Okay, so with the latest v0.2.56 tag we still get a 70ms delay in the first call to `uBit.audio.levelSPL->getValue()` (about three 255 byte buffers at 11000 sample rate), but it sounds like we cannot improve that.