lancaster-university/codal-microbit-v2 #309: microphone values not higher than decimal 131 ... ?

6a5h4, 2023-04-30:
i tried to program it with microsoft makecode editor in blocks, but i saw, the output value of the microphone is not 0-255 as told, its only 0-131 ... what i have to do to get a full 8-bit level?

6a5h4, 2023-05-07:
Hi @microbit-carlos ,

i tested now v3-beta and in blocks it works (mic volume up to 255 possible), but v3-beta has another big problem that throws me back for 3 days! And there it is:

I programmed in phyton until now, and it seems to work, but every change was not applied to the .hex-file, so i ever copied the same phyton software to the micro:bit-v2, so it couldn't work as expected. I needed 3 days to discover that the compiler only compiles the blocks-Programm, never the phyton program editor changes... so one project had more than one software version in it!
I switched above back to blocks and was wondering to see my blocks-program that was 3 days old (!!) ... think you need to rework a lot of the makecode-editor before you can leave the beta status...

microbit-carlos, 2023-05-09:
Hi @6a5h4,

Thanks for the report, could you open a new issue in the https://github.com/microsoft/pxt-microbit/issues tracker to let the MakeCode team the issues you've found? Thanks!

microbit-carlos, 2023-11-03:
I'll close this as the original CODAL issue has been resolved and any MakeCode editor specific issues should be open in https://github.com/microsoft/pxt-microbit/issues.
Thanks for the report!

closed 2023-11-03 by microbit-carlos