lancaster-university/codal-microbit-v2 #313: Driving the speaker pin with hardware PWM produces only noise the first time

microbit-carlos, 2023-05-05:
description incoming.

microbit-carlos, 2023-06-16:
This is still replicable in v0.2.55.

microbit-carlos, 2023-07-27:
I can confirm this is still replicable with CODAL v0.2.57

martinwork, 2023-07-27:
The first call to setAnalogPeriodUs does nothing because the pin hasn't been configured as an analogue output.
https://github.com/martinwork/codal-nrf52/blob/master/source/NRF52Pin.cpp#L633

microbit-carlos, 2023-07-28:
Thanks Martin!
I can confirm that switch the order and executing first `setAnalogValue()` and then `setAnalogPeriodUs()` fixes the issue.

However, it makes sense for the user to want to set up the analog out configuration before changing the pin mode to start outputting the signal.

martinwork, 2023-07-29:
Yes, I don't know why it doesn't call setAnalogValue(0) if the pin isn't already configured, but it might be awkward to remember the setting until setAnalogValue is called.
https://github.com/lancaster-university/codal-nrf52/blob/master/source/NRF52Pin.cpp#L631

The DAL would have to change too.
https://github.com/lancaster-university/microbit-dal/blob/js-event-semantics/source/drivers/MicroBitPin.cpp#L417

It can't find an explicit mention in the CODAL source or Python docs, but it's there in DAL and MakeCode.
https://github.com/lancaster-university/microbit-dal/blob/js-event-semantics/source/drivers/MicroBitPin.cpp#L413
https://lancaster-university.github.io/microbit-docs/ubit/io/#setanalogperiodus
https://makecode.microbit.org/reference/pins/analog-set-period
https://microbit-micropython.readthedocs.io/en/v2-docs/pin.html?highlight=write_analog#microbit.MicroBitDigitalPin.set_analog_period

microbit-carlos, 2023-08-11:
@finneyj this MicroPython issue is also related:
- https://github.com/microbit-foundation/micropython-microbit-v2/issues/143