lancaster-university/codal-microbit-v2 #315: Make the Sound Expressions effect default parameters/steps CODAL defaults

microbit-carlos, 2023-05-15:
Neither MakeCode nor MicroPython expose to the users a way to change the fx params and steps.
For consistency we should use the same values for both platforms, to be able to generate the same sounds.
In that case it makes sense to make these the default values inside CODAL instead of having both platform setting the same values.

https://github.com/microsoft/pxt/blob/5e2ff19c7d44bc659b997530c96b7ac9ad6a6551/webapp/src/components/soundEffectEditor/soundUtil.ts#L24-L42

martinwork, 2024-01-02:
Reported as a Python problem in support ticket https://support.microbit.org/helpdesk/tickets/70448 (private)

microbit-carlos, 2024-03-26:
@JohnVidler the preferred MicroPython solution is to set the parameters as #defines and MicroPython can use those:
- https://github.com/microbit-foundation/micropython-microbit-v2/issues/155#issuecomment-2019118084

JohnVidler, 2024-03-26:
I've defined a few SFX defaults over in this PR, would this do the trick?

https://github.com/lancaster-university/codal-microbit-v2/pull/417

microbit-carlos, 2024-03-27:
PR has been merged, next steps is to check that MicroPython is implementing it and submitting a PR to MakeCode.

microbit-carlos, 2024-03-28:
MicroPython work tracked in:
- https://github.com/microbit-foundation/micropython-microbit-v2/issues/155#issuecomment-1996714623

MakeCode in:
- https://github.com/microsoft/pxt-microbit/issues/5569

This CODAL issue can be closed as complete.

closed 2024-03-28 by microbit-carlos