lancaster-university/codal-microbit-v2 #327: MY_DATA on iPad - mime type of Download CSV data is "text/plain"

martinwork, 2023-05-21:
Could the mime type of MY_DATA Download CSV be "text/csv"?
I want to convert the mime type to a file extension, needed because I haven't managed to find the file name associated with the blob.

microbit-carlos, 2023-06-16:
Sounds very reasonable, could you raise a PR with the change? Thanks!

martinwork, 2023-06-16:
@microbit-carlos Will changing this do it?
https://github.com/lancaster-university/codal-microbit-v2/blob/e4caa98354432f83c65a2b3b23829677e570211d/resources/logfs/header.html#LL48C65-L48C75

Does that get replaced by the online version?

martinwork, 2023-06-20:
PR https://github.com/lancaster-university/codal-microbit-v2/pull/339

microbit-carlos, 2023-11-13:
PR https://github.com/lancaster-university/codal-microbit-v2/pull/339 merged 🎉

closed 2023-11-13 by microbit-carlos