lancaster-university/codal-microbit-v2 #333: uBit.audio.levelSPL->upstream.isConnected() is always false

martinwork, 2023-05-24:
There's a problem with the inheritance of isConnected() 

uBit.audio.levelSPL->upstream is a DataSource, initialised with a SplitterChannel.

uBit.audio.levelSPL->upstream.isConnected() seems to be finding DataSource::isConnected, not SplitterChannel::isConnected.

I guess at least DataSource::isConnected should be virtual.
https://github.com/lancaster-university/codal-core/blob/bf65ebed2003049bd026e1f685bb756e189c9258/inc/streams/DataStream.h#L70

At the moment, this test always succeeds.
https://github.com/lancaster-university/codal-core/blob/bf65ebed2003049bd026e1f685bb756e189c9258/source/streams/LevelDetectorSPL.cpp#L211

```
#include "MicroBit.h"

MicroBit uBit;

void forever()
{
    while (true)
    {
        DataSource       *datasource   = &uBit.audio.levelSPL->upstream;
        SplitterChannel  *channel      = (SplitterChannel *) datasource;
        bool              dsConnected  = datasource->isConnected();
        bool              chConnected  = channel->isConnected();
        bool              connected    = uBit.audio.levelSPL->upstream.isConnected();

        CODAL_TIMESTAMP t0 = system_timer_current_time_us();
        int level = uBit.audio.levelSPL->getValue();
        CODAL_TIMESTAMP t1 = system_timer_current_time_us();

        uBit.serial.printf( "%d %d %d %d %d\n", (int) ( connected ? 1 : 0), (int) ( dsConnected ? 1 : 0), (int) ( chConnected ? 1 : 0), (int) level, (int)( t1 - t0));

        if ( level > 0)
        {
            uBit.display.image.setPixelValue( 0, 0, uBit.display.image.getPixelValue(0, 0) ? 0 : 255);
            uBit.sleep(50);
            uBit.display.image.setPixelValue( 0, 0, uBit.display.image.getPixelValue(0, 0) ? 0 : 255);
        }
        uBit.sleep(20);
    }
}

int main()
{
    uBit.init();
    //uBit.audio.activateMic();
    create_fiber( forever);
    release_fiber();
}

```

microbit-carlos, 2023-06-14:
Hi @JohnVidler,  I noticed I'm assigned to this ticket, do you need my input for something specific?

JohnVidler, 2023-06-14:
Honestly not sure why you're on this one. I'll reassign it!

JohnVidler, 2023-07-04:
Now fixed on master, see https://github.com/lancaster-university/codal-core/blob/master/inc/streams/DataStream.h#L70

The object routing seems to be ok now, so marking this as done

closed 2023-07-04 by JohnVidler