lancaster-university/codal-microbit-v2 #338: MakeCode CI failure

microbit-carlos, 2023-06-15:
Currently the MakeCode CI builds are failing:
https://github.com/lancaster-university/codal-microbit-v2/actions/workflows/makecode.yml
[image]

The error is not very descriptive:

```
building bundled libs/blocksprj
*** yotta target bbc-microbit-classic-gcc@https://github.com/lancaster-university/yotta-target-bbc-microbit-classic-gcc
info: get versions for bbc-microbit-classic-gcc
error: Expecting value: line 1 column 1 (char 0)
INTERNAL ERROR: Error: yotta target bbc-microbit-classic-gcc@https://github.com/lancaster-university/yotta-target-bbc-microbit-classic-gcc: exit code 255
    at ChildProcess.<anonymous> (/home/runner/work/codal-microbit-v2/codal-microbit-v2/node_modules/pxt-core/built/buildengine.js:293:24)
    at ChildProcess.emit (node:events:513:28)
    at maybeClose (node:internal/child_process:1100:16)
    at Process.ChildProcess._handle.onexit (node:internal/child_process:304:5)
```

```
Digest: sha256:48fe950715eabc42c6438de7a5c81f15ccdfb7c79b808b6b2fc2f3c30e733c5c
Status: Downloaded newer image for pext/yotta:update-yotta3
/usr/local/lib/python2.7/dist-packages/jwt/utils.py:8: CryptographyDeprecationWarning: Python 2 is no longer supported by the Python core team. Support for it is now deprecated in cryptography, and will be removed in the next release.
  from cryptography.hazmat.primitives.asymmetric.utils import (
info: get versions for bbc-microbit-classic-gcc
error: No JSON object could be decoded
INTERNAL ERROR: Error: Exit code: 255 from cd built/dockeryt; docker run --rm -v /home/runner/work/codal-microbit-v2/codal-microbit-v2/libs/blocksprj/built/dockeryt/:/src -w /src --env YOTTA_GITHUB_AUTHTOKEN=*** pext/yotta:update-yotta3 yotta target bbc-microbit-classic-gcc@https://github.com/lancaster-university/yotta-target-bbc-microbit-classic-gcc
    at ChildProcess.<anonymous> (/home/runner/work/codal-microbit-v2/codal-microbit-v2/node_modules/pxt-core/built/nodeutil.js:78:24)
    at ChildProcess.emit (node:events:513:28)
    at maybeClose (node:internal/child_process:1100:16)
    at Process.ChildProcess._handle.onexit (node:internal/child_process:304:5)
```

However, they run correctly in my fork:
https://github.com/microbit-carlos/codal-microbit-v2/actions/runs/5282488564

The only difference I can think of is the population of the `` secret in the repo settings?

microbit-carlos, 2023-06-15:
I've pushed to this repo a branch which only builds CODAL with MakeCode using the `PXT_COMPILE_SWITCHES=csv---mbcodal` env var, and that runs correctly, so the issue is isolated to the DAL/yotta build:
https://github.com/lancaster-university/codal-microbit-v2/actions/runs/5282570183

JohnVidler, 2023-07-04:
Do we have a way forward for the deprecation of yotta? Otherwise this is just going to keep reappearing :(

microbit-carlos, 2023-07-04:
The proposal is to replace it with CODAL-style CMake files and build.py, but in the meantime we'll need to to continue using Yotta with GitHub tokens.

microbit-carlos, 2023-07-14:
@JohnVidler as the current branch will be tagged for MakeCode live, could you fix this issue before the tag (and restart the latest workflow run) so that we can get a CI build of MakeCode before tagging?

microbit-carlos, 2023-07-14:
Okay, so the docker build will fail anyway due to https://github.com/microsoft/pxt-microbit/issues/5313, but the "local build" (`nodocker`) in CI should still work to completion once this is fixed.

microbit-carlos, 2023-07-31:
I've updated the token in this repo with a fine grained read-only token from my account.
That fixes the issue building the "no docker" MakeCode build, but due to https://github.com/lancaster-university/codal-microbit-v2/issues/338 the "docker" build will fail, as the docker image is not available in Docker Hub.

As a temporary measure, to be able to tell if the `nodocker` builds fail, I'll disable the build with docker until we can fetch the image from a public registry.

microbit-carlos, 2024-02-27:
The docker image has  been pushed to DockerHub and we can restore the MakeCode builds with Docker:
- https://github.com/lancaster-university/codal-microbit-v2/pull/407

closed 2024-02-27 by microbit-carlos