lancaster-university/codal-microbit-v2 #340: Memory leak when SerialStreamer is used

microbit-carlos, 2023-06-20:
This programme runs for about 1 min 30 s and then we get a [MICROBIT_HEAP_ERROR 030 error](https://github.com/lancaster-university/pxt-microbit-samsung/blob/c92e3b4cc700bb6c0cbb2a86c119bf38e09cd07f/sim/state/misc.ts#L14).

```cpp
#include "MicroBit.h"
#include "SerialStreamer.h"

MicroBit uBit;

int main() {
    uBit.init();

    SerialStreamer *streamer = new SerialStreamer(uBit.audio.processor->output, SERIAL_STREAM_MODE_BINARY);
    uBit.audio.requestActivation();
    uBit.audio.activateMic();

    while (true) {
        uBit.sleep(1000);
    }
}
```

martinwork, 2023-09-19:
Most often the panic is in isReadOnlyInline() from incr() when refCount == 1, implying a buffer being copied has been destroyed. I have also seen a panic from decr().
https://github.com/lancaster-university/codal-core/blob/master/source/types/RefCounted.cpp#L52

It doesn’t seem to be heap corruption, but that a RefCounted buffer has been released too many times and destroyed.

I have seen a panic in CodalHeapAllocator when a ref counted buffer was being freed.
https://github.com/lancaster-university/codal-core/blob/master/source/core/CodalHeapAllocator.cpp#L378

I have seen SerialStreamer::lastBuffer crash when assigning a new pull because it’s previous ref counted object has been freed.
[image]

I think one problem is that the buffer operations are not interrupt safe.

I have analysed one example, built with GNU 12.3.1, involving DataStream::nextBuffer, which seems to show DataStream::pull() interrupted at the wrong instant, for a visit to DataStream::pullRequest(), which leads to a double release of DataStream::nextBuffer.
https://github.com/lancaster-university/codal-core/blob/master/source/streams/DataStream.cpp#L160
https://github.com/lancaster-university/codal-core/blob/master/source/streams/DataStream.cpp#L193

A panic occurs as DataStream::pull() tries to return tmp, because its ptr has been released and destroyed.
[image]

Line numbers may be different because of added trace code:
https://github.com/martinwork/microbit-v2-samples/tree/test-serialstreamer
https://github.com/martinwork/codal-core/tree/test-serialstreamer
https://github.com/martinwork/codal-microbit-v2/tree/test-refcounted

Without the trace code additions, it can take a lot longer to panic, but it does eventually.

See the log below. As I understand it...
- In DataStream::pull(), this->nextBuffer = ManagedBuffer() is interrupted inside decr(), before the refCount is reduced, and before the ptr (0x20006EF0) is changed.
- Another buffer (0x200024B8) using 6EF0 is reassigned, which reduces the refCount.
- Inside DataStream::pullRequest, nextBuffer, with the same ptr 6EF0, is reassigned, so it reduces the 6EF0 refCount again (I think this is the extra decrement).
- When the interrupt returns to the original nextBuffer decr() call, it will go ahead with the original intended decrement, which brings the ref count to 1.

<details>

<summary>Log of RefCounted operations</summary>

```
Key:
iBufBuf		incr in the constructor
dBuf=/iBuf= 	decr/incr in operator =
dBuf~ 	        decr in destructor

some lines linking addresses to variables have a dummy NULL ptr.

0x20006EF0,1,0x2001FD58::iBufBuf refCount == 1
```

|sequence|ptr|count|object|fn|
| ------------- | --- | ------------- | ------------- | ------------- |
|439185|0x00000000|0|0x20008A18|lastBuffer SerialStreamer::pullRequest|
|439186|0x00000000|0|**0x200024D4**|nextBuffer DataStream::pull|
|// tmp is assigned|
|439187|**0x20006EF0**|5|0x2001FD30|iBufBuf|
|439188|0x00000000|0|0x2001FD30|tmp DataStream::pull|
|// nextBuffer enters operator =, then decr|
|439189|**0x20006EF0**|7|**0x200024D4**|dBuf=|
|// note: no iBuf=|
|// interrupt?|
|439190|0x200092E4|3|0x2001FC74|iBufBuf|
|// another variable using 6EF0 is reassigned|
|// note the refCount is unchanged from above|
|439191|**0x20006EF0**|7|0x200024B8|dBuf=|
|439192|0x200092E4|5|0x200024B8|iBuf=|
|439193|0x00000000|0|**0x200024D4**|nextBuffer DataStream::pullRequest|
|439194|0x200092E4|7|0x2001FC48|iBufBuf|
|// nextBuffer is assigned, calling decr on 6EF0 again|
|439195|**0x20006EF0**|5|**0x200024D4**|dBuf=|
|439196|0x200092E4|9|**0x200024D4**|iBuf=|
|// now the 6EF0 refCount is 3|
|// other irrelevant ops|
|439197|0x200092E4|11|0x2001FC48|dBuf~|
|439198|0x200092E4|9|0x2001FC74|dBuf~|
|439199|0x20008C78|3|0x2001FC74|init|
|439200|0x200092E4|7|0x200025DC|dBuf=|
|439201|0x20008C78|3|0x200025DC|iBuf=|
|439202|0x20008C78|5|0x2001FC74|dBuf~|
|// return from interrupt?|
|// complete the original decr() and destroy|
|439203|**0x20006EF0**|1|**0x200024D4**|destroy|

</details>

martinwork, 2023-09-28:
@JohnVidler I didn't have much luck catching it in the debugger, but I added some DMESG to the relevant functions and captured repeated traces like below.

SerialStreamer::pullRequest seems indirectly to pull the nextBuffer from the same DataStream (at 0x200024C8) that demux writes to,

NRF52ADCChannel::demux calls the ADC channel's non-blocking output DataStream::pullRequest to update its nextBuffer.
```
NRF52ADCChannel::demux 0x200024B0 output 0x200024C8
DataStream::pullRequest 0x200024C8 DataSink 0x20006118 DataSource 0x200024B0 blocking 0
DataStream::pullRequest 0x200024C8 setting nextBuffer
NRF52ADCChannel::demux EXIT
```

Next StreamSplitter::pullRequest routes down through SplitterChannel and StreamNormalizer to SerialStreamer::pullRequest.
```
StreamSplitter::pullRequest
SplitterChannel::pullRequest 0x2000616C output 0x2000618C
StreamNormalizer::pullRequest output 0x200061B0
DataStream::pullRequest 0x200061B0 DataSink 0x20006FC8 DataSource 0x20006190 blocking 1
SerialStreamer::pullRequest
```

SerialStreamer::pullRequest calls pull on StreamNormalizer::output (a blocking DataStream) which routes via StreamNormalizer, SplitterChannel, and StreamSplitter::getBuffer to the ADC channel's output DataStream:pull, which returns the nextBuffer. 
```
DataStream::pull 0x200061B0 DataSink 0x20006FC8 DataSource 0x20006190 blocking 1
DataStream::pull 0x200061B0 calling upStream->pull()
StreamNormalizer::pull upstream 0x2000616C
SplitterChannel::pull 0x2000616C
StreamSplitter::getBuffer upstream 0x200024C8
DataStream::pull 0x200024C8 DataSink 0x20006118 DataSource 0x200024B0 blocking 0
DataStream::pull 0x200024C8 returning nextBuffer
SplitterChannel::pullRequest 0x200061D8 output 0x00000000
StreamSplitter::pullRequest EXIT
```

martinwork, 2025-04-15:
I think this might have been fixed by
https://github.com/lancaster-university/codal-core/commit/bf59d6c5c3987bf8938438b7b42bb2181dda3e2c

martinwork, 2025-04-16:
Ah, not fixed, but it takes longer than 1 min 30 s...

microbit-carlos, 2025-04-24:
Thanks for testing Martin!

With the main branch I had to create a channel with the splitter and configure it with a 11K sample rate (in CODAL 2.65 we changed the default mic rate from 11K to 44K and at that rate it seems to affect the serial streamer, perhaps the `pullRequest()` method is called faster than it's capable to output serial data). But even without doing that change, I could also replicate the issue after a few more minutes.

However, with the audio-refactor branch I only get 768 bytes out of the microphone and after that it just hangs there.  This is the code I used for testing, did you do something differently?

```cpp
#include "MicroBit.h"
#include "SerialStreamer.h"

MicroBit uBit;

int main() {
    uBit.init();

    static SplitterChannel *splitterChannel = uBit.audio.splitter->createChannel();
    // splitterChannel->requestSampleRate( 11000 ); // <-- This is only needed for builds from the main branch
    SerialStreamer *streamer = new SerialStreamer(*splitterChannel, SERIAL_STREAM_MODE_BINARY);
    uBit.audio.requestActivation();

    while (true) {
        uBit.sleep(1000);
    }
}
```

martinwork, 2025-04-24:
@microbit-carlos I think I used the original sample, and probably changed it to output decimal, because of using tera term.
https://github.com/lancaster-university/codal-microbit-v2/issues/340#issue-1765357264

I usually test with BLE pairing enabled. I might have had DMESG enabled, but I just tested it without (though not for long).
```
    "config":{
        "DEVICE_DMESG_BUFFER_SIZE":0,
        "DMESG_SERIAL_DEBUG":0,
        "DMESG_ENABLE":0,

        "MICROBIT_BLE_ENABLED": 0,
        "MICROBIT_BLE_PAIRING_MODE": 1,
        "MICROBIT_BLE_DFU_SERVICE": 1,
        "MICROBIT_BLE_DEVICE_INFORMATION_SERVICE": 1,
        "MICROBIT_BLE_EVENT_SERVICE": 1,
        "MICROBIT_BLE_PARTIAL_FLASHING": 0,
        "MICROBIT_BLE_OPEN": 0,
        "MICROBIT_BLE_SECURITY_LEVEL": "SECURITY_MODE_ENCRYPTION_NO_MITM"
    }
```

I think `uBit.audio.activateMic();` is required in the start up to make it keep going. 

On the other hand, I don't think uBit.audio.requestActivation(); is needed, because that's about the output side.

I'll leave the new sample running....

martinwork, 2025-04-25:
The new sample with audio refactor branch went 030 sometime overnight.

microbit-carlos, 2025-04-25:
Thanks Martin! It seems to be related to pairing mode somehow?
- https://github.com/lancaster-university/codal-microbit-v2/issues/479

However, for me the first time I loaded this hex file it threw a 030 error in a couple of minutes, but I haven't been able to replicate it again in a few 5min attempts (I'll leave it running for a while to see what happens):

[MICROBIT-serialstreamer-binary-030.hex.zip](https://github.com/user-attachments/files/19906236/MICROBIT-serialstreamer-binary-030.hex.zip)

```cpp
#include "MicroBit.h"

#include "SerialStreamer.h"

MicroBit uBit;

int main() {
    uBit.init();

    static SplitterChannel *splitterChannel = uBit.audio.splitter->createChannel();
    SerialStreamer *streamer = new SerialStreamer(*splitterChannel, SERIAL_STREAM_MODE_BINARY);
    uBit.audio.requestActivation();
    uBit.audio.activateMic();

    while (true) {
        uBit.sleep(1000);
    }
```

```json
{
    "target": {
        "name": "codal-microbit-v2",
        "url": "https://github.com/lancaster-university/codal-microbit-v2",
        "branch": "audio-refactor",
        "type": "git"
    },
    "config":{
        "MICROBIT_BLE_ENABLED" : 0,
        "MICROBIT_BLE_PAIRING_MODE": 1
    }
}
```

martinwork, 2025-04-25:
The original sample has shown 030 too, after a couple of hours. I was wondering if attaching to the splitter instead of the mic output was a factor, - apparently not.

microbit-carlos, 2025-04-25:
I did two other runs of 2h with the hex file from https://github.com/lancaster-university/codal-microbit-v2/issues/340#issuecomment-2829913032 and it didn't trigger the error in these attemprs. I'll see if I can leave it overnight at some point.

After that I left a very similar programme (with `splitterChannel->dataWanted(DATASTREAM_WANTED)`, as discussed in https://github.com/lancaster-university/codal-microbit-v2/issues/479) and that trigger the error after 33 min the first time, and after 24 min the second time. This second test was done after disconnecting and reconnecting the USB cable, but after that, the board was reset simply by disconnecting and reconnecting the serial console (a serial break command triggers a DAPLink SWD reset, similar to pressing the reset button). And each subsequent attempt took less time, it then replicated after 18 min, then after 6 min, and then after 5 min.
Not sure if the reset difference is a real factor or a red herring, but I thought it worth mentioning it.

Either way, while this new hex file seems to trigger it more easily (unsure why that might be), it is not a deterministic/consistent error with either hex file.

[MICROBIT-serialstreamer-datawanted-030.hex.zip](https://github.com/user-attachments/files/19912250/MICROBIT-serialstreamer-datawanted-030.hex.zip)

```cpp
#include "MicroBit.h"
#include "SerialStreamer.h"

MicroBit uBit;

int main() {
    uBit.init();

    static SplitterChannel *splitterChannel = uBit.audio.splitter->createChannel();
    SerialStreamer *streamer = new SerialStreamer(*splitterChannel, SERIAL_STREAM_MODE_BINARY);
    uBit.audio.requestActivation();
    splitterChannel->dataWanted(DATASTREAM_WANTED);

    while (true) {
        uBit.sleep(1000);
    }

}
```
```json
{
    "target": {
        "name": "codal-microbit-v2",
        "url": "https://github.com/lancaster-university/codal-microbit-v2",
        "branch": "audio-refactor",
        "type": "git"
    },
    "config":{
        "MICROBIT_BLE_ENABLED" : 0,
        "MICROBIT_BLE_PAIRING_MODE": 0
    }
}
```