lancaster-university/codal-microbit-v2 #345: `TouchSensor::removeTouchButton()` doesn't always work

microbit-carlos, 2023-06-28:
When a ping is registered as a TouchButton and it is set as digital high, the voltage on the pin is less than Vin. For example, when USB connected and the MCU is running at 3.3V, setting `uBit.io.logo.setDigitalValue(1)` produces ~2.6V on the pin.

Using `TouchSensor::removeTouchButton()` should technically remove the touch functionality from the pin, and setting it high should reach the 3.3V again. However, with this example programme we can see that is not always the case.

Button A sets the pin log high, and button B runs `removeTouchButton()`. To run this test we need a voltmeter to measure the voltage in the face logo.
Restarting the micro:bit and pressing A and the B and we can see that sometimes the voltage on the logo is 3.3V and sometimes is ~2.6V.
[MICROBIT.hex.zip](https://github.com/lancaster-university/codal-microbit-v2/files/11898343/MICROBIT.hex.zip)
```cpp
#include "MicroBit.h"

MicroBit uBit;

int main() {
    uBit.init();

    while (true) {
        if (uBit.buttonA.isPressed()) {
            uBit.io.logo.setDigitalValue(1);
            uBit.display.print("A");
        } else if (uBit.buttonB.isPressed()) {
            if (uBit.touchSensor.removeTouchButton(&uBit.logo) != DEVICE_OK) {
                uBit.display.print("X");
                while (true) uBit.sleep(1000);
            }
            //uBit.io.logo.setDigitalValue(0);
            uBit.display.print("B");
        } 
        uBit.sleep(250);
    }
}
```

JohnVidler, 2023-06-28:
Just to clarify - this is that the `HIGH` voltage is not reaching 3v3.

I've not been able to test this yet, but does this also happen if we use an edge connector pin in capacitive sensing mode, or is it just the logo that is affected?

microbit-carlos, 2023-06-28:
No sorry, that will be covered in:
- https://github.com/lancaster-university/codal-microbit-v2/issues/346

This is that running `uBit.touchSensor.removeTouchButton()` doesn't always have the same result.
Together with `setDigitalValue(1)` then pin voltage either goes to the expected 3.3V ✅, or it goes down to V1.7 ❌

JohnVidler, 2023-07-19:
Should this actually be the case? Technically as we don't track the pin state on anything other than the pin peripheral itself, by removing the touch sensor we're getting into an undefined state.

We _could_ proactively set the pin high or low, but we can't possibly know what the application requriements are for this, so is it better to leave this as it is, and require that if applications need a pin to be in a specific state after removing touch sensing they should actively drive high or low?

microbit-carlos, 2023-07-19:
Removing the touch sensor should leave the pin in whatever is the default state for any other digital pin, no? How are the "normal" digital pins configured on startup?

finneyj, 2023-08-02:
Thanks folks. I've looked into this - the issue was coming from the NRF52TouchSensor not letting go of the NRF52 GPIOTE channels when a button is disconnected.

I've run a fair few tests, including the ones listed above. This should now be fixed in main branches of the codal-core and codal-nrf52.

microbit-carlos, 2023-08-03:
Okay, we can close this one as fixed! 🎉