lancaster-university/codal-microbit-v2 #348: Triple press for Bluetooth mode - shorten animation

martinwork, 2023-07-03:
The delay while the screen is filled is only necessary for A+B+RESET, to check A and B are held.

https://github.com/lancaster-university/codal-microbit-v2/blob/master/model/MicroBit.cpp#L251

@jaustin Bypass the screen fill, or just speed it up?

jaustin, 2023-07-03:
I think make it really fast if `triple_reset` ?
https://github.com/lancaster-university/codal-microbit-v2/blob/master/model/MicroBit.cpp#L254C19-L254C19

I'd say drop this to 40ms in the A+B case in 10ms in the case of `triple_reset`?

microbit-carlos, 2023-08-03:
@martinwork is there a PR for this issue created already? We had the recollection of something maybe being prepared, but couldn't find it.

martinwork, 2023-08-03:
@microbit-carlos My recollection is that @jaustin asked me to create an issue, not a PR.

microbit-carlos, 2023-08-11:
Oh right, thanks for clarifying. Talking with Jonny, we definitely want to start working on this, would you be able to implement it?

microbit-carlos, 2023-08-14:
Thanks Martin!
I guess for consistency we should reduce the V1 DAL delay to 40ms as well, could you raise that PR too? Thanks!

martinwork, 2023-08-14:
@microbit-carlos I have updated PR https://github.com/lancaster-university/codal-microbit-v2/pull/368 with a 20ms delay as discussed. I'll create a V1 DAL PR for the js-event-semantics branch. also with a 20ms delay.

martinwork, 2023-08-14:
Created https://github.com/lancaster-university/microbit/pull/20

microbit-carlos, 2023-09-15:
PR https://github.com/lancaster-university/codal-microbit-v2/pull/368 has been merged and this can be closed as resolved 🎉 

The V1 DAL equivalent task is captured in:
- https://github.com/lancaster-university/microbit-dal/issues/491