lancaster-university/codal-microbit-v2 #349: How can I use the "BLETest" demo?

wxzed, 2023-07-04:
How can I use the "BLETest" demo? I have already modified codal.json, and I can see "BLE ABDILMTU/P" on the board. However, even after completing compass.calibrate, my phone is still unable to detect the BLE advertising packet. Is there anything else I need to do?

martinwork, 2023-07-04:
Depending on how you have modified codal.json, you may need to reset micro:bit to Bluetooth pairing mode (press RESET 3 times) in order to pair.

There is an example codal.json for Bluetooth in codal.ble.json, which uses just works mode.
https://github.com/lancaster-university/microbit-v2-samples/blob/master/codal.ble.json

wxzed, 2023-07-05:
Thank you for your response. When I set "MICROBIT_BLE_PAIRING_MODE": 1, I can enter pairing mode by pressing the reset button three times. After resetting again, I can see the Bluetooth broadcast. However, when I set "MICROBIT_BLE_PAIRING_MODE": 0, pressing the reset button three times does not enter pairing mode, indicating that the configuration is effective. However, I still cannot receive the BLE broadcast. I would like to be able to detect BLE broadcasts without going through the pairing process. How should I configure it?

martinwork, 2023-07-05:
See these options in https://github.com/lancaster-university/codal-microbit-v2/blob/master/inc/MicroBitConfig.h#L151

Set MICROBIT_BLE_OPEN to 1 to easily disable BLE security entirely, and give values to other relevant configurations
https://github.com/lancaster-university/codal-microbit-v2/blob/master/inc/MicroBitConfig.h#L179

wxzed, 2023-07-06:
Thank you for your response. Based on your reply, I have successfully implemented the functionality I desired.

closed 2023-07-06 by wxzed