lancaster-university/codal-microbit-v2 #354: `MICROBIT_BLE_UTILITY_SERVICE_PAIRING` defaults for MakeCode & MicroPython

microbit-carlos, 2023-07-24:
For MakeCode:
- In a normal project with Radio:
    - Without the datalogging extension the utility service shouldn't be compiled at all
    - If datalogging extension is used, it should add the utility service to pairing mode only (not application mode)
- When the BLE extension is added:
    - The utility service should always be added to the pairing mode
    - The utility service should always be added to the application mode 

For MicroPython:
-  The utility service should always be added to the pairing mode

martinwork, 2023-07-25:
Does this means the Data Logger extension needs to be modified to set both MICROBIT_BLE_UTILITY_SERVICE_PAIRING and MICROBIT_BLE_UTILITY_SERVICE to 1, maybe here:
https://github.com/microsoft/pxt-microbit/blob/master/libs/datalogger/pxt.json#L11
```
    "microbit": {
        "config": {
            "MICROBIT_BLE_UTILITY_SERVICE": "1",
            "MICROBIT_BLE_UTILITY_SERVICE_PAIRING": "1"
        }
    },
```
Or should those definitions be added to the flashlog extension?
https://github.com/microsoft/pxt-microbit/blob/master/libs/flashlog/pxt.json#L18

That should work because...

MICROBIT_BLE_UTILITY_SERVICE_PAIRING is checked when
CONFIG_ENABLED(DEVICE_BLE) && CONFIG_ENABLED(MICROBIT_BLE_PAIRING_MODE)
which is usually the case in MakeCode.

MICROBIT_BLE_UTILITY_SERVICE is checked when
CONFIG_ENABLED(DEVICE_BLE) && CONFIG_ENABLED(MICROBIT_BLE_ENABLED)
which is when the Bluetooth extension is included.

If the service should always be present when the Bluetooth extension is added, even if Data Logger is not, then perhaps the Bluetooth extension also needs to be modified to set both MICROBIT_BLE_UTILITY_SERVICE_PAIRING and MICROBIT_BLE_UTILITY_SERVICE to 1.

At the moment the CODAL service code is compiled when DEVICE_BLE is 1, like all other services. Do we need to wrap it in additional conditions, so it's only compiled when it might be used? 
https://github.com/lancaster-university/codal-microbit-v2/blob/master/source/bluetooth/MicroBitUtilityService.cpp#L31
https://github.com/lancaster-university/codal-microbit-v2/blob/master/inc/bluetooth/MicroBitUtilityService.h#L30

microbit-carlos, 2023-11-10:
Right now the `MICROBIT_BLE_UTILITY_SERVICE` and `MICROBIT_BLE_UTILITY_SERVICE_PAIRING` flags are disabled in CODAL by default.

For the next pxt release with the current CODAL we are enabling the `MICROBIT_BLE_UTILITY_SERVICE` and `MICROBIT_BLE_UTILITY_SERVICE_PAIRING` flags to the build, but only when the datalog extension is added. This is achieved via PR, pushed now to be able to have it out before a specific deadline:
- https://github.com/microsoft/pxt-microbit/pull/5346

This is way was done to avoid using the extra resources when datalogging is not used in MakeCode, which can be controlled via the presence of the datalog extension.

However we don't have a way to to determine if a CODAL user programme uses the feature until compile time, at which point it might be late to enable the flags for the rest of CODAL, like the pairing/bluetooth mode.
So the default for CODAL should be to have the features turned on.

For MicroPython builds we can control the flag for the pairing/bluetooth mode in the codal.json file, so either default is fine.

microbit-carlos, 2024-03-05:
Conclusion:
- We need to set the CODAL default to ON, so that CODAL and MicroPython programmes can have it enabled without additional user action (as there issn't a way to automatically enable it if this feature is used by user code)

microbit-carlos, 2024-03-27:
Change of plans 😓

Basically MakeCode guards agains the pxt core and an extension having different values for the config flags. Rather than request this feature to MakeCode let's change the approach to be able to complete this quickly.

So, we'll need to do the following:
- Leave the default **off** in CODAL
    - ✅ No action
- Ensure MicroPython build has this flag enabled 
     - ✅ PR https://github.com/microbit-foundation/micropython-microbit-v2/pull/177
- Ensure MakeCode has this flag enabled only in the datalog extension
    - ✅ No action
- Update the example coda.ble.json file to ensure these options are visible  
    - ✅ Done in https://github.com/lancaster-university/microbit-v2-samples/commit/5198bb3649da09bfd6ad22742549bbe05b389224

closed 2024-03-27 by microbit-carlos