lancaster-university/codal-microbit-v2 #356: Issues changing `splitterChannel` sample rate.

microbit-carlos, 2023-07-26:
From issue https://github.com/microsoft/pxt-microbit/issues/5312 as identified by @srietkerk:
- https://github.com/microsoft/pxt-microbit/issues/5312#issuecomment-1648663404

> @JohnVidler I think this is a CODAL issue that specifically has to do with `splitterChannel->requestSampleRate(float samplerate)`.
> 
> It took me a bit to realize, but the problem emerges with the bottom sample rate call being a larger value than the top sample rate call. Once the sample rate gets set to 18000 for the upstream, the value of the input sample rate will never change due to this check here: https://github.com/lancaster-university/codal-core/blob/763a2d1a916d4db3fe3f3f2b1eaa9947cc8d0a7e/source/streams/StreamSplitter.cpp#L126. If we try to press A again, the upstream input sample rate is larger than what we are trying to change the sample rate to (11000), so we never get to request that sample rate to the upstream and thus the input sample rate for stays at 18000 while the playback rate changes back down to 11000. Thus, when we go to change both the sample rate and the playback to 18000, nothing happens to the input sample rate, but the output sample rate goes back to 18000, thus making the playback on the second go-round and beyond sound not sped up.
> 
> This can be seen by trying the following program Changing the sample rate in the same two places works as expected. The playback is slow every time:
> [image]
> 
> This problem came to my attention when I noticed that just setting the playback sample rate in the second `set sample rate` call did not cause the same bug as noticed above.
> [image]

microbit-carlos, 2023-07-26:
@srietkerk would you be able to provide C++ sample code to replicate? That will help significantly.

martinwork, 2023-07-26:
Related: https://github.com/lancaster-university/codal-microbit-v2/issues/284

srietkerk, 2023-07-26:
I can't supply what it looks like in C++ from micro:bit directly, as far as I know. However, I can point you to how the extension is calling the sample rate function. Hopefully that helps.

Setting the sample rate happens here: https://github.com/microsoft/pxt-microbit/blob/c6a9d193378b90753be0a9b3d21fd40c4f013b09/libs/audio-recording/recording.cpp#L174

and here: https://github.com/microsoft/pxt-microbit/blob/c6a9d193378b90753be0a9b3d21fd40c4f013b09/libs/audio-recording/recording.cpp#L207

Let me know if there is any other way that I can help.

JohnVidler, 2024-02-26:
This should now be _finally_ fixed with the release of v0.2.65.

closed 2024-02-26 by JohnVidler