lancaster-university/codal-microbit-v2 #357: Pin 0 digital input may be high by default on V2 micro:bit

microbit-giles, 2023-07-26:
To reproduce:
- Go to https://microbit.org/projects/make-it-code-it/pir-movement-alarm/
- Download the transmitter HEX file and flash to micro:bit V2
- Both I and a colleague observe that the diamond is shown immediately when nothing is connected to the micro:bit. We expect nothing to be shown on the display when nothing is connected to the micro:bit. This suggests pin 0 may be floating or high by default.

The HEX file seems to work as expected on a micro:bit V1 - no diamond shown by default. Diamond is shown when 3v pin connected to pin 0, and disappears when disconnected.
[image]

Are there any mitigations we can add in code, or does this need some other kind of fix?

microbit-carlos, 2023-07-26:
Based on issue https://github.com/lancaster-university/codal-microbit-v2/issues/17 V2 should be configured to match V1 and set the pins default state to pull down:
- https://github.com/lancaster-university/codal-microbit-v2/issues/17

Looking at the source code we have the default pull configured to down in the target.json files:
https://github.com/lancaster-university/codal-microbit-v2/blob/3fa546f7ef5fd34cb9167b41045f8d7b6235b67e/target.json#L26
https://github.com/lancaster-university/codal-microbit-v2/blob/3fa546f7ef5fd34cb9167b41045f8d7b6235b67e/target-locked.json#L18

And the default mode is applied here:
- [codal-nrf52/source/NRF52Pin.cpp#L154-L156](https://github.com/lancaster-university/codal-nrf52/blob/84b1aeec870addcfb6027a924dbfe83f1f10c6db/source/NRF52Pin.cpp#L154-L156)

microbit-carlos, 2023-07-26:
> I think it must be an old hex file with a bug that has been fixed.
> 
> I see the same problem with the downloaded hex, but not with a newly built hex.

Ah, thanks @martinwork!
As this hex file is hosted in the website we'll have to update it in the CMS to a newer version of MakeCode.

microbit-carlos, 2023-08-02:
Okay, so the issue was from an old MakeCode release and has been fixed in current CODAL, so we can close this as fixed.