lancaster-university/codal-microbit-v2 #365: CodalAssert.cpp:26:5: error: 'codal_dmesg_flush' was not declared in this scope

martinwork, 2023-08-10:
These errors stop current master branches building when DMESG is not enabled.

```
../libraries/codal-core/source/core/CodalAssert.cpp: In function 'void __codal_assert__(const char*, int, bool, const char*, const char*)':
../libraries/codal-core/source/core/CodalAssert.cpp:26:5: error: 'codal_dmesg_flush' was not declared in this scope
   26 |     codal_dmesg_flush();
      |     ^~~~~~~~~~~~~~~~~
../libraries/codal-core/source/core/CodalAssert.cpp: In function 'void __codal_fault__(const char*, int, const char*)':
../libraries/codal-core/source/core/CodalAssert.cpp:38:5: error: 'codal_dmesg_flush' was not declared in this scope
   38 |     codal_dmesg_flush();
      |     ^~~~~~~~~~~~~~~~~
../libraries/codal-core/source/core/CodalAssert.cpp: In function 'void __codal_assert_pass__(const char*, int, const char*)':
../libraries/codal-core/source/core/CodalAssert.cpp:48:5: error: 'codal_dmesg_flush' was not declared in this scope
   48 |     codal_dmesg_flush();
      |     ^~~~~~~~~~~~~~~~~
../libraries/codal-core/source/core/CodalAssert.cpp: In function 'void __codal_assert_fail__(const char*, int, const char*)':
../libraries/codal-core/source/core/CodalAssert.cpp:63:5: error: 'codal_dmesg_flush' was not declared in this scope
   63 |     codal_dmesg_flush();
      |     ^~~~~~~~~~~~~~~~~
```

microbit-carlos, 2023-08-11:
I can confirm I get the same error when setting `"dev": true` in codal.json to checkout the latest commits.

JohnVidler, 2023-08-11:
Patched out as of https://github.com/lancaster-university/codal-core/commit/ff914443a8f1b96e99031a88b1d92061b0af1a88

closed 2023-08-11 by JohnVidler