lancaster-university/codal-microbit-v2 #375: MakeCode simple parrot programme runs out of memory

microbit-carlos, 2023-09-15:
This simple MakeCode parrot programme from https://github.com/microsoft/pxt-microbit/issues/5312 runs out of memory consistently after a few recordings:
-  https://makecode.microbit.org/_VziHPeD7a3XJ
[image]

```ts
input.onButtonPressed(Button.A, function () {
    record.setSampleRate(11000)
    basic.showString("?")
    record.startRecording(record.BlockingState.Blocking)
    basic.showLeds(`
        . # . . .
        . # # . .
        . # # # .
        . # # . .
        . # . . .
        `)
    record.setSampleRate(18000)
    record.playAudio(record.BlockingState.Blocking)
    basic.clearScreen()
})
```

It might run out of memory with error 020 after a couple of recordings, or a few more. So far I've always managed to trigger it within 5 attempts.

This can be replicated with MakeCode Live [v6.0.18](https://github.com/Microsoft/pxt-microbit/releases/tag/v6.0.18) with CODAL v0.2.57:
- https://makecode.microbit.org/
- https://makecode.microbit.org/v6.0.18

I can also be replicated with MakeCode [v6.0.16](https://github.com/Microsoft/pxt-microbit/releases/tag/v6.0.16) with CODAL v0.2.56:
- https://makecode.microbit.org/v6.0.16

Going back to version  [v6.0.15](https://github.com/Microsoft/pxt-microbit/releases/tag/v6.0.15) with CODAL v0.2.55 and I cannot replicate anymore:
- https://makecode.microbit.org/v6.0.15

So it looks like it was introduced in v0.2.56:
https://github.com/lancaster-university/codal-microbit-v2/compare/v0.2.55...v0.2.56

microbit-carlos, 2023-09-15:
On Monday we should figure out if we can issue quick hotfix or if we need to ask the MakeCode team to revert back to CODAL v0.2.55.

JohnVidler, 2023-09-18:
Just a note here to say I'm currently investigating this.

microbit-carlos, 2023-09-26:
Tag v0.2.60 has increased the maximum size for stream recordings, as a temporary workaround to this issue while we continue investigating it: https://github.com/lancaster-university/codal-core/commit/a1163541d10c0a0d3dd9b2e8ac1b8906100449ad

It's been merged in pxt-microbit and we are hoping to be able to release to MakeCode Live asap:
- https://github.com/microsoft/pxt-microbit/pull/5360

JohnVidler, 2023-11-01:
With the updated memory configuration in `v0.2.61` Makecode seems much happier, and I'm now unable to reproduce the 020 crash.

There seems to be another issue with the sample rate being forgotten on the second run, but I don't believe this is related to this particular issue, so I'm going to close this one as done for now.

[image]
(Makecode versions for reference)

closed 2023-11-01 by JohnVidler