lancaster-university/codal-microbit-v2 #376: Microphone takes longer to turn on the second time is used

microbit-carlos, 2023-09-15:
With simple MakeCode parrot programme from https://github.com/microsoft/pxt-microbit/issues/5312, when button A is pressed the first time the microphone goes online very quickly. Any subsquent button presses and the microphone takes about a second to turn on from the moment the button is pressed.:
-  https://makecode.microbit.org/_VziHPeD7a3XJ
[image]

```ts
input.onButtonPressed(Button.A, function () {
    record.setSampleRate(11000)
    basic.showString("?")
    record.startRecording(record.BlockingState.Blocking)
    basic.showLeds(`
        . # . . .
        . # # . .
        . # # # .
        . # # . .
        . # . . .
        `)
    record.setSampleRate(18000)
    record.playAudio(record.BlockingState.Blocking)
    basic.clearScreen()
})
```

Tested with MakeCode using CODAL versions v0.2.55 to v0.2.57, but it's possible it is present in previous versions as well, as I haven't tested earlier ones.

microbit-carlos, 2023-09-15:
Okay, tested in MakeCode v6.0.1 with CODAL v0.2.47 and the present is still present there.

microbit-carlos, 2023-09-20:
changing the `[ show string [ "?" ] ]` block with "show image" with 0 ms delay shows that in that case the mic LED comes up almost immediately after the image.
[image]

```ts
input.onButtonPressed(Button.A, function () {
    record.setSampleRate(11000)
    images.createImage(`
        . # # # .
        # . . . #
        # . . . #
        # . . . #
        . # # # .
        `).showImage(0, 0)
    record.startRecording(record.BlockingState.Blocking)
    basic.showLeds(`
        . # . . .
        . # # . .
        . # # # .
        . # # . .
        . # . . .
        `)
    record.setSampleRate(18000)
    record.playAudio(record.BlockingState.Blocking)
    basic.clearScreen()
})
```

Or if the `interval` second parameter is used in `basic.showString("?", 0)` to set the delay to 0ms, then it also works as expected.

So @JohnVidler suspicion that the first run doesn't respect the default 400 ms delay from `basic.showString("?")` is right.
But the delay on the second run is much longer, using a stopwatch it's about 2 seconds.

JohnVidler, 2023-10-12:
Do we have a CODAL repro for this? Is this a Makecode issue or a timing issue in CODAL?

microbit-carlos, 2023-10-12:
We haven't tried it in CODAL code directly, so that might be the first step.

microbit-carlos, 2024-03-11:
I can confirm that this issue does not reproduce anymore on the current Live MakeCode, so the delay issue is likely to have been on the MakeCode side and fixed at some point in the last few months.

closed 2024-03-11 by microbit-carlos