lancaster-university/codal-microbit-v2 #380: StreamRecording.cpp build error on latest codal-core repo

JasJassal, 2023-09-24:
Following the build instructions from microbit-v2-samples results in pulling down the codal-core repo headed at the following patch:
d7ea466 (HEAD) Exposd a call to dump the heap state when not in full memory debug mode

This errors with:
microbit-v2-samples/libraries/codal-core/source/streams/StreamRecording.cpp:80:9: error: 'codal_dmesg_flush' was not declared in this scope
         codal_dmesg_flush();

Rolling back to HEAD (ommit a116354) on codal-core seems to result is successful build.

JasJassal, 2023-09-24:
Realise now I probably should have raised this on microbit-v2-samples repo.

microbit-carlos, 2023-09-26:
Thanks for the report @JasJassal!
This should be fixed in the latest commits, could you clean your environment and try again (you can delete the `build` and `librarires` folder to download everything again and rebuild from zero).

JasJassal, 2023-09-26:
Verified builds using build.py without issue. Thanks, closing issue

JasJassal, 2023-09-26:
Verified builds using build.py. Thanks.

closed 2023-09-26 by JasJassal