lancaster-university/codal-microbit-v2 #392: Revert changes in codal-microbit-nrf5sdk that modify SDK files directly

microbit-carlos, 2023-11-10:
Creating the issue here as this is the place used to manage microbit related tickets.

The goal of the https://github.com/microbit-foundation/codal-microbit-nrf5sdk/ repository is to keep the SDK files in the `nRF5SDK` folder untouched, and to place any mods in `nRF5SDK_mods`, as there are scripts that would overwrite those changes when upgrading the SDK.

So we need to revert https://github.com/microbit-foundation/codal-microbit-nrf5sdk/commit/5714cbe338c3544793bbb841a7ae81708a3ffae1.  And if we are there already, we could also revert https://github.com/microbit-foundation/codal-microbit-nrf5sdk/commit/44400bf165d76f76aee570aa05d61417e05f5270, which the GitHub comments on the commit also suggested.

JohnVidler, 2024-03-11:
My commit (https://github.com/microbit-foundation/codal-microbit-nrf5sdk/commit/5714cbe338c3544793bbb841a7ae81708a3ffae1) has already been reverted, and I'll sort a mods patch for the second one for the next tag.

JohnVidler, 2024-04-15:
Dropping this for now, as we don't support the compiler versions that would currently emit these warnings (I'm just being weird/testing future releases by running newer ones)

closed 2024-04-15 by JohnVidler

microbit-carlos, 2024-04-15:
> Dropping this for now, as we don't support the compiler versions that would currently emit these warnings (I'm just being weird/testing future releases by running newer ones)

Not sure I understand, do you mean that CODAL does not support the latest GCC releases? Or why was the patch needed to begin with?

This ticket was also meant to cover reverting https://github.com/microbit-foundation/codal-microbit-nrf5sdk/commit/44400bf165d76f76aee570aa05d61417e05f5270, which is still present in the SDK repo.