lancaster-university/codal-microbit-v2 #397: Waking up turns on the microphone even if it is never used

microbit-carlos, 2023-11-27:
With current MakeCode Live using CODAL [v0.2.63](https://github.com/lancaster-university/codal-microbit-v2/releases/tag/v0.2.63).

When this programme wakes up the microphone LED stays permanently ON:

[image]
```ts
input.onButtonPressed(Button.A, function () {
    power.lowPowerRequest()
})
power.fullPowerOn(FullPowerSource.B)
basic.showIcon(IconNames.Heart)
```
[microbit-ad.hex.zip](https://github.com/lancaster-university/codal-microbit-v2/files/13473115/microbit-ad.hex.zip)

microbit-carlos, 2024-02-12:
@JohnVidler I believe this still need to be fixed, or have you pushed a fix to a non-main branch?

JohnVidler, 2024-03-11:
This is now fixed in `master`; and I can't reproduce this on the beta build for `v0.2.66`.

Calling this complete unless we see a regression.