lancaster-university/codal-microbit-v2 #401: Set audio pin creates a 44.2kHz 50% PWM after audio playback

fabianhugo, 2024-01-29:
**Describe the bug**
When using the `set audio pin` block in the `pins` category or the `uBit.audio.setPin(uBit.io.P0);` function in microbit-v2-samples and then playing any sounds, e.g. `uBit.audio.soundExpressions.play("giggle");` leaves a 44.2kHz PWM signal with a duty cycle of 50% on the previously set pin. 
[image]. 
This causes some speakers to heat up, and could therefore lead to broken speakers or other damage.

**To Reproduce (makecode)**
1. Create makecode project use the `set audio pin` block in the `pins` category, choose P0
2. Use the `play giggle unitl done` 
3. Transfer program
4. Probe pin P0

**To Reproduce (microbit-samples-v2)**
1. Insert the following code to `main.cpp`: 
`    uBit.audio.setPin(uBit.io.P0);
    uBit.audio.soundExpressions.play("giggle");`
2. Transfer program
3. Probe pin P0

jaustin, 2024-01-29:
Related https://github.com/microsoft/pxt-microbit/issues/3883 - you can use the `setSilenceLevel` API in C++ or Typescript.  @fabianhugo do you have any more info about the speakers that 'heat up' please?
CC @finneyj

fabianhugo, 2024-01-30:
Thanks a lot for your swift reply and for pointing towards the other issue. `music.setSilenceLevel(0)` solved it :)  
The speaker that heated up is the speaker from the dFRobots Boson Kit driven by a MOSFET. 
[image]
Can be closed. Thanks!