lancaster-university/codal-microbit-v2 #403: MicroBitButton.h missing in MicroBit.h when DEVICE_BLE=0

tballmsft, 2024-02-09:
When building pxt-microbit with DEVICE_BLE against this repo, we get a compile error as MicroBitButton is not defined.  I traced this down to the following issue in MicroBit.h in this repo:

> MicroBitButton.h is not included directly in MicroBit.h, but only indirectly by MicroBitButtonService.h, which is only included in MicroBit.h when DEVICE_BLE is enabled

finneyj, 2024-02-09:
Thanks @tballmsft!

Easy one - can we squeeze a fix for this into our planned next release @JohnVidler?

martinwork, 2024-02-09:
I suspect MicroBitButton.h was only included in the BLE code because it was in the V1 code, and it's really a mistake: MicroBitButtonService doesn't seem to use MicroBitButton and should be using DEVICE_ macros rather than MICROBIT_.

MicroBitButton is in the compat folder and only occurs here
https://github.com/microsoft/pxt-microbit/blob/2dccb52fcfe9edd6a420cfd5718c4e80452a03ad/libs/core/pins.cpp#L647

tballmsft, 2024-02-09:
It's not critical to fix this in lancaster, as we have a solution for pxt-microbit. In libs\core\pxtcore.h, I also see

```
#ifndef __PXTCORE_H
#define __PXTCORE_H

#include "MicroBit.h"
#include "MicroBitImage.h"
#include "ManagedString.h"
#include "ManagedType.h"
```

So clearly MicroBit.h wasn't designed to include everything we need in pxt-microbit, across V1 and V2, but to provide compat with V1.

tballmsft, 2024-02-09:
Nevertheless, it's odd to have MicroBitButton.h appear in MicroBit.h for one build and not for another.

microbit-carlos, 2024-02-13:
Okay, so it looks we just need to include `MicroBitButton.h` into `MicroBitCompat.h`, as `MicroBitCompat.h` is already included in `MicroBit.h`.

microbit-carlos, 2024-02-26:
Fixed in [v0.2.65](https://github.com/lancaster-university/codal-microbit-v2/releases/tag/v0.2.65) by including `MicroBitButton.h` in `MicroBitCompat.h`:  https://github.com/lancaster-university/codal-microbit-v2/commit/0c67e627f29eaaef1c7821710eebc75b73cd0441

closed 2024-02-26 by microbit-carlos