lancaster-university/codal-microbit-v2 #405: In a tight loop, mostly deep sleeping, takes 600 milliseconds to go back to sleep

microbit-carlos, 2024-02-12:
With a bare minimum programme like this one:

```cpp
#include "MicroBit.h"

MicroBit uBit;

int main() {
    uBit.init();

    while (true) {
        uBit.power.deepSleep(1000);
    }
}
```

[image]

It takes about 600 msec to fully go to deep sleep:
[image]

microbit-carlos, 2024-02-12:
Okay, if in the infinite loop the programme yields for a second we see a lower time to go to sleep:

```cpp
    while (true) {
        uBit.power.deepSleep(1000);
        uBit.sleep(1000);
    }
```

So top level view:
[image]

If we zoom in, we can see the first half (one second) with `uBit.sleep()` and the half (one second) with `uBit.power.deepSleep()`:
[image]
[image]

And if we zoom into at the 2nd half, when it is going into `uBit.power.deepSleep()`, we can see that it takes about 95 ms to go to sleep:
[image]

So it must be the combination of going to sleep and waking up in short succession that causes the 600 ms delay.

martinwork, 2024-07-28:
500ms is here
https://github.com/lancaster-university/codal-microbit-v2/blob/50ae9945ade9c02924e3e85986f704c3c14dd311/inc/MicroBitPowerManager.h#L164

AFAICR the main thought was that rapid sleep/wake cycles would use more power than staying awake. Maybe a better estimate could come from the measurements.