lancaster-university/codal-microbit-v2 #427: A StreamRecording with low splitter sampling rate fails to stop recording

microbit-carlos, 2024-05-01:
With this example code, setting the recording rate to 320 samples per second works, but setting it to 319 hangs when recording and never "stops".

```cpp
#include "MicroBit.h"
#include "StreamRecording.h"

MicroBit uBit;

const int SAMPLE_RATE = 319;
const int BUFFER_LEN = SAMPLE_RATE * 5;

int main() {
    uBit.init();

    static SplitterChannel *splitterChannel = uBit.audio.splitter->createChannel();
    splitterChannel->requestSampleRate( SAMPLE_RATE );
    static StreamRecording *recording = new StreamRecording(*splitterChannel, BUFFER_LEN);
    // Setting the mixer channel to a different sampling rate doesn't make a difference
    static MixerChannel *recordingChannel = uBit.audio.mixer.addChannel(*recording, SAMPLE_RATE);
    recordingChannel->setVolume(75.0);
    MicroBitAudio::requestActivation();

    while (true) {
        if (uBit.buttonA.isPressed()) {
            recording->erase();

            recording->recordAsync();
            while (recording->isRecording()) {
                uBit.display.print("R");
                uBit.sleep(5);
            }
            recording->stop();

            uBit.display.clear();
        } else if (uBit.buttonB.isPressed()) {
            recording->playAsync();
            while (recording->isPlaying()) {
                uBit.display.print("P");
                uBit.sleep(20);
            }
            uBit.display.clear();
        }
        uBit.sleep(200);
    }
}
```
- [319.hex.zip](https://github.com/lancaster-university/codal-microbit-v2/files/15175696/319.hex.zip)
- [320.hex.zip](https://github.com/lancaster-university/codal-microbit-v2/files/15175697/320.hex.zip)

To replicate:
- Compile this example or download 319.hex & flash it to the micro:bit
- Press button A to start recording
   - Recording should only last 5 seconds
   - The letter R should show on the display for as long as the device is recording
- No matter how long you wait, the recording never stop
   - There isn't a 020 panic (memory error) thrown either, so if it is indeed recording it must be doing it within the allocated buffer

martinwork, 2024-08-17:
Single byte buffers are quietly rejected, though the format has one byte per sample 
https://github.com/lancaster-university/codal-core/blob/df05db9e15499bd8906618192a4d482e3836c62f/source/streams/StreamRecording.cpp#L96

`data == ManagedBuffer() `doesn't seem to add anything.

To reject buffers without a whole sample, the test could be
`if( data.length() < DATASTREAM_FORMAT_BYTES_PER_SAMPLE( this->upStream.getFormat()) {
`

With this, the sample rate borderline for "records forever" moves to 142/143, as in https://github.com/lancaster-university/codal-microbit-v2/issues/428

With low sample rates, I think StreamRecording will build a list of tiny buffers, with a 12+ byte overhead for each node
https://github.com/lancaster-university/codal-core/blob/df05db9e15499bd8906618192a4d482e3836c62f/inc/streams/StreamRecording.h#L19

microbit-carlos, 2025-05-16:
But we can request a lower sampling rate from the microphone right? And in that case it's possible the bug is still present in that use case (which came up during MicroPython testing).

We can probably test this by removing `splitterChannel->requestSampleRate( SAMPLE_RATE );` in the example code and adding the `mic-> setSampleRate()` call after setting the pipeline.

closed 2025-05-16 by microbit-carlos