lancaster-university/codal-microbit-v2 #440: Investigate why CODAL v0.2.65 reduces the recording length in MakeCode

microbit-carlos, 2024-08-07:
Reported in:
- https://github.com/microsoft/pxt-microbit/issues/5796

martinwork, 2024-08-07:
@microbit-carlos The default sample rate was changed from 11000 to 44100.
https://github.com/lancaster-university/codal-microbit-v2/commit/2e664e6294749c1444c002d2276674c3210ba0b8

Recording time is restored by adding a `set sample rate` block.

microbit-carlos, 2024-08-08:
Yes, thanks Martin!
It was changed from a magic number to a config option in https://github.com/lancaster-university/codal-microbit-v2/commit/d54ce9b784ef1bba22c075b281bb877abf53ea18, so set to 11K, but that also affected the mixer channel default, so maybe that's why it was pushed up to top value 44.1K.

MakeCode was depending on the default value, so the recording extension needs to set splitter channel sample rate to the default used in the extension.

microbit-carlos, 2024-08-08:
PR raised in MakeCode:
- https://github.com/microsoft/pxt-microbit/pull/5805

microbit-carlos, 2024-08-12:
Fix merged in MakeCode, so we can close this as completed.

closed 2024-08-12 by microbit-carlos