lancaster-university/codal-microbit-v2 #441: ADC period 22us only supports 4 channels

martinwork, 2024-08-19:
More than 3 analogue inputs, reduces mic rate

Try this in MakeCode live, then beta
https://makecode.microbit.org/_a277qf3sTHYs
https://makecode.microbit.org/_4q5L1eCcWFK9 (for import to beta with the pin blocks fixed)
Press A to record – see recording time in console
Press B to add 4 analogue inputs
Press A to record – see increased recording time in beta

martinwork, 2024-08-19:
Similar C++ example: [recording.zip](https://github.com/user-attachments/files/16658772/recording.zip)

The hex in the zip has
`DMESG("possibleSamples %d", possibleSamples);
`

inserted at
https://github.com/lancaster-university/codal-nrf52/blob/8802eb49140e0389e535cb6160d9080efd951ba7/source/NRF52ADC.cpp#L559

martinwork, 2024-08-19:
The NRF52ADC period used to be set in MicroBit.cpp to 91us (note: 1e6/11000 == 90.9), but has been changed by MicroBitAudio to 45 (1e6/22000), then 90 (1e6/11000) and now 22 (1e6/44100).

The last change says "back up to 44100" but that number has not been used to set the NRF52ADC rate before, as far as I can see. Perhaps that referred to the Mixer2 output channel speed.

It looks like MicroBitAudio is turning the NRF52ADC speed dial up to 11 out of 10, even for programs that don't use the microphone. I don't think a MakeCode program can reduce the  NRF52ADC rate.

------------------------------------------------------------------------------------
History... Originally, the default NRF52ADC period was set to 91us
https://github.com/lancaster-university/codal-microbit-v2/blob/a8470a76d9f3420f1daa1ecaf3eda80732bb857c/model/MicroBit.cpp#L69

MicroBitAudio started setting the default ADC period to 1e6/22000 (45) in https://github.com/lancaster-university/codal-microbit-v2/commit/98d7dc960b0af9b5249fa47f4755b0ef9e260594

MicroBitAudio was changed to set 1e6/11000 (90) in https://github.com/lancaster-university/codal-microbit-v2/commit/ceba6d7ed038af894443012e95ba7e3bfee33221

Macro CONFIG_MIXER_DEFAULT_CHANNEL_SAMPLERATE == 11000 was added after issue https://github.com/lancaster-university/codal-microbit-v2/issues/311 in https://github.com/lancaster-university/codal-microbit-v2/commit/d54ce9b784ef1bba22c075b281bb877abf53ea18

Finally, CONFIG_MIXER_DEFAULT_CHANNEL_SAMPLERATE was increased to 44100 in https://github.com/lancaster-university/codal-microbit-v2/commit/2e664e6294749c1444c002d2276674c3210ba0b8

----------------------------------------------------------------------------------------
Trying to understand CONFIG_MIXER_DEFAULT_CHANNEL_SAMPLERATE... It is
- the default rate for a Mixer2 output channel 
- the default NRF52ADC sample rate

https://github.com/search?q=repo%3Alancaster-university%2Fcodal-microbit-v2%20CONFIG_MIXER_DEFAULT_CHANNEL_SAMPLERATE&type=code

These two things don't seem logically closely connected. 

I'm not sure why MicroBitAudio needs to set the NRF52ADC rate in its constructor.

Does LevelDetectorSPL need the same high rate as sound recording?
Won't the NRF52ADC rate get increased if a faster rate is requested?
Isn't it the input and output rates for a StreamRecording that need to match for https://github.com/lancaster-university/codal-microbit-v2/issues/311?
```
    SplitterChannel *splitterChannel = uBit.audio.splitter->createChannel();
    splitterChannel->requestSampleRate( SAMPLE_RATE );
    StreamRecording *recording = new StreamRecording(*splitterChannel, BUFFER_LEN);
```

Could the mic input pipeline be created on first use?
https://github.com/lancaster-university/codal-microbit-v2/blob/a8470a76d9f3420f1daa1ecaf3eda80732bb857c/source/MicroBitAudio.cpp#L74

martinwork, 2024-09-13:
Sound level is more noisy at 22us / 44100Hz.
When it’s quiet, MakeCode v6 sees lots of zeros, but v7 has a very jumpy graph.
https://makecode.microbit.org/_MDw9Tafqu9k3

<details><summary>C++</summary>

```
#include "MicroBit.h"

MicroBit uBit;

void forever()
{
    while (true)
    {
        int level = uBit.audio.levelSPL->getValue();

        uBit.serial.printf( "s:%d\r\n", (int) level);
        uBit.sleep(20);
    }
}

int main()
{
    uBit.init();
    //uBit.audio.activateMic();
    create_fiber( forever);
    release_fiber();
}
```

</details>

finneyj, 2025-02-20:
yep - default sample rate is back down to 11kHz in audio-refactor.... Hopefully that will address the issues here?

martinwork, 2025-04-28:
@finneyj Yes it seems OK now - tested audio refactor branch with
```
#include "MicroBit.h"

MicroBit uBit;

int decibels2makecode( int db)
{
    db = max( 52, min( db, 120));
    db = (db - 52) * 255 / ( 120 - 52);
    db = max( 0, min( db, 255));
    return db;
}

void forever()
{
    while (true)
    {
        uBit.display.scroll( ManagedString( (int) decibels2makecode( uBit.audio.levelSPL->getValue())));
        uBit.sleep(20);
    }
}

int main()
{
    uBit.init();
    create_fiber( forever);
    release_fiber();
}
```

closed 2025-05-16 by microbit-carlos