lancaster-university/codal-microbit-v2 #442: Gibberish in the beginning of the recording data

microbit-carlos, 2024-08-23:
When recording data from the microphone, the beginning of the data is clearly not right.

I've taken measurements with a recording and playback sampling rate of ~44.1 KHz, using the `StreamRecording` and retrieving the data with the `SerialStreamer`. During sound recording, I've placed my finger on top of the microphone hole to try to minimise background noise, and once the data stabilises, it stays very close to the zero line.

The top graph plots 11k samples, 250ms at 44.1Khz. The second graph zooms into the first 2560 samples:
[image]

The ADC DMA buffers are 512 bytes, and with a 14-bit sampling rate, we get 256 samples per buffer.
In most captures that I've done it looks like the data is only stable after 2560 samples, which is interestingly exactly 10 DMA buffers.

<details>
<summary>Source code to replicate</summary>

```cpp
#include "MicroBit.h"
#include "StreamRecording.h"
#include "SerialStreamer.h"

MicroBit uBit;

int main() {
    uBit.init();

    static SplitterChannel *splitterChannel = uBit.audio.splitter->createChannel();
    static StreamRecording *recording = new StreamRecording(*splitterChannel);

    static MixerChannel *mixerChannel = uBit.audio.mixer.addChannel(*recording);

    static SerialStreamer *streamer = new SerialStreamer(*recording, SERIAL_STREAM_MODE_DECIMAL);

    MicroBitAudio::requestActivation();
    mixerChannel->setVolume(75.0);
    uBit.audio.mixer.setVolume(1023);

    uBit.serial.printf("Recording rate: %d\n", (int)recording->getSampleRate());
    uBit.serial.printf("Mixer rate: %d\n", (int)uBit.audio.mixer.getSampleRate());

    while (true) {
        if (uBit.buttonA.isPressed()) {
            uBit.sleep(500);
            recording->recordAsync();
            uBit.display.print("R");
            uBit.sleep(250);
            recording->stop();
        } else if (uBit.buttonB.isPressed()) {
            uBit.display.print("P");
            recording->connect(*mixerChannel);
            recording->play();
        } else if (uBit.logo.isPressed()) {
            recording->connect(*streamer);
            recording->play();
        }
        uBit.display.clear();
        uBit.sleep(100);
    }
}

```

</details>

<details>
<summary>Serial output</summary>

Didn't include all the serial output at the end, stopped around the time the data stabilises.

```
Recording rate: 45454
Mixer rate: 44100
30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 
[... 164 lines cut ...]
-2 -2 0 0 0 -1 -1 0 0 -1 0 0 0 -1 0 0 
0 -1 -1 0 0 0 0 0 0 0 0 0 -1 -3 0 -2 
-1 -1 0 0 0 0 0 -1 0 0 1 0 -1 -1 0 0 
0 -1 -1 0 0 -1 0 -1 0 0 0 -1 -1 -1 0 0 
0 1 -1 -1 -1 0 0 -2 -1 0 0 0 -1 -1 -1 0 
0 0 0 0 0 1 0 0 0 0 -1 -1 0 0 -1 -1 
0 0 -1 0 -2 -2 0 -1 1 0 0 0 1 0 0 -1 
-1 0 -1 0 0 0 0 0 0 0 0 -1 -1 0 0 0 
-1 0 0 0 -1 0 -1 -2 0 0 0 0 0 0 0 0 
0 0 -1 -1 -2 0 -1 -1 0 0 0 0 0 -1 0 0 
0 0 0 0 0 0 -1 -1 -1 0 0 0 0 0 -1 0 
0 -1 0 -2 0 0 1 0 0 0 0 -1 -1 -2 -1 0 
0 -1 0 0 0 0 0 0 0 -1 0 0 0 0 0 0 
0 0 0 0 0 0 -1 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 2 -1 0 1 -1 0 0 1 0 
-1 0 1 0 0 1 0 0 0 0 0 0 0 0 1 0 
1 -1 0 0 0 0 0 0 0 0 0 0 0 0 0 1 
0 -1 0 0 1 0 0 0 -1 0 1 1 0 -1 0 0 
1 0 0 0 0 0 0 0 0 0 0 0 0 0 0 1 
0 1 1 0 0 1 0 0 0 -2 0 0 1 0 0 0 
```

</details>

[MICROBIT.hex.zip](https://github.com/user-attachments/files/16732107/MICROBIT.hex.zip)

martinwork, 2024-08-26:
I think the gibberish is produced in NRF52ADCChannel::demux. The first buffer in each recording is an old one stored by mic→output. The next few are recorded before, while or shortly after the mic is activated. What settle time does the mic need? The long "tail" of steps is perhaps the StreamNormalizer adjusting the calculated zero offset.

**Two possible workarounds without changing CODAL:**

- Keep the mic activated and buffers flowing through mic→output with levelSPL->activateForEvents(true)
- Better?... Immediately before a recording, use levelSPL->getValue() to activate the mic, allow the settle time plus zero adjustment, and stop it being deactivated before recording starts. levelSPL disables itself after 75ms.

**I think this is how MicroBitAudio mic input is operating...**

As the program starts, the mic is 
- created in the MicroBitAudio constructor
- activated when StreamRecording (or StreamNormalizer?) is connected.
- deactivated because no buffers are pulled

The mic ADC channel continues to generate buffers and drive the pipeline, which is blocked in mic→output, and only results in MicroBitAudio calling deactivateMic() twice for each buffer (see SPLITTER_TICK below).

No buffers pass through the StreamNormalizer until something pulls on the splitter.

When StreamRecording is started, it simply starts accepting buffers. Though the pull request is prompted by the ADC producing a new buffer, if the pipeline was not flowing, the first buffer pulled is whatever mic→output was holding when it stopped flowing. Pulling that unblocks mic→output, and causes the splitter and rawsplitter to message MicroBitAudio to activate the mic.

Once StreamRecording has ended, deactivateMic() is called repeatedly again because mic→output blocks when no buffers are pulled.

uBit.audio.levelSPL can activate the mic and provide a pull on rawsplitter to keep mic→output unblocked and the mic active.

The first block of the first recording after RESET is always slightly off, because it’s the first one through the StreamNormalizer, establishing the zero offset.

Even after uBit.audio.levelSPL→activateForEvents(true), at the end of a StreamRecording, MicroBitAudio sees events 1032, 4 and 10, all of which lead to deactivateMic(). So the mic light briefly flickers off before subsequent events activate it again, which seems to generate some gibberish.

**Details...**

I put a serial number in the first byte of each NRF52ADCChannel::demux buffer, and copied it in StreamNormalizer::pull(). That revealed that the first buffer in a recording was old. DMESGing the first few bytes of each buffer in those places seemed to confirm they haven’t been corrupted.

I don’t understand why the non-blocking mic→output DataStream holds the first buffer it receives until it is pulled, and drops newer buffers. I expected the opposite.
https://github.com/lancaster-university/codal-core/blob/509086cc8590465041b15493ab52b56e7071c110/source/streams/DataStream.cpp#L188

While the ADC is driving the pipeline, rawsplitter and splitter continually send SPLITTER_TICK messages that cause MicroBitAudio to call activateMic or deactivateMic twice for each ADC buffer.
https://github.com/lancaster-university/codal-core/blob/509086cc8590465041b15493ab52b56e7071c110/source/streams/StreamSplitter.cpp#L255
SPLITTER_TICK was added quite recently.
https://github.com/lancaster-university/codal-core/commit/c064d96883a04605ba8d0a3cd12428ccf021d5ad

finneyj, 2025-02-20:
I've added some settling time into the system - see audio-refactor branch. Is this issue still present?

microbit-carlos, 2025-03-31:
@finneyj I've tested this with the latest `audio-refactor` branches (manually pulled the latest commits in each library) today and still have a similar issue.

[MICROBIT-442.hex.zip](https://github.com/user-attachments/files/19537090/MICROBIT-442.hex.zip)
[image]

If it doesn't replicate when the hex is flashed tried pressing the reset button or unplugging and replugging the USB cable.

- codal-core 5c0d922e5cd85ff034d57c76e4f61dc0014b42b1
- codal-nrf52 ae11dcc9e3df4133155ad0fa3950408f6ab922f0
- codal-microbit-nrf5sdk 4b8abc690f6c9fca6132e6db5ee13a795a263f88
- codal-microbit-v2 6f47847abc6bc46fcbb727c9e8a9745cd55c5fe0

martinwork, 2025-03-31:
@finneyj The second part of this comment might be relevant. 
https://github.com/lancaster-university/codal-microbit-v2/pull/457#issuecomment-2677078062

I haven't retested after recent changes.

finneyj, 2025-05-16:
Would be good to know if this is resolved with latest updates @martinwork  @microbit-carlos

microbit-carlos, 2025-06-16:
@martinwork I assume this problem is still present when using the latest commits from all the main branches?
(`target.dev = true` can be set in `codal.json` to easily checkout the latest).

martinwork, 2025-06-17:
There is still something, yes.

<details>
<summary>streamed data</summary>

```
-8 -8 -5 -6 -6 -6 -6 -7 -5 -5 -5 -5 -5 -5 -4 -5
-5 -5 -5 -5 -5 -5 -4 -3 -4 -5 -5 -4 -4 -5 -5 -4
-4 -4 -3 -4 -4 -4 -3 -4 -5 -5 -5 -5 -3 -4 -3 -4
-4 -5 -5 -5 -5 -4 -4 -4 -3 -4 -3 -4 -4 -3 -3 -4
-4 -4 -4 -4 -3 -3 -3 -3 -4 -4 -4 -4 -3 -4 -4 -5
-4 -3 -3 -3 -3 -4 -4 -3 -3 -4 -3 -4 -3 -3 -5 -4
-3 -5 -4 -3 -3 -3 -3 -3 -3 -3 -3 -3 -3 -4 -3 -4
-3 -3 -4 -3 -3 -4 -3 -3 -4 -3 -3 -2 -2 -3 -3 -3
-3 -3 -2 -4 -3 -3 -2 -3 -3 -2 -3 -2 -3 -3 -3 -2
-3 -2 -2 -2 -2 -2 -2 -3 -2 -2 -4 -2 -3 -3 -2 -2
-2 -2 -2 -2 -3 -3 -3 -3 -3 -3 -3 -3 -3 -3 -3 -3
-3 -3 -3 -3 -2 -2 -3 -4 -4 -2 -3 -3 -3 -3 -3 -3
-3 -3 -2 -2 -2 -2 -2 -3 -2 -2 -2 -2 -2 -2 -2 -2
-2 -2 -3 -3 -2 -3 -2 -1 -2 -1 -2 -2 -3 -2 -3 -3
-2 -2 -2 -3 -4 -2 -2 -3 -2 -3 -2 -2 -2 -2 -2 -2
-1 -2 -2 -2 -3 -2 -2 -2 -2 -1 -2 -2 -2 -2 -2 -3
0 -1 -2 0 -1 -1 0 0 0 0 -2 0 0 0 0 -1
-1 0 0 0 0 0 -1 0 0 0 0 0 0 0 0 0
0 -1 0 0 0 0 0 0 0 0 0 0 0 0 0 0
-1 0 0 0 0 0 0 0 -1 0 0 0 0 -1 0 0
0 0 0 -1 0 0 0 0 0 0 0 0 0 0 0 0
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
0 0 0 0 0 -1 0 0 0 0 0 0 0 1 1 0
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
0 0 0 0 0 0 0 -1 0 0 0 0 0 0 0 0
0 0 0 0 0 0 -1 0 1 0 0 0 0 0 0 0
0 0 0 0 0 0 0 0 0 -1 0 0 0 0 0 0
0 0 0 0 -1 0 1 1 0 0 0 0 0 0 0 0
0 0 0 0 0 1 0 0 0 0 0 0 0 0 0 0
0 1 0 0 0 1 0 0 0 0 1 1 0 0 0 0

```

And with analogue channels

```
-1 0 0 0 0 1 1 1 0 1 -1 1 0 0 1 0
-1 0 0 0 0 0 0 0 1 -1 0 0 0 0 0 0
0 -1 -1 10 9 8 9 9 10 10 10 10 10 9 10 10
11 9 10 7 10 9 9 10 11 11 9 10 10 11 10 11
8 9 10 40 40 40 40 40 40 40 40 40 40 34 -2 -55
-118 79 25 -26 -71 -112 107 71 40 8 -20 -45 -68 -88 -110 -127
114 102 88 75 62 54 43 38 30 23 18 13 8 4 0 -2
-5 -8 -10 -15 -16 -17 -17 -18 -19 -20 -22 -21 -23 -22 -23 -24
-27 -22 -25 -24 -27 -24 -23 -23 -21 -21 -22 -24 -23 -21 -21 -20
-22 -23 -23 -22 -23 -21 -20 -20 -18 -18 -17 -19 -18 -18 -18 -20
-18 -17 -20 -19 -18 -21 -16 -16 -16 -16 -14 -15 -15 -16 -13 -14
-13 -14 -14 -14 -16 -16 -14 -14 -17 -12 -12 -13 -14 -14 -13 -12
-13 -12 -13 -15 -14 -15 -12 -13 -11 -12 -13 -11 -12 -12 -12 -11
-10 -11 -9 -12 -12 -10 -10 -9 -9 -8 -8 -9 -10 -8 -9 -9
-10 -9 -6 -8 -9 -7 -9 -8 -7 -8 -6 -7 -7 -7 -8 -7
-8 -8 -9 -9 -10 -10 -7 -9 -7 -8 -4 -6 -8 -6 -6 -8
-2 -4 -4 -6 -6 -4 -5 -6 -4 -4 -3 -1 -2 -1 -2 -2
-1 -1 -1 -2 -2 -2 -2 -2 -4 -3 -3 -2 -2 -2 -2 -2
-1 -1 -1 -3 -1 -2 0 -1 -1 0 0 0 0 -1 0 -1
-2 0 0 -2 0 -2 -1 -1 -1 -3 -3 0 -2 -3 -2 -2
-3 -3 -1 -3 -4 -4 -2 -3 -1 -1 -3 -4 0 -2 0 -1
-1 -2 -1 -2 -4 -2 -2 -3 0 -2 -1 0 0 -1 -1 0
0 0 0 -1 0 0 -1 0 -2 -1 0 -2 -1 -3 -2 0
0 -1 0 -1 -1 -1 -2 -1 -2 0 -1 0 0 0 0 0
0 0 0 -2 -1 -3 -1 1 0 1 0 0 0 0 0 -1
0 0 1 0 -2 0 0 0 0 0 0 0 0 0 -1 0
0 0 1 -1 2 0 0 0 0 0 -1 0 0 0 0 1
-2 0 0 -1 -2 0 0 0 -1 0 0 -1 -2 0 0 0
0 0 1 0 1 0 1 1 0 1 0 4 1 1 1 2
2 2 0 -1 1 0 0 -1 1 0 0 1 3 0 0 0
1 1 0 0 0 1 -1 1 1 -1 0 1 2 2 1 0
1 0 0 0 0 0 0 0 1 2 2 2 1 0 1 -15
2 1 1 0 0 0 0 -2 0 -1 0 0 2 0 0 0
1 1 1 0 0 0 0 0 1 2 0 1 2 0 1 0
1 2 0 0 0 0 2 1 1 0 2 1 0 1 2 1
1 0 1 1 1 0 1 1 1 2 1 1 2 0 2 0
0 2 1 2 0 0 0 0 2 2 1 2 0 2 0 0
0 1 0 0 0 0 2 1 2 0 0 0 0 0 1 1
0 1 1 0 1 0 2 1 2 2 0 3 2 1 2 2
0 2 1 2 2 0 1 2 3 2 1 1 0 2 3 0
1 2 3 0 0 0 1 1 3 3 2 2 3 1 2 0
2 1 1 2 2 1 0 2 1 2 3 1 3 2 3 2
1 2 2 0 1 1 0 1 2 1 1 2 2 2 1 2
0 3 3 0 1 1 2 3 1 2 1 1 1 1 1 1
2 2 2 1 0 0 2 2 2 3 2 2 3 2 2 2
1 2 2 2 0 1 2 1 3 2 2 2 3 4 2 4
3 1 3 2 4 1 1 3 2 1 3 2 2 4 4 2
2 2 1 4 1 2 2 2 1 2 2 3 2 1 3 0
1 2 0 3 1 1 1 2 3 2 1 3 3 1 2 2
2 2 1 0 0 0 1 1 2 0 1 1 2 1 1 2
0 1 0 1 1 1 0 1 0 1 1 1 0 0 2 2
0 3 3 0 0 2 -1 0 3 0 0 2 2 3 1 4
4 3 3 1 2 2 2 3 3 3 4 2 4 3 1 4
1 1 2 2 3 1 3 2 0 0 1 0 0 2 1 2
0 1 0 0 -2 0 2 2 0 2 0 4 1 1 1 1
3 2 1 2 0 0 3 1 1 0 1 0 1 0 1 2
2 2 0 0 1 0 2 2 1 3 1 2 2 2 1 1
1 1 1 1 0 1 1 1 3 3 3 3 4 2 3 2
2 2 2 2 0 1 3 2 3 2 2 1 -1 3 1 3
2 1 1 2 0 0 2 0 2 2 2 2 2 1 3 2
3 3 3 2 3 1 3 2 1 2 3 2 2 0 2 3
4 3 2 3 0 0 0 0 0 0 1 1 2 2 1 0
2 2 2 1 0 1 3 1 3 0 2 1 0 2 2 3
1 1 1 2 1 1 3 2 2 3 3 2 2 1 2 2
0 1 0 -1 0 0 0 0 2 0 0 0 3 0 2 1
1 2 0 1 1 1 1 0 0 0 1 0 0 0 0 1
1 1 0 0 0 -1 0 1 -2 0 1 1 0 1 1 0
1 0 0 1 -1 0 0 1 1 -1 0 1 0 2 0 0
0 -1 0 -1 0 0 1 0 0 1 1 1 0 1 1 0
0 1 1 1 0 0 0 1 0 0 2 2 0 2 1 1
-18 -1 1 0 -1 -2 0 0 0 1 0 0 2 1 0 0
1 1 0 0 -2 0 0 0 0 1 0 2 0 0 1 2
1 1 0 0 0 0 1 0 0 1 1 1 1 0 1 1
0 1 2 2 1 3 0 3 2 1 2 1 0 3 2 0
0 0 0 0 0 -2 1 1 0 1 1 1 1 1 2 0
0 1 1 2 2 1 3 4 2 2 1 2 1 1 2 4
4 3 2 1 2 2 0 1 2 0 3 2 2 1 0 2
1 2 2 1 2 0 1 1 0 2 0 1 2 0 3 0
1 0 3 0 0 0 0 -2 0 0 0 1 0 1 2 1
0 1 0 0 0 1 0 3 2 2 1 0 1 1 0 3
1 1 1 1 0 2 0 0 0 0 0 0 0 1 1 1
2 0 1 0 0 -1 0 0 0 1 1 1 1 0 1 0
0 1 1 0 0 0 0 0 0 0 0 0 0 1 0 2
2 2 2 1 1 2 0 1 2 1 0 1 1 0 1 0
0 1 0 -1 0 0 1 0 0 0 1 0 1 0 2 0
0 1 1 -1 2 2 0 0 1 0 1 2 1 2 0 1
1 0 2 0 -2 0 3 1 2 0 1 1 2 2 0 0
0 2 1 1 0 0 0 2 0 1 -1 2 1 1 0 2
0 1 2 0 2 1 1 1 -1 0 0 0 1 1 2 2
0 1 0 1 2 3 2 2 1 0 2 0 -1 0 2 0
0 1 2 2 0 2 3 2 1 1 1 2 2 0 0 1

```

</details>

martinwork, 2025-06-18:
@microbit-carlos Ah yes, those graphs do illustrate the problem. I think both show the normaliser adjusting in steps from pre-settle data to find the "quiet" level. The second one has pre-settle data at the beginning, because 4 times as many buffers are generated before the mic's settle time has passed.