lancaster-university/codal-microbit-v2 #446: Missing sound data between the first and second data buffer from the microphone

microbit-carlos, 2024-09-10:
Using a custom `DataSink`, as done by MicroPython, we can see that when pulling from the splitter channel, we end up with what it looks like missing data between the first and second buffer from the ADC.

This test was carried out with the current CODAL configuration where the ADC is set to 45454 Hz, and the example code then sets the splitter channel sample rate to 7812.
As the ADC DMA buffers are 512 bytes, and we have 14-bits per sample, we get 256 samples per buffer. Downsampling that buffer from 45454 to 7812, and we get 44 bytes left (`256 / (45454 / 7812) ≈ 44`).

This example code is configured use a custom `DataSink` to pull 128 bytes 10 times in a row (each time connecting and disconnecting from the pipeline) and the data can be printed to serial.

Then I've played a 382 Hz sine wave (just picked a random frequency that easily illustrates this issue) with my phone next to the micro:bit microphone and plotted the data. We can see that in all of the plots the wave glitches at the 45th byte.
Interestingly we don't see the same at the 89th byte, so I wonder if this only affects the boundary between the 1st and 2nd byte from the moment the splitter channel is connected?

<details>
<summary>Source code</summary>

Press button A to record, and button B to print data to serial.

```cpp
#include "MicroBit.h"
#include "StreamRecording.h"

MicroBit uBit;

const int SAMPLE_RATE = 7812;
const int BUFFERS_LEN = 10;
const int BUFFERS_SIZE = 128;
uint8_t buffers[BUFFERS_LEN][BUFFERS_SIZE];

class MyStreamRecording : public DataSink {
public:
    SplitterChannel *upStream;
    uint8_t *dest;
    int dest_pos_ptr;
    size_t dest_max;
    bool request_stop;

    MyStreamRecording(SplitterChannel *source) : upStream(source) { }
    virtual ~MyStreamRecording() { }

    float getSampleRate() {
        return this->upStream->getSampleRate();
    }
    bool isRecording() {
        return upStream->isConnected();
    }
    void stopRecording() {
        this->request_stop = true;
    }
    void recordAsync(uint8_t *buf, size_t max_len) {
        this->dest = buf;
        this->dest_max = max_len;
        this->dest_pos_ptr = 0;
        this->request_stop = false;

        upStream->connect(*this);
    }
    virtual int pullRequest() {
        uint8_t *pull_buf = this->dest + this->dest_pos_ptr;
        size_t n = this->dest_max - this->dest_pos_ptr;

        if (n > 0) {
            n = this->upStream->pullInto(pull_buf, n) - pull_buf;
        }

        if (n == 0 || this->request_stop) {
            this->upStream->disconnect();
            this->request_stop = false;
        } else {
            // Convert signed 8-bit to unsigned 8-bit data.
            for (size_t i = 0; i < n; ++i) {
                pull_buf[i] += 128;
            }
            this->dest_pos_ptr += n;
        }

        return DEVICE_OK;
    }
};

int main() {
    uBit.init();

    SplitterChannel *splitterChannel = uBit.audio.splitter->createChannel();
    splitterChannel->requestSampleRate(SAMPLE_RATE);
    MyStreamRecording *recording = new MyStreamRecording(splitterChannel);

    memset(buffers, 128, sizeof(buffers));

    while (true) {
        if (uBit.buttonA.isPressed()) {
            // Record data into buffers
            uBit.display.print("R");
            uBit.sleep(500);
            for (int i = 0; i < BUFFERS_LEN; ++i) {
                recording->recordAsync(buffers[i], BUFFERS_SIZE);
                while (recording->isRecording()) {
                    uBit.sleep(1);
                }
            }
        }
        if (uBit.buttonB.isPressed()) {
            // Print buffers to serial in CSV format
            uBit.display.print("P");
            for (int i = 0; i < BUFFERS_SIZE; ++i) {
                for (int j = 0; j < BUFFERS_LEN - 1; ++j) {
                    uBit.serial.printf("%d, ", buffers[j][i]);
                }
                uBit.serial.printf("%d\n", buffers[BUFFERS_LEN - 1][i]);
            }
        }
        uBit.display.clear();
        uBit.sleep(100);
    }
}

```

</details>

(I've omitted the first two buffers, as they contain bad data as caused by issue https://github.com/lancaster-university/codal-microbit-v2/issues/442)

[image]
[image]
[image]

microbit-carlos, 2024-09-11:
To see if we could capture more than one glitch per "pipeline stream" (as the example code connects and disconnects the stream for each of the 128 byte plots), I've tried increasing each block of pulled data from 128 bytes to 256 bytes, and a few different sound frequencies (in case small offsets made it more obvious in the plot).

But, it looks like we really are only seeing the issue between the 1st and 2nd pull, at byte 44:
[image]

martinwork, 2024-09-11:
@microbit-carlos The first packet in each recording is an old packet, and the latest packet is dropped.

Based on a previous look, I think... When the StreamRecorder connects, the next generated ADC packet will prompt the pull request, but the  buffer pulled from  mic->output will be whatever it was holding before the packet generation, and the buffer the ADC just sent will be dropped.

The mic->output DataStream holds each buffer it receives until it is pulled, and drops newer packets. I don’t understand why it does this. I expected the opposite. I think perhaps it needs a double buffer system, like the ADC hardware uses, with the event handler switching the buffer for the ADC interrupt to use before it forwards the last buffer received.
https://github.com/lancaster-university/codal-core/blob/509086cc8590465041b15493ab52b56e7071c110/source/streams/DataStream.cpp#L188

By adding a serial number in the first byte of each ADC packet, and preserving it in StreamNormalizer, the packet numbers in each recording can be dumped. See below, two A presses.

It's clear that a packet is dropped in each recording, and the second batch starts with the old packet 39 that was not pulled through in the last batch.

For the output below, I made the serial number a static. With the serial number as a member variable in NRF52ADCChannel, the numbers restart for the second batch (the first recording is 39 4 5), because the channel is released and recreated during the pause between A presses.

```
0 1 2
3 5 6
7 9 10
11 13 14
15 17 18
19 21 22
23 25 26
27 29 30
31 33 34
35 37 38

39 183 184
185 187 188
189 191 192
193 195 196
197 199 200
201 203 204
205 207 208
209 211 212
213 215 216
217 219 220
```

finneyj, 2025-02-20:
I've updated a lot of behaviour in the audio-refactor branch. Can someone please re-test this against the code in there to see if this issue is still present?

martinwork, 2025-02-21:
It's looking good @finneyj ! Buffers now arrive in StreamRecording as they are generated by the ADC.

The log below shows two recordings started with a button press...  The first one looks like there might be a start-up buffer dropped. The second starts with the very next ADC buffer after the one that ended the first.

I need to test more: track when the mic is turned off/on; try with another ADC channel being logged, look at the buffer contents etc.

The recording seemed noisy when played back compared to current MakeCode - need to retest that too!

I had to change my sample to remove `splitterChannel->requestSampleRate(11000);`

```
demux 0
demux 1
demux 2
Recording rate: 11111
Mixer rate: 44100
demux 3
demux 4
StreamRecording buffer[0] 4
demux 5
StreamRecording buffer[0] 5

etc.

demux 202
StreamRecording buffer[0] 202
demux 203
StreamRecording buffer[0] 203
demux 204
StreamRecording buffer[0] 204
Recording length:   51200 bytes
Recording rate:     11111 Hz
Recording duration: 4608 ms

second recording

demux 205
StreamRecording buffer[0] 205
demux 206
StreamRecording buffer[0] 206
demux 207
StreamRecording buffer[0] 207

etc
```

microbit-carlos, 2025-03-31:
Thanks @martinwork! Are you able to re-test that and provide results now that Joe is working through tickets in the audio pipeline?

martinwork, 2025-04-08:
@microbit-carlos I did some more testing. It seems to be working as above, and as I commented on the PR
https://github.com/lancaster-university/codal-microbit-v2/pull/457#issuecomment-2677078062 

And https://github.com/lancaster-university/codal-microbit-v2/pull/457#issuecomment-2677080946

The startup delay is still one
https://github.com/lancaster-university/codal-microbit-v2/blob/audio-refactor/source/MicroBitAudio.cpp#L116

I'll test some more.

finneyj, 2025-04-08:
Thanks @martinwork 

Can I just check you're working on the up to date audio-refactor branches @martinwork? (there are branches of codal-core, codal-nrf52 and codal-microbit-v2) They're quite far ahead of master now, and I'm hoping to merge them once we've cleared up these issues....

martinwork, 2025-04-08:
Yes @finneyj , I think so. I have created branches from master in each of my forks, and merged the audio-refactor branches

https://github.com/lancaster-university/codal-microbit-v2/compare/audio-refactor...martinwork:codal-microbit-v2:mw-test-audio-refactor?expand=1

https://github.com/lancaster-university/codal-nrf52/compare/audio-refactor...martinwork:mw-test-audio-refactor?expand=1

https://github.com/lancaster-university/codal-core/compare/audio-refactor...martinwork:mw-test-audio-refactor?expand=1

finneyj, 2025-04-08:
OK - cool.

So reading your issue here: https://github.com/lancaster-university/codal-microbit-v2/pull/457#issuecomment-2677078062

There seem to be a whole array of problems you've detected. Are you saying these are all still present in the most recent codebase?

martinwork, 2025-04-08:
> Are you saying these are all still present in the most recent codebase?

I think so, but I was mainly testing this issue. I'll test some more and try to make individual issues

Here's my test sample:
https://github.com/martinwork/microbit-v2-samples/blob/mw-test-audio-refactor/source/test-audio-refactor.cpp

I no longer see "missing" data between first and second buffers. There are no old buffers in the recording. But there is warm-up noise.

RECORD
demux 8
demux 9
demux 10
demux 11
demux 12
Recording length:   1280 bytes
Recording rate:     11111 Hz
Recording duration: 115 ms
RECORD done
STREAM
8 -125 119 106 94 83 73 64 57 46 44 38 32 28 22 19 15 12 9 6 3 1 0 -1 -3 -5 -5 -7 -8 -10 -12 -13
-12 -11 -11 -12 -11 -12 -12 -13 -13 -13 -12 -12 -13 -12 -13 -14 -12 -14 -14 -13 -13 -12 -13 -13 -13 -12 -11 -12 -11 -12 -11 -11
-12 -12 -11 -11 -10 -10 -11 -10 -10 -10 -10 -10 -10 -10 -10 -9 -10 -10 -9 -9 -11 -9 -9 -9 -8 -8 -9 -8 -8 -9 -8 -7
-8 -8 -8 -7 -7 -7 -7 -7 -6 -6 -7 -7 -6 -6 -6 -7 -7 -6 -6 -6 -6 -6 -7 -6 -6 -11 -8 -6 -5 -7 -5 -6
-6 -6 -5 -5 -5 -5 -5 -5 -6 -6 -5 -5 -6 -5 -5 -5 -5 -5 -4 -4 -4 -3 -5 -4 -3 -4 -4 -4 -4 -5 -5 -6
-6 -5 -4 -4 -3 -4 -5 -4 -4 -4 -4 -4 -4 -3 -4 -4 -5 -3 -4 -4 -4 -3 -3 -5 -5 -4 -3 -4 -3 -4 -4 -2
-3 -3 -4 -4 -4 -2 -3 -3 -3 -4 -3 -2 -3 -3 -3 -2 -3 -3 -3 -3 -3 -2 -3 -3 -2 -2 -2 -2 -3 -3 -4 -5
-2 -3 -3 -3 -3 -3 -2 -2 -2 -3 -2 -3 -2 -2 -1 -3 -2 -1 -2 -2 -2 -2 -2 -3 -2 -2 -2 -2 -1 -2 -1 -2

9 -2 -2 -2 -1 -2 -3 -2 -2 -1 -2 -3 -2 -1 -1 -2 -2 -2 -2 -1 0 -1 0 0 -2 -1 0 0 0 -1 -1 -1
-1 -1 -2 -2 -1 0 0 0 0 0 -1 0 0 0 0 -2 0 0 0 -1 0 -1 0 0 -1 -1 -1 0 -2 -1 -2 0
0 0 -1 0 -1 -2 -1 -1 0 -1 0 0 -1 -1 0 -1 -1 -2 0 -1 0 -1 0 0 -1 -1 0 -1 -2 0 0 -1
-1 0 0 -1 0 0 0 -1 -1 -1 -1 -1 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 -1 -2 0
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 -1 0 0 0 0 0 0 0 0 0 0
-1 0 0 0 0 0 0 0 0 0 0 -1 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
0 -1 -1 0 0 0 0 0 -1 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
0 1 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 1 0 0 0 0 0 0 0 0 0 0 0 0 0

finneyj, 2025-04-08:
"I'll test some more and try to make individual issues"  - thanks @martinwork - that would be really helpful!!

I agree - the noise you're seeing above looks like the adc stabilising. Increasing the startup delay variable should address that. 

I guess there's a tradeoff here. Fixing the cold start noise that occurs once per hard reboot, against the additional latency introduced into every reactivation of a "warm" pipeline... The latter could be a significant effect? Perhaps we need a more nuanced solution that can detect if the device has just be cold booted or not?

finneyj, 2025-05-15:
hi @martinwork.

I found a stale buffer lurking around in the DataStream component, that could have been contributing to this issue. I fixed it in this patch: https://github.com/lancaster-university/codal-core/commit/8f816685127189c64276b672e92b9545542a4aa2

martinwork, 2025-05-19:
@finneyj @microbit-carlos I no longer see this issue of "missing" data, which was actually due to an old buffer being used, so I think we can close.

microbit-carlos, 2025-05-21:
Awesome, thanks both! 🎉

closed 2025-05-21 by microbit-carlos