lancaster-university/codal-microbit-v2 #449: The value of LEVEL_DETECTOR_SPL_8BIT_000_POINT (32) will never be reached

microbit-carlos, 2024-09-17:
Because the `LevelDetectorSPL` minVal is set to 52:
https://github.com/lancaster-university/codal-microbit-v2/blob/v0.2.68/source/MicroBitAudio.cpp#L82

And `LevelDetectorSPL` will clamp as that being the min. https://github.com/lancaster-university/codal-core/blob/509086cc8590465041b15493ab52b56e7071c110/source/streams/LevelDetectorSPL.cpp#L142-L147

microbit-carlos, 2024-09-17:
Previous discussion:
- https://github.com/lancaster-university/codal-microbit-v2/issues/236

When `LEVEL_DETECTOR_SPL_8BIT_000_POINT` was introduced it was set to 35.0, but the default value for minVal was already set to 52:
- https://github.com/lancaster-university/codal-core/commit/d376f1a851457effaf34058c62bd6c42c5ada6a0
- https://github.com/lancaster-university/codal-core/blob/d376f1a851457effaf34058c62bd6c42c5ada6a0/inc/streams/LevelDetectorSPL.h#L97-L100

However, at that time, the constructor in codal-microbit-v2 was setting minVal to 0, it was only changed to 52 a year later:
https://github.com/lancaster-university/codal-microbit-v2/commit/d6a777e34775727e35af4acb65bdd76a43b4d070

finneyj, 2025-02-20:
FYI - The audio-refactor branch now holds an implementation that goes back to 35db.

closed 2025-03-25 by microbit-carlos