lancaster-university/codal-microbit-v2 #463: Can't play 2 samples at the same time on 2 channels of a mixer

RThom6, 2025-01-24:
```C++
#include "Microbit.h"
#include "Tests.h"

MicroBit uBit;

static MemorySource *sampleSource = NULL;
static MemorySource *sample2 = NULL;
static MixerChannel *channel = NULL;
static MixerChannel *channel2 = NULL;

int main()
{
    uBit.init();

    uBit.audio.enable();

    if (sampleSource == NULL) {
        sampleSource = new MemorySource();
        sampleSource->setFormat(DATASTREAM_FORMAT_8BIT_UNSIGNED);
        sample2 = new MemorySource();
        sample2->setFormat(DATASTREAM_FORMAT_8BIT_UNSIGNED);
    }

    channel2 = uBit.audio.mixer.addChannel(*sample2);
    channel = uBit.audio.mixer.addChannel(*sampleSource);
    channel->setSampleRate(16000);
    channel2->setSampleRate(11000);
    ManagedBuffer b((uint8_t *)hello, sizeof(hello)); // hello sample present in lancaster-university/microbit-v2-samples
   
    sampleSource->playAsync(b);
    sample2->playAsync(b);
    uBit.sleep(5000);
}

```

`playAsync(ManagedBuffer b)` in codal-core MemorySource.cpp just plays them one after the other. Upon changing implementation to match `playAsync(const void *data, int length, int count)` now the audio doesn't really play at all. Debugging with DMESG for min and max values of the output of each buffer show its very compressed.

[image]

martinwork, 2025-01-24:
It looks like there's a bug that the ManagedBuffer override of playAsync calls play rather than playAsync
https://github.com/lancaster-university/codal-core/blob/master/source/streams/MemorySource.cpp#L160

If the data is in buffers, you can workaround that like
`sampleSource->playAsync( buffer.getBytes(), buffer.length());
`
or
`sampleSource->playAsync( &buffer[0], buffer.length());
`

Comparing with the sample, 
https://github.com/lancaster-university/microbit-v2-samples/blob/master/source/samples/SpeakerTest.cpp#L937
I think the addChanel calls need to set the sampleRange.
`addChannel(*sampleSource, 16000, 255);
`

The sample also calls setBufferSize, but that doesn't seem to be essential.

martinwork, 2025-01-24:
PR: https://github.com/lancaster-university/codal-core/pull/180

closed 2025-02-20 by finneyj