lancaster-university/codal-microbit-v2 #468: audio refactor: Programme freezes after reset

microbit-carlos, 2025-03-13:
@finneyj, with the current audio pipeline this programme only runs after a cold power up.
Pressing the reset button, or reflashing the same hex file and the programme gets stuck (but not frozen, as the display shows the character `A` correctly) in the first call to `uBit.audio.levelSPL->getValue()`.

[MICROBIT.hex.zip](https://github.com/user-attachments/files/19235221/MICROBIT.hex.zip)
```cpp
#include "MicroBit.h"

MicroBit uBit;

int main() {
    uBit.init();

    uBit.display.print("A");
    uBit.serial.printf("\nstart\n");
    uBit.audio.enable();
    while (true) {
        uBit.serial.printf("%d\n", (int)uBit.audio.levelSPL->getValue());
        uBit.sleep(20);
    }
}
```
```json
{
    "target": {
        "name": "codal-microbit-v2",
        "url": "https://github.com/lancaster-university/codal-microbit-v2",
        "branch": "audio-refactor",
        "type": "git"
    },
    "config":{
        "MICROBIT_BLE_ENABLED" : 0,
        "MICROBIT_BLE_PAIRING_MODE": 0
    }
}
```
With codal-microbit-v2 at commit [4a5aab37f97c7c5048cb7e317f461391da629cb1](https://github.com/lancaster-university/codal-microbit-v2/commit/4a5aab37f97c7c5048cb7e317f461391da629cb1).

microbit-carlos, 2025-03-13:
It looks like the resource lock is waiting there forever, as commenting it out "fixes" it.
https://github.com/lancaster-university/codal-core/blob/audio-refactor/source/streams/LevelDetectorSPL.cpp#L263

microbit-carlos, 2025-03-24:
Hey Joe!
Have you tried to replicate with the hex file I attached as well? As that could discard any hardware difference.

finneyj, 2025-03-25:
Yep - the binary definitely misbehaves, but not a version I build locally from source. The commit hash of codal-microbit-v2 matches your definition above too.

Can you tell if you have up to date branches of the relevant libraries? You'll need to be on the audio-refactor branch of codal-core and codal-nrf52 also... This is what I have:

codal-core:  5f1d1751391b1abef533c7ed16945ba036dab849 (HEAD -> audio-refactor, origin/audio-refactor)
codal-microbit-nrf5sdk: d41d5c7ebe53a1d01935e61d4ffa891e5112e119 (HEAD, origin/master, origin/HEAD, master)
codal-nrf52: ae11dcc9e3df4133155ad0fa3950408f6ab922f0 (HEAD -> audio-refactor, origin/audio-refactor)
codal-microbit-v2: 4a5aab37f97c7c5048cb7e317f461391da629cb1 (HEAD -> audio-refactor, origin/audio-refactor)

microbit-carlos, 2025-03-25:
Thanks Joe! At least we know the hex file replicates the issue, so it's likely hardware-independent (and also excludes DAPLink, which I often mess with and could have dubious builds running).

I've first tried replicating the build with this docker image, to standarise toolchains:
```bash
rm -rf build/
docker run -v $(pwd):/home --rm ghcr.io/carlosperate/microbit-toolchain:latest python build.py
```

And that replicates as well: [MICROBIT-docker.hex.zip](https://github.com/user-attachments/files/19446175/MICROBIT-docker.hex.zip)

These are the commits (all clean) I have checkout, which were the git clones automatically created by the CODAL build system using the codal.json (pointing to the codal-microbit-v2 `audio-refactor` branch) listed in my original message:
- codal-microbit-v2 4a5aab37f97c7c5048cb7e317f461391da629cb1
- codal-core 5f1d1751391b1abef533c7ed16945ba036dab849
- codal-nrf52 ae11dcc9e3df4133155ad0fa3950408f6ab922f0
- codal-microbit-nrf5sdk d41d5c7ebe53a1d01935e61d4ffa891e5112e119

Those match to yours, so maybe it is a toolchain difference? If you can replicate with the docker image as well that would be an interesting find.

finneyj, 2025-03-25:
Thanks @microbit-carlos. 

yes - Sounds like it's related to something in the toolchain.... I'll keep looking.

finneyj, 2025-03-27:
OK - found it. Actually a simple race condition between two code paths in the driver.., pullRequest() was responsible for awakening any waiting fibers, but periodicTimeout() was responsible for shutting down the component when a timeout occurs. 

This patch updates the latter take into account any fibers that may be waiting.

microbit-carlos, 2025-03-28:
Thanks Joe!
I've updated the lock file in https://github.com/lancaster-university/codal-microbit-v2/pull/457/commits/b71998c8fff13ec143939252861254b034396698, tested again and it's fixed 🥳

closed 2025-03-28 by microbit-carlos