lancaster-university/codal-microbit-v2 #469: audio refactor: Takes ~40ms for `levelSPL->getValue()` to return valid data.

microbit-carlos, 2025-03-13:
A regression from:
- https://github.com/lancaster-university/codal-microbit-v2/issues/301

On a cold power up, `uBit.audio.levelSPL->getValue()` returns zero for 38 to 40 ms since the first call.

```cpp
#include "MicroBit.h"

MicroBit uBit;

int main() {
    uBit.init();

    // Wait until button press to give us time to connect a serial terminal
    while (!uBit.buttonA.isPressed());

    uBit.serial.printf("\nstart\n");
    uint32_t startTime = uBit.systemTime();
    while (true) {
        uBit.serial.printf("(%dms) %d\n", uBit.systemTime() - startTime, (int)uBit.audio.levelSPL->getValue());
        uBit.sleep(1);
    }
}
```
```json
{
    "target": {
        "name": "codal-microbit-v2",
        "url": "https://github.com/lancaster-university/codal-microbit-v2",
        "branch": "audio-refactor",
        "type": "git"
    },
    "config":{
        "MICROBIT_BLE_ENABLED" : 0,
        "MICROBIT_BLE_PAIRING_MODE": 0
    }
}
```
With codal-microbit-v2 at commit [4a5aab37f97c7c5048cb7e317f461391da629cb1](https://github.com/lancaster-university/codal-microbit-v2/commit/4a5aab37f97c7c5048cb7e317f461391da629cb1).

Output:
```
start
(0ms) 0
(2ms) 0
(6ms) 0
(10ms) 0
(14ms) 0
(18ms) 0
(22ms) 0
(26ms) 0
(30ms) 0
(34ms) 0
(38ms) 0
(42ms) 47
```

CC @finneyj

finneyj, 2025-03-31:
Thanks @microbit-carlos. 

Fixed here:
https://github.com/lancaster-university/codal-core/commit/9fed73982ab392343ab5a8b47ca731b14c3eb6cb

microbit-carlos, 2025-03-31:
Thanks Joe! Can confirm it now blocks for around ~50ms and the first returned value looks valid 🎉

closed 2025-03-31 by microbit-carlos