lancaster-university/codal-microbit-v2 #470: audio refactor: Listening and then ignoring mic events keeps the microphone enabled.

microbit-carlos, 2025-03-25:
@finneyj, with the current audio pipeline this keep the mic LED permanently on.

[MICROBIT.hex.zip](https://github.com/user-attachments/files/19448441/MICROBIT.hex.zip)

```cpp
#include "MicroBit.h"

MicroBit uBit;

static void onLoud(MicroBitEvent) {
    uBit.serial.printf("LOUD\n");
}
static void onQuiet(MicroBitEvent) {
    uBit.serial.printf("QUIET\n");
}

int main() {
    uBit.init();

    uBit.messageBus.listen(DEVICE_ID_SYSTEM_LEVEL_DETECTOR, LEVEL_THRESHOLD_HIGH, onLoud);
    uBit.messageBus.listen(DEVICE_ID_SYSTEM_LEVEL_DETECTOR, LEVEL_THRESHOLD_LOW, onQuiet);
    uBit.sleep(1000);
    uBit.messageBus.ignore(DEVICE_ID_SYSTEM_LEVEL_DETECTOR, LEVEL_THRESHOLD_HIGH, onLoud);
    uBit.messageBus.ignore(DEVICE_ID_SYSTEM_LEVEL_DETECTOR, LEVEL_THRESHOLD_LOW, onQuiet);

    uBit.display.print("A");
    uBit.serial.printf("\nstart\n");

    while (true) {
        uBit.sleep(420);
    }
}

```
```json
{
    "target": {
        "name": "codal-microbit-v2",
        "url": "https://github.com/lancaster-university/codal-microbit-v2",
        "branch": "audio-refactor",
        "type": "git"
    },
    "config":{
        "MICROBIT_BLE_ENABLED" : 0,
        "MICROBIT_BLE_PAIRING_MODE": 0
    }
}
```
With codal-microbit-v2 at commit [4a5aab37f97c7c5048cb7e317f461391da629cb1](https://github.com/lancaster-university/codal-microbit-v2/commit/4a5aab37f97c7c5048cb7e317f461391da629cb1).

microbit-carlos, 2025-03-25:
This might not be a regression, as the same behaviour can be observed building agains the main branch and [v0.2.68](https://github.com/lancaster-university/codal-microbit-v2/releases/tag/v0.2.68) tag.

finneyj, 2025-03-28:
Yes! I just noticed this independently whilst investigating one of your other bugs. Interesting!

I'll take a look @microbit-carlos...

finneyj, 2025-03-31:
OK - I've added a new message bus listener mode to detect removal of event listeners, and implemented this for LevelDetectorSPL on Microbit.

Work completes in these patches: https://github.com/lancaster-university/codal-core/commit/5c0d922e5cd85ff034d57c76e4f61dc0014b42b1 and https://github.com/lancaster-university/codal-microbit-v2/commit/6f47847abc6bc46fcbb727c9e8a9745cd55c5fe0

microbit-carlos, 2025-03-31:
Great, thanks Joe! I can confirm this is fixed 🎉

closed 2025-03-31 by microbit-carlos