lancaster-university/codal-microbit-v2 #471: audio refactor: Sound Level 8 bit result doesn't reach max value

microbit-carlos, 2025-03-25:
@finneyj, with the current audio pipeline and blowing on the microphone hard, the sound level in 8-bit format doesn't reach a value above 230ish.

[MICROBIT.hex.zip](https://github.com/user-attachments/files/19451126/MICROBIT.hex.zip)

```cpp
#include "MicroBit.h"

MicroBit uBit;

int main() {
    uBit.init();

    uBit.display.print("A");
    uBit.serial.printf("\nstart\n");
    while (true) {
        uBit.serial.printf("%d (%d)\n",
           (int)uBit.audio.levelSPL->getValue(LEVEL_DETECTOR_SPL_DB),
           (int)uBit.audio.levelSPL->getValue(LEVEL_DETECTOR_SPL_8BIT));
        uBit.sleep(50);
    }
}
```
```json
{
    "target": {
        "name": "codal-microbit-v2",
        "url": "https://github.com/lancaster-university/codal-microbit-v2",
        "branch": "audio-refactor",
        "type": "git"
    },
    "config":{
        "MICROBIT_BLE_ENABLED" : 0,
        "MICROBIT_BLE_PAIRING_MODE": 0
    }
}
```
With codal-microbit-v2 at commit [4a5aab37f97c7c5048cb7e317f461391da629cb1](https://github.com/lancaster-university/codal-microbit-v2/commit/4a5aab37f97c7c5048cb7e317f461391da629cb1).

finneyj, 2025-05-15:
I just tested this, and it seems to reach 255 for me after the changes I made discussed in #472.

So I think we can close this off when we're ready.

microbit-carlos, 2025-05-16:
Yes, this now fixes, thanks Joe! 🎉

closed 2025-05-16 by microbit-carlos