lancaster-university/codal-microbit-v2 #475: Occasional corruption of neopixel wire data causing visual glitching

kevinjwalters, 2025-04-01:
I've been looking into glitches on RGB LEDs in MicroPython v2,1.2 on a BBC micro:bit V2. By accident/luck I caught one in a [video of the 60 RGB pixels on a Kitronik ZIP Halo HD](https://www.youtube.com/watch?v=lpnIn2PC4GA) at 04:25. There are two video frames at 25fps of it. My code updates the LEDs continuously at 5-25 times a second rather than just once a second. I suspect this may not have been reported by others as this is only happened every (very roughly) 1 in 10000 executions of `show()`.

Since then I've tried these on five other later revision ZIP Halo HD boards with their own micro:bit V2 and they all do this. MicroPython has the drive strength [set to normal rather than high on the V2 creating a discrepancy with the V1](https://github.com/microbit-foundation/micropython-microbit-v2/issues/83). Before I started using a logic analyzer I ruled out drive strength being an issue.

With the help of an 4tronix Bit:Zero Breakout, an Ikalogic SQ25 and a bit of programming ingenuity I've caught various traces of the wire protocol looking very broken. There's some of my older capture in the comments in https://github.com/microbit-foundation/micropython-microbit-v2/issues/227 . I was detected the problems by displaying dim patterns and using the micro:bit's ability to measure ambient light. I've moved on to a time based detection now as `show()` runs too fast (e.g. 1.45ms rather than 1.79ms for 60 LEDs), at least for one type of corruption. This suggests the problem is more likely to lie in the code which transmits the data which AIUI lives in https://github.com/lancaster-university/codal-microbit-v2 but issues are handled in this repo according to @microbit-carlos.

Some more recent traces follow. CH 2 is pulsed just before the `show()` is executed, CH 3 is pulsed when the program thinks there's an error to trigger the (end of) capture by the logic analyzer (this one is small and has limited memory and no continuous streaming). I've been experimenting with increasing the number of WS2812B pixels in software as the ones in 60 harmlessly disappear into the ether.

I have started playing with the pixel length as I'm wondering if there's a problematic buffer somewhere and the length/multiple affects things. The ones below are detected by short `show()` time. While one was running I saw a brief small flash which it did not detect suggesting a Case 3 may exist.

###  Corruption Case 1 - 92 pixels

[image]

First section of obvious corruption:

[image]

Second section:

[image]

### Corruption Case 2 -  95 pixels

This one has a load of spiky rubbish before it messing up the protocol's "reset" preamble. The LEDs for this one actually look okay but I think it's because they don't see a legal protocol and ignore it.

[image]

The "spikes" are all the same, 27 pulses which resemble the data transmissions.

[image]

The rest looks similar to Case 1.

[image]

[image]

---

The [documentation](https://microbit-micropython.readthedocs.io/en/latest/neopixel.html) mentions:

> From our tests, the Microbit NeoPixel module can drive up to around 256 NeoPixels. Anything above that and you may experience weird bugs and issues.

I'm not sure if that's because the original implementation blocked all interrupts or because of this bug or both, or something else!

---

Other neopixel tickets:
 - I see there's an unresolved bug which doesn't cause any issues in https://github.com/lancaster-university/codal-microbit-v2/issues/241 - perhaps that's worth cleaning up if this code is being worked on.
 -

finneyj, 2025-04-03:
Hi @kevinjwalters - thanks for the detailed bug report!

From your description it does sound a little bit like the corruption is coming from the codal side of the driver. 

@carlosperate - I noticed a conditional compile that is very prominent in the code:
https://github.com/lancaster-university/codal-nrf52/blob/6a7a5bb430d29088d7bb4cd2f948d3c66f9e1bae/source/neopixel.cpp#L9

HARDWARE_NEOPIXEL is enabled by default in C++ builds. Can you confirm that this isn't being overridden by micropython? I don't want to look at he wrong code path :)

Also @kevinjwalters - I wonder if it might be an interesting experiment to set HARDWARE_NEOPIXEL to "0" in a test micropython build, just to see if the bug reproduces with the old software only implementation? That could narrow down the problem to the handling of the hardware PWM module.... (which does look likely based on a quick look at the data analyser traces above... those long period square waves make that seem likely!)

I'll take a look at the code in the meantime....

finneyj, 2025-04-03:
Hmm... I do notice that there is a code path here that allows access to the instance variables of the NRF52PWM class from interrupt context:

https://github.com/lancaster-university/codal-nrf52/blob/6a7a5bb430d29088d7bb4cd2f948d3c66f9e1bae/source/NRF52PWM.cpp#L318

Which means they really should be declared volatile, but aren't currently:

https://github.com/lancaster-university/codal-nrf52/blob/master/inc/NRF52PWM.h#L32

Could be a potential cause of some occasional corruption...

microbit-carlos, 2025-04-03:
Thanks for looking into this Joe!

> HARDWARE_NEOPIXEL is enabled by default in C++ builds. Can you confirm that this isn't being overridden by micropython? I don't want to look at he wrong code path :)

I can confirm MicroPython does not change `HARDWARE_NEOPIXEL` and it uses the PWM path.

>Which means they really should be declared volatile, but aren't currently:

@finneyj I've just cannonballed this a bit, but would a MicroPython build with this be enough to test this?

[image]

@kevinjwalters do you know how to send MicroPython Python files via seria with tools like https://github.com/ntoll/microfs ?
I've created this MicroPython build with the volatile patch. You would need to drag and drop this hex file into the micro:bit and then use microsf to send the python code as a `main.py` file (so that it runs on startup).

[micropython-volatile.hex.zip](https://github.com/user-attachments/files/19585975/micropython-volatile.hex.zip)

kevinjwalters, 2025-04-03:
I did another test with some code that uses RGB values that create a numerically asending stream of numbers in the (GRB) wire protocol to help identify what goes missing.

[image]

This had 3.5 transmissions in it, i chopped out the half and left three in there. I assumed first two were good and third was problematic. An interesting small detail about the first one is it has an _extra pulse on the end that should not be there_.

[image]

The third one looks similar to Case 1 but isn't identical, it has a small pulse at the start and the first gap is low.

[image]

Manually decoding the numbers it looks like this starts with 0b000111 (7 with 2 bits missing), then 0b00001000 (8) meaning 0-6 and 2 bits of 7 went AWOL.

[image]

kevinjwalters, 2025-04-03:
@microbit-carlos I'll give it a go this week. I use Thonny normally for interactive copying of files to micro:bit. I think I have the microfs on my Pi 4.

kevinjwalters, 2025-04-03:
@finneyj I have not used the volatile patch MicroPython yet.

I have another ultra simple example showing the extra bogus pulse on the end.

```
>>> from microbit import pin8, pin9
>>> import neopixel
>>> pix = neopixel.NeoPixel(pin8, 3)
>>> pix[0] = (255, 0, 0)
>>> pix[1] = (63, 63, 63)
>>> pix[2] = (8, 0, 8)
>>> pix.show() ; pin9.write_digital(1)
```

[image]

One other curiosity there is pin9 goes high before data and reset period has completed. I thought MicroPython used a (fully) blocking `show()`. It must block to a major extent as I've been detecting problems by short durations for 60 or more pixels. Perhaps it returns when the last few are sent?

finneyj, 2025-04-03:
Cool - thanks @kevinjwalters. 

Yes, the PWM driver is double buffered, so most likely the thread will be released when the final buffer has entered the DMA hardware and there's nothing else to do....

kevinjwalters, 2025-04-03:
The patched MicroPython still exhibits the problem. The numerically ascending code seems to be good at quickly generating an erroneous `show()`. It's [neopixel-thirty-bug-timedetector-13.py](https://github.com/kevinjwalters/micropython-examples/blob/master/microbit/neopixel-thirty-bug-timedetector-13.py)

```
MicroPython v1.22.0 on 2025-04-03; micro:bit v2.1.1 with nRF52833
```

just produced

```
R13 ZIPCOUNT 60
R13 SLOW 1932
```

This is interesting, a quick glance and it looks like a false positive but looking closer this capture shows good integrity on the data but a _short overall transmission_ suggesting some type of truncation.

[image]

[image]

The decode for LED 52 to LED 60 (numbers 1 to 60) is missing because LED 53's data only has 13 proper pulses then it gets left high.

kevinjwalters, 2025-04-03:
I can change the number of pixels while program is running on that last version, I bumped it up to 63 and it produced something that looks like Case 2 on volatile patched MicroPython.  The CH 2 pulse is just before the `show()` call so the spiky rubbishis clearly leftover from previous `show()`.

[image]

kevinjwalters, 2025-04-03:
> Cool - thanks [@kevinjwalters](https://github.com/kevinjwalters).
> 
> Yes, the PWM driver is double buffered, so most likely the thread will be released when the final buffer has entered the DMA hardware and there's nothing else to do....

Is that safe then if the code does this

```
pixels.show()
pixels.show()  # immediately after first
```

?

kevinjwalters, 2025-04-08:
@martinwork @microbit-carlos Is MakeCode useful here? I think it sits on CODAL and it's incredibly fast to code simple things in. (I am away from my micro:bits at the moment.)

finneyj, 2025-04-08:
It does indeed @kevinjwalters.

I haven't heard any reports of issues with makecode, which is why we thought it might be a good idea to eliminate micropython from the list of possible causes... MakeCode programs do tend to be quite short though, micropython typically gives us more of a workout. :)

martinwork, 2025-04-08:
@microbit-carlos I'll have a look.

martinwork, 2025-04-10:
@microbit-carlos @finneyj @kevinjwalters 
C++ version of
https://github.com/kevinjwalters/micropython-examples/blob/26ebc18b173612025081591c2bf8fd4602468984/microbit/neopixel-thirty-bug-timedetector-13.py#L1

[neopy.zip](https://github.com/user-attachments/files/19684629/neopy.zip)

kevinjwalters, 2025-04-10:
I did a quick port of the critical functionality in MakeCode https://makecode.microbit.org/_fm9Md2dicbvp but I have no idea what implementation this is using for sending data to NeoPixel.

I've been running this along side the C++ version (thanks @martinwork) on two micro:bits and they are both running fine based on 30 mins runtime.

kevinjwalters, 2025-04-11:
@microbit-carlos Is there a potential clash here between the PWM nRF52 resources the CODAL uses for sending the pixel data and other stuff in MicroPython? If this isn't obvious then is backout to the nRF51 simple, proven version an interim solution until this problem is understood?

kevinjwalters, 2025-04-13:
I stripped the bug reproduction right back to minimal functionality and [neopixel-thirty-bug-timedetector-16.py](https://github.com/kevinjwalters/micropython-examples/blob/master/microbit/neopixel-thirty-bug-timedetector-16.py) will error almost immediately and same code with a `gc.collect()` in the loop has lasted 12 hours. The collect will change the timing a bit so I'd say this is definite but it's worth investigating by someone who understand how PWM/interrupts interact with garbage collection. Are heap allocations in C++ using the same heap as MicroPython?

I had a look at the code and I noticed there's no an explicit setting of the interrupt priority for PWM. Is that intentional? There's three blocks in there, here's first.

https://github.com/lancaster-university/codal-nrf52/blob/bdfa1eca85dbbd6fc8aaf5fc4c3fd00c1f65762d/source/NRF52PWM.cpp#L76-L78

The pattern elsewhere in the code is handler+priority+clear+enable.

https://github.com/lancaster-university/codal-nrf52/blob/bdfa1eca85dbbd6fc8aaf5fc4c3fd00c1f65762d/source/NRF52PDM.cpp#L84-L93

kevinjwalters, 2025-04-13:
I am not using deep function calls or recursion and can't think of any intentional heavy stack usage on my part.

I get a lot of 030 errors on any long running code (~12-48 hours), see https://github.com/microbit-foundation/micropython-microbit-v2/issues/228

So how do the interrupt priorities mesh together when they are set in multiple places? The settings with constants are easy to pick out are (I can't guarantee these are executed but I see them in the code):

| Interrupt  | codal-microbit-v2 | codal-nrf52 | micropython-microbit-v2 |
| ------------- | ------------- | -------------- | ---- | 
| SAADC_IRQn | 5 | 0 | | 
| RADIO_IRQn | 4 | 2 | 3 |

finneyj, 2025-04-14:
The most specific priority will take prescedence. So in you example above, I would expect the priorities to be 5 and 3 respectively. 
The irq priorities can be changed dynamically, but the code *should* be set up such that the most specific repo takes priority.

Wait - you said "I get a lot of 030 errors on *any* long running code (~12-48 hours)" ?

030 errors are generally only caused by corruption of the codal heap from somewhere. If you're seeing 030 errors on other programs, then maybe there's something else causing this problem?

kevinjwalters, 2025-04-14:
Yep, 030 are very common for anything I do that's long running. I thought I'd try and get this one out the way before dealing with that but they could be linked.

For this code 

```
    ManagedBuffer buffer(outputBufferSize);

    uint16_t *out = (uint16_t *) &buffer[0];
    uint16_t *end = (uint16_t *) &buffer[buffer.length()];

    while (out < end)
```

from https://github.com/lancaster-university/codal-nrf52/blob/16edb84e43f5faeb64b93cb9cd0ee1c4260ae31b/source/WS2812B.cpp#L89-L94 is that just wrong because (from inspection) it's using a `ManagedBuffer` which operates on _byte_ counts but then trying to use the buffer to store `unt16_t` which are _two bytes_ so half the buffer will be off the end of the allocated storage? `buffer.length() // sizeof(uint16_t)` ?

martinwork, 2025-04-14:
SAADC_IRQn - codal-nrf52 sets it in the constructor, then it's changed by codal-microbit-v2.
RADIO_IRQn - codal-microbit-v2 doesn't use the codal-nrf52 radio code. micropython-microbit-v2 also has it's own radio code, so I think the priority might be different in python after it's set in the radio initialisation.

Could the python code neopixel problems be because the HFCLK is not enabled? Does it still happen with radio enabled?

kevinjwalters, 2025-04-15:
I've got some new weirdness I can't explain. I wanted to look again at different pixel counts but I had what I thought was a bug where `show()` would return in around 170us. I spent ages looking at that and it was true, I can get the micro:bit 2 into a state where `show()` returns really quickly until the board is reset with the button or `machine.reset()`, the soft reset with control-d from REPL does not restore health.

[neopixel-thirty-bug-timedetector-21.py](https://github.com/kevinjwalters/micropython-examples/blob/master/microbit/neopixel-thirty-bug-timedetector-21.py) shows all this. It tries to do 5 minute runs at various pixel counts but it will pause if it gets a slow request (the baseline is now measured) and then print `BGRD` for sub 200us show(). This is typical output via Thonny's serial window, I normally used control-d to start main.py but here I'm using `machine.reset()` to get into a known state.

```
Traceback (most recent call last):
  File "main.py", line 191, in <module>
KeyboardInterrupt:
MicroPython v1.18 on 2023-10-30; micro:bit v2.1.2 with nRF52833
Type "help()" for more information.
>>>
>>> import machine
>>> machine.reset()
R21 MATH 445175
R21 SUMM 5 0 392332
R21 SUMM 10 114309 238073
R21 BGRD 167
R21 MATH 1002564
Traceback (most recent call last):
  File "main.py", line 191, in <module>
KeyboardInterrupt:
MicroPython v1.18 on 2023-10-30; micro:bit v2.1.2 with nRF52833
Type "help()" for more information.
>>>
MPY: soft reboot
R21 MATH 1004595
R21 BGRD 174
R21 MATH 1001602
```

From the `SUMM`ary That means there were 0 errors for 5 pixels, and 114309 (48%) for 10. Those run over a 5 minute period assuming ticks_ms() is okay.

There's some simple maths in a big loop which took 445ms when all was ok but after MicroPython on micro:bit V2 goes into funny state then it's reported as taking 1003ms, 1005ms, 1002ms so either `ticks_us()` breaks or something is running in background stealing CPU. 

I just re-ran it in Arduino's Serial Monitor for the benefit of the timestamping ([remember MicroPython's timing is a bit out](https://github.com/microbit-foundation/micropython-microbit-v2/issues/225)) to check that final `MATH` print line.

```
12:17:25.454 -> >>> exit
12:17:27.783 -> Traceback (most recent call last):
12:17:27.783 ->   File "<stdin>", line 1, in <module>
12:17:27.827 -> NameError: name 'exit' isn't defined
12:17:27.827 -> >>> 
12:18:18.625 -> >>> 
12:18:19.168 -> >>> import machine
12:18:27.212 -> >>> machine.reset()
12:18:31.444 -> R21 MATH 444994
12:23:34.421 -> R21 SUMM 5 0 392333
12:28:37.451 -> R21 SUMM 10 0 277721
12:33:40.593 -> R21 SUMM 16 26 240419
12:38:43.493 -> R21 SUMM 21 0 214126
12:43:46.554 -> R21 SUMM 26 0 191504
12:48:49.703 -> R21 SUMM 32 1 173198
12:53:52.687 -> R21 SUMM 60 0 112333
12:58:55.816 -> R21 SUMM 64 0 111737
13:03:58.877 -> R21 SUMM 95 7 82386
13:09:01.786 -> R21 SUMM 5 0 392333
13:14:04.820 -> R21 SUMM 10 174790 217209
13:14:04.898 -> R21 BGRD 168
13:14:05.950 -> R21 MATH 1002887
```

05.950-04.898 = 1.052 so basic maths / for loops / code is now 2.25x slower.

---

Thinking about Thonny I've always assumed it did something clever to present the REPL in a window and allow file copying and possibly some variable printing. https://github.com/thonny/thonny/wiki/MicroPython/6c18c19b5271b907e51d7005ddb3bcec67ca41a2#thonny-repl-vs-plain-micropython-serial-repl mentions it puts MicroPython in "raw REPL mode".

martinwork, 2025-04-15:
I reduced the period to a few seconds and printed the baseline array.
Why do the times in each baseline array increase?
```
this is for 5 pixels
R21 SHMN 240 [280, 281, 281, 281, 287, 287, 287, 287, 288, 288, 288, 288, 288, 288, 288, 288]

R21 SHMN 664 [742, 743, 753, 758, 758, 758, 758, 759, 759, 767, 798, 805, 812, 838, 977, 977]

R21 SHMN 2818 [3142, 3148, 3149, 3150, 3151, 3153, 3168, 3170, 3176, 3178, 3181, 3198, 3201, 3216, 3232, 3247]

back to 5
R21 SHMN 235 [281, 281, 281, 281, 281, 282, 282, 282, 288, 322, 322, 322, 334, 335, 335, 375]
```

kevinjwalters, 2025-04-15:
@martinwork  I assume the values for each pixel count differ a little because of timing accuracy issues and interpreted language. They ascend monotonically because of the `.sort()` in the program - that's there to facilitate IQR - I'm trying to ignore anything that's way too fast (broken) or a bit slow. MicroPython uses a blocking implementation for `show()` so as there are more pixels it takes longer to send the data (800 kbits per second plus reset period before and after) assuming everything is working correctly.

martinwork, 2025-04-15:
Ah yes, of course! I had convinced myself the other times were all on the high side of the baseline values, but that's not true either - the averages are quite close. 

I'm not seeing a large number of times less than show_min_us. I'm building with https://python.microbit.org/ and capturing the output with tera term.

Seems like a large variation in times though. With certain pixel counts (e.g. 10) having a larger range in proportion to the average.

kevinjwalters, 2025-04-17:
@martinwork Any thoughts on that persistent terrible slowness that kicks in? 

On other matters, ideally I'd like a workaround for the [030 errors](https://github.com/microbit-foundation/micropython-microbit-v2/issues/228). Is there a way for these to either auto-resset micro:bit or display, say, 5 times and then auto-reset?

kevinjwalters, 2025-04-19:
This bug has another variety where most or all of the `show()` are corrupted. It's both problematic and attractive at the same time. Control-C and restart does not fix it, a reset of the micro:bit is required. [YouTube video](https://www.youtube.com/watch?v=YfX25L3Wfts) showing the second case I have seen of this, much rarer than the one-off glitches.

martinwork, 2025-04-20:
With a sleep_ms(1) added into the timing loops, [neopixel-thirty-bug-timedetector-21.py](https://github.com/kevinjwalters/micropython-examples/blob/master/microbit/neopixel-thirty-bug-timedetector-21.py) seems to run OK. Without the sleep, it seems to take about 2.5 to 3 minutes to go wrong, so I also increased the period to 10 minutes

I have been experimenting with a C++ sample, but failed to make the same thing happen. 

With a pause between shows, the PWM code repeats the same sequence. No pause leads to some variation, which is part of the source of the range of timings. When the call returns the data is usually still being sent, and the next call waits for the previous data to finish. Sometimes the new data is streamed on; other times the PWM is stopped and restarted.

Python timing variation seems to be 30-50% compared to about 10% with C++.

martinwork, 2025-04-20:
This modified version of the original sample works: [neopixel-thirty-bug-timedetector-21-works.py.zip](https://github.com/user-attachments/files/19825621/neopixel-thirty-bug-timedetector-21-works.py.zip). The last change, when it started working, was to remove the pin pings.

Running the original with sleep_us(100) commented out in one_ping, there was 1 "slow" count.
```
R21 MATH 444322
R21 SUMM 5 0 543703
R21 SUMM 10 1 355516
```

kevinjwalters, 2025-04-20:
@martinwork I rechristened your version. I left it running on my desk on two ZIP Halo HDs. These can be useful for visually catching errors as if only 21 pixels are set and then a load of coloured gibberish appears on 22nd and above pixels it will persist until over-written by later period large pixel counts.

```
MPY: soft reboot
R22 MATH 444881
R22 SUMM 10 0 407701
R22 SUMM 16 0 331240
R22 SUMM 21 0 280334  <-- at least one error occurred during this period which was not detected by timing test
```

Either the timing test doesn't catch all problems or the threshold errs on the side of removing false positives as I just got an error, see photograph below.

[image]

I've had them running for about 1.5 hours and all reported 0 errors so far, errors must be incredibly rare for this code. I just had _another_ visually obvious error with 21 pixel updates on same ZIP Halo HD.

````
R22 SUMM 10 0 409637
R22 SUMM 16 0 331730
R22 SUMM 21 0 280356
R22 SUMM 26 0 243589
R22 SUMM 32 0 214091
R22 SUMM 60 0 129003
R22 SUMM 64 0 127825
R22 SUMM 95 0 91170
R22 SUMM 10 0 408933
R22 SUMM 16 0 331764
R22 SUMM 21 0 280322
R22 SUMM 26 0 243560
R22 SUMM 32 0 213910
R22 SUMM 60 0 128862
R22 SUMM 64 0 127783
R22 SUMM 95 0 91047
R22 SUMM 10 0 408922
R22 SUMM 16 0 331774
R22 SUMM 21 1 280372  <-- this error might be the same as the second visually apparent error
```

---

@finneyj A small amount more effort but allowing the buffer size to be set for this particular debugging via kwarg on MicroPython's Neopixel object would allow for more experimentation. There's already a `setBufferSize()` member function to override the default value set in the constructor. That would also allow it to be shrunk down to do practical testing on starvation to ensure it's robust.

For the CODAL heap I noticed a research article saying it's reentrant. I'm still not sure how this interacts or doesn't interact with the micropython heap. I don't do much low-level stuff but it appears unnecessary to be doing dynamic allocations during interrupts here? My uninformed instinct would be to allocate the two buffers holding pwm values right at the start (or just one if it all happened to fit in one) and then just free them from the interrupt handler when done.

kevinjwalters, 2025-04-20:
For `R22` it eventually went errortastic but it hasn't yet stopped due to the ultra-quick sub 200us oddity...
 
```
MicroPython v1.18 on 2023-10-30; micro:bit v2.1.2 with nRF52833
Type "help()" for more information.
>>>
>>>
MPY: soft reboot
R22 MATH 443287
R22 SUMM 10 0 409637
R22 SUMM 16 0 331730
R22 SUMM 21 0 280356
R22 SUMM 26 0 243589
R22 SUMM 32 0 214091
R22 SUMM 60 0 129003
R22 SUMM 64 0 127825
R22 SUMM 95 0 91170
R22 SUMM 10 0 408933
R22 SUMM 16 0 331764
R22 SUMM 21 0 280322
R22 SUMM 26 0 243560
R22 SUMM 32 0 213910
R22 SUMM 60 0 128862
R22 SUMM 64 0 127783
R22 SUMM 95 0 91047
R22 SUMM 10 0 408922
R22 SUMM 16 0 331774
R22 SUMM 21 0 280364
R22 SUMM 26 0 243574
R22 SUMM 32 0 214023
R22 SUMM 60 0 128884
R22 SUMM 64 0 127779
R22 SUMM 95 0 91109
R22 SUMM 10 0 408500
R22 SUMM 16 0 331757
R22 SUMM 21 0 280349
R22 SUMM 26 0 243546
R22 SUMM 32 0 214077
R22 SUMM 60 0 128888
R22 SUMM 64 0 127777
R22 SUMM 95 0 91116
R22 SUMM 10 0 408497
R22 SUMM 16 0 331745
R22 SUMM 21 3 280357
R22 SUMM 26 0 243582
R22 SUMM 32 0 213968
R22 SUMM 60 0 128879
R22 SUMM 64 0 127795
R22 SUMM 95 0 91160
R22 SUMM 10 0 408595
R22 SUMM 16 1 331691
R22 SUMM 21 0 280348
R22 SUMM 26 0 243574
R22 SUMM 32 0 213886
R22 SUMM 60 0 128877
R22 SUMM 64 0 127785
R22 SUMM 95 0 91047
R22 SUMM 10 0 408920
R22 SUMM 16 0 331744
R22 SUMM 21 1 280360
R22 SUMM 26 0 243586
R22 SUMM 32 0 214032
R22 SUMM 60 0 128993
R22 SUMM 64 0 127818
R22 SUMM 95 0 91167
R22 SUMM 10 0 408585
R22 SUMM 16 0 331697
R22 SUMM 21 0 280393
R22 SUMM 26 0 243556
R22 SUMM 32 0 213955
R22 SUMM 60 0 128878
R22 SUMM 64 0 127769
R22 SUMM 95 0 91121
R22 SUMM 10 0 408428
R22 SUMM 16 1 331692
R22 SUMM 21 0 280322
R22 SUMM 26 0 243578
R22 SUMM 32 0 213903
R22 SUMM 60 0 128883
R22 SUMM 64 0 127775
R22 SUMM 95 0 91043
R22 SUMM 10 0 408667
R22 SUMM 16 0 331722
R22 SUMM 21 0 280363
R22 SUMM 26 0 243565
R22 SUMM 32 0 214040
R22 SUMM 60 0 129028
R22 SUMM 64 0 127811
R22 SUMM 95 0 91170
R22 SUMM 10 0 408855
R22 SUMM 16 2 331696
R22 SUMM 21 101603 316442
R22 SUMM 26 117296 319764
R22 SUMM 32 91956 299685
```

kevinjwalters, 2025-04-21:
I thought I'd have a play with `sleep_ms()` vs `sleep_us()` in new [neopixel-thirty-bug-timedetector-23.py](https://github.com/kevinjwalters/micropython-examples/blob/master/microbit/neopixel-thirty-bug-timedetector-23.py). Nothing conclusive to far, but this just `030` errored in 17 minutes, that's very unusual after such a short run time although the up time on that micro:bit would be 18 hours.

```
MPY: soft reboot
R23 COUNTS (1, 6, 11, 17, 22, 27, 33, 60, 63, 64)
R23 MATH 445251
R23 SUMM 1 0 14113 sleep_ms
R23 SUMM 1 0 12995 sleep_us
R23 SUMM 6 0 14151 sleep_ms
R23 SUMM 6 0 12904 sleep_us
R23 SUMM 11 0 13621 sleep_ms
R23 SUMM 11 0 12056 sleep_us
R23 SUMM 17 5 13078 sleep_ms
R23 SUMM 17 5 11689 sleep_us
R23 SUMM 22 0 11714 sleep_ms
R23 SUMM 22 0 11392 sleep_us
R23 SUMM 27 0 11302 sleep_ms
R23 SUMM 27 2 11080 sleep_us
R23 SUMM 33 0 11203 sleep_ms
R23 SUMM 33 0 10741 sleep_us
R23 SUMM 60 0 9380 sleep_ms
R23 SUMM 60 0 9116 sleep_us
R23 SUMM 63 0 9379 sleep_ms
```

I'll do a few more runs tomorrow.

kevinjwalters, 2025-04-21:
This mornings results from R23, this one only does a minute for each test.

**ZHH B**

```
MicroPython v1.18 on 2023-10-30; micro:bit v2.1.2 with nRF52833
Type "help()" for more information.
>>>
>>>
MPY: soft reboot
R23 COUNTS (1, 6, 11, 17, 22, 27, 33, 60, 63, 64)
R23 MATH 444128
R23 SUMM 1 0 14012 sleep_ms
R23 SUMM 1 0 12999 sleep_us
R23 SUMM 6 0 14097 sleep_ms
R23 SUMM 6 0 12913 sleep_us
R23 SUMM 11 0 13658 sleep_ms
R23 SUMM 11 0 12056 sleep_us
R23 SUMM 17 0 13101 sleep_ms
R23 SUMM 17 4 11688 sleep_us
R23 SUMM 22 0 11597 sleep_ms
R23 SUMM 22 0 11402 sleep_us
R23 SUMM 27 0 11339 sleep_ms
R23 SUMM 27 0 11092 sleep_us
R23 SUMM 33 0 11211 sleep_ms
R23 SUMM 33 0 10739 sleep_us
R23 SUMM 60 0 9514 sleep_ms
R23 SUMM 60 0 9209 sleep_us
R23 SUMM 63 0 9528 sleep_ms
R23 SUMM 63 0 9207 sleep_us
R23 SUMM 64 0 9622 sleep_ms
R23 SUMM 64 0 9202 sleep_us
R23 SUMM 1 0 14096 sleep_ms
R23 SUMM 1 0 12999 sleep_us
R23 SUMM 6 0 14113 sleep_ms
R23 SUMM 6 0 12913 sleep_us
R23 SUMM 11 0 13666 sleep_ms
R23 SUMM 11 0 12056 sleep_us
R23 SUMM 17 0 13161 sleep_ms
R23 SUMM 17 5 11690 sleep_us
R23 SUMM 22 0 11599 sleep_ms
R23 SUMM 22 0 11401 sleep_us
R23 SUMM 27 0 11318 sleep_ms
R23 SUMM 27 0 11092 sleep_us
R23 SUMM 33 0 11208 sleep_ms
R23 SUMM 33 0 10743 sleep_us
R23 SUMM 60 0 9431 sleep_ms
R23 SUMM 60 0 9210 sleep_us
R23 SUMM 63 0 9542 sleep_ms
R23 SUMM 63 0 9209 sleep_us
R23 SUMM 64 0 9636 sleep_ms
R23 SUMM 64 0 9208 sleep_us
R23 SUMM 1 0 14032 sleep_ms
R23 SUMM 1 0 12999 sleep_us
R23 SUMM 6 0 14126 sleep_ms
R23 SUMM 6 0 12913 sleep_us
R23 SUMM 11 0 13674 sleep_ms
R23 SUMM 11 0 12055 sleep_us
R23 SUMM 17 0 13106 sleep_ms
R23 SUMM 17 0 11690 sleep_us
R23 SUMM 22 0 11546 sleep_ms
R23 SUMM 22 0 11402 sleep_us
R23 SUMM 27 0 11420 sleep_ms
R23 SUMM 27 1 11092 sleep_us
R23 SUMM 33 0 11215 sleep_ms
R23 SUMM 33 4 10750 sleep_us
R23 SUMM 60 0 9435 sleep_ms
R23 SUMM 60 1 9211 sleep_us
R23 SUMM 63 0 9531 sleep_ms
R23 SUMM 63 0 9206 sleep_us
R23 SUMM 64 0 9606 sleep_ms
R23 SUMM 64 0 9209 sleep_us
R23 SUMM 1 0 14042 sleep_ms
R23 SUMM 1 0 12999 sleep_us
R23 SUMM 6 0 14104 sleep_ms
R23 SUMM 6 0 12913 sleep_us
R23 SUMM 11 0 13637 sleep_ms
R23 SUMM 11 0 12056 sleep_us
R23 SUMM 17 0 13075 sleep_ms
R23 SUMM 17 1 11690 sleep_us
R23 SUMM 22 0 11590 sleep_ms
R23 SUMM 22 0 11400 sleep_us
R23 SUMM 27 0 11334 sleep_ms
R23 SUMM 27 0 11092 sleep_us
R23 SUMM 33 0 11210 sleep_ms
R23 SUMM 33 2 10737 sleep_us
R23 SUMM 60 0 9472 sleep_ms
R23 SUMM 60 0 9205 sleep_us
R23 SUMM 63 0 9546 sleep_ms
R23 SUMM 63 0 9204 sleep_us
R23 SUMM 64 0 9619 sleep_ms
R23 SUMM 64 0 9203 sleep_us
R23 SUMM 1 0 14106 sleep_ms
R23 SUMM 1 0 12999 sleep_us
R23 SUMM 6 0 14112 sleep_ms
R23 SUMM 6 0 12916 sleep_us
R23 SUMM 11 0 13726 sleep_ms
then 030 error
```

---

**ZHH C** (other board on my desk)

```
MPY: soft reboot
R23 COUNTS (1, 6, 11, 17, 22, 27, 33, 60, 63, 64)
R23 MATH 443576
R23 SUMM 1 0 14074 sleep_ms
R23 SUMM 1 0 12996 sleep_us
R23 SUMM 6 0 14156 sleep_ms
R23 SUMM 6 0 12908 sleep_us
R23 SUMM 11 0 13693 sleep_ms
R23 SUMM 11 9484 11911 sleep_us
R23 BGRD 170
R23 MATH 1008601
R23 SUMM 17 0 1 sleep_ms
R23 BGRD 176
R23 MATH 1008613
```

martinwork, 2025-04-22:
It looks like the python heap is built in a static array. gc_collect seems to occur automatically every few seconds and takes 2ms.
https://github.com/microbit-foundation/micropython-microbit-v2/blob/3f22f306bcbc3461b0c3c60702b74bbce6689013/src/codal_port/main.c#L48

I added some trace to the codal part of the micropython runtime, and noticed **buffers being pulled with samplesSent rising much higher than expected**.
https://github.com/lancaster-university/codal-nrf52/blob/master/source/WS2812B.cpp#L84

I built this runtime [MICROBIT.zip](https://github.com/user-attachments/files/19856735/MICROBIT.zip) as below, and used mu to merge this python code [neopixel-thirty-bug-timedetector-21-P0.zip](https://github.com/user-attachments/files/19856742/neopixel-thirty-bug-timedetector-21-P0.zip), in which I changed the neopixel pin to P0, removed the gc.collect()s, and made the first pixel pulse green.

After a minute or so the DMESG output starts reporting very large samplesSent numbers - thousands rather than hundreds. The pixel changes slow down.

It looks like gc_collect() isn't the trigger. It still happens with gc.disable() & gc.enable() wrapping the main show(), and even wrapping the whole inner while loop (need to comment out the zip_px[0] = line, so it's not visible without DMESG output)

I think the same slow down occurs with a hex built in the Python Editor, but takes longer (30 minutes?) to slow down
[neopixel-thirty-bug-timedetector-21-P0-release.hex.zip](https://github.com/user-attachments/files/19857485/neopixel-thirty-bug-timedetector-21-P0-release.hex.zip)
  
<details>
<summary> To build the runtime </summary>

Clone and build https://github.com/microbit-foundation/micropython-microbit-v2
with modified **src/codal_app/codal.json** to add DMESG
```
        "DEVICE_DMESG_BUFFER_SIZE":1024,
        "DMESG_SERIAL_DEBUG":1,
        "DMESG_ENABLE":1,
```
and modified **lib/codal/libraries/codal-nrf52/source/WS2812B.cpp** to add
`
#include "codaldmesg.h"
`
and here https://github.com/lancaster-university/codal-nrf52/blob/master/source/WS2812B.cpp#L120
```
    if ( samplesSent > 128 * ( ( totalSamples + 127) / 128))
    {
        DMESG( "%d %d", (int) *(uint8_t *) data, (int) samplesSent);
    }
```

</details>

kevinjwalters, 2025-04-24:
Is there an easy way to see if stack/heap/memory corruption is happening here and to see what data is being written where it shouldn't be? There must be a few stacks in play here beyond basic C one. Perhaps Python, perhaps CODAL fiber things, interrupt handlers?

It seems to have gone but CircuitPython had a "sentinel" byte which was written to an area of the stack to help to see if it had been trashed: https://github.com/adafruit/circuitpython/blob/18207cd547e1ffbfbf1ead9f5a32eea69d079ec6/py/stackctrl.c Based on naming I always assumed this came from MicroPython but I'm wrong, it was introduced with https://github.com/adafruit/circuitpython/pull/175

kevinjwalters, 2025-04-24:
@martinwork I loaded up the version of MicroPython you made. I note it identifies itself as `MicroPython v1.23.0 on 2025-04-22; micro:bit v2.1.1 with nRF52833`.

I ran it for about 20 minutes with [neopixel-thirty-bug-timedetector-16.py](https://github.com/kevinjwalters/micropython-examples/blob/master/microbit/neopixel-thirty-bug-timedetector-16.py). That normally fails very quickly. I gave up with that and switched to [timedetector-23.py](https://github.com/kevinjwalters/micropython-examples/blob/master/microbit/neopixel-thirty-bug-timedetector-23.py). It's interesting to see the maths test that normally takes 440-445ms takes 482ms, something is either less efficient in this version or it's just the DEBUG overhead.

```
MPY: soft reboot
R23 COUNTS (1, 6, 11, 17, 22, 27, 33, 60, 63, 64)
R23 MATH 481841
R23 SUMM 1 0 14321 sleep_ms
UIPM: RECEIVED UNKNWON FRAME
...same line repeated another 64 times...
R23 SUMM 1 0 12972 sleep_us
R23 SUMM 6 0 14317 sleep_ms
UIPM: RECEIVED UNKNWON FRAME
...same line repeated another 64 times...
R23 SUMM 6 0 12954 sleep_us
0 512
0 640
0 768
0 896
0 1024
0 1152
0 1280
0 1408
0 1536
0 1664
0 1792
0 0 16768
0 16896
0 17024
0 17152
0 17280
0 17408
0 17536
0 176640 31360
0 31488
0 31616
0 31744
0 31872
0 32000
0 32128
0 32256
0 32384^M0 13696
0 13824
0 13952
0 14080
0 14208
0 14336
0 14464
0 14592^M0 28288
0 28416
0 28544
0 28672
0 28800
0 28928
0 29056
0 29184
0 29312
00 14464
0 14592
0 14720
0 14848
0 14976
0 15104
0 15232
0 15360
0 10 29184
0 29312
```

Are `UIPM: RECEIVED UNKNWON FRAME` to be expected?

Here's the putty capture: [putty-zhhb-20250424-debugmp-r16-i.txt](https://github.com/user-attachments/files/19898257/putty-zhhb-20250424-debugmp-r16-i.txt)

microbit-carlos, 2025-04-25:
Just a quick about MakeCode, as it uses its own ASM driver (which is also why the neopixel extension is not compatible with BLE), so it's not surprising this issue cannot be replicated there.
The neopixel extension (https://github.com/microsoft/pxt-neopixel/) still calls the ASM `sendBuffer()` implementation from https://github.com/microsoft/pxt-ws2812b.

microbit-carlos, 2025-04-25:
I'm a little lost, as there is a lot to digest here, but has it been confirmed that the issue is still present when disabling the garbage collector when calling `show()`?
If that's the case, do we have any other theories as to what might be causing the heap corruption?

martinwork, 2025-04-25:
I wanted to find anything that went wrong reasonably quickly and every time, so I could try changing something (like adding volatile) and know whether it was having any effect, but I haven't got around to that yet!

The rapid pulses in the python code in https://github.com/lancaster-university/codal-microbit-v2/issues/475#issuecomment-2822617643 slow down visibly with both current Python Editor and my test runtime.

With GC disabled, I couldn't see a way to produce the visible pulse, but the DMESG output from the test runtime behaves the same.

martinwork, 2025-04-30:
@finneyj My C++ sample also triggers dataReady < 0, after adding target_wait_us(250) in the loop,  

I've made just one changed in CODAL - to panic 998
https://github.com/martinwork/codal-nrf52/blob/badeee189e87007010c343ef2f52150944abebde/source/NRF52PWM.cpp#L258

Here's the hex... expect panic 998 within a minute or two
[neopix-250us.zip](https://github.com/user-attachments/files/19978686/neopix-250us.zip)

This hex has a shorter wait... notice slight flashes on the second pixel, and it tends to just stop.
[neopix-200us.zip](https://github.com/user-attachments/files/19978716/neopix-200us.zip)

It may be my imagination, but both hexes seem to behave differently after pressing RESET, compared to when just flashed. 

<details>
<summary>C++ source</summary>

```
#include "MicroBit.h"

MicroBit uBit;

int bufferSize = 30;

uint8_t *buffer = NULL;

void forever()
{
    uBit.sleep(1000);

    buffer = (uint8_t *) malloc( bufferSize);
    memset( buffer, 0, bufferSize);
    neopixel_send_buffer( uBit.io.pin[0], buffer, bufferSize);

    uBit.sleep(500);

    for ( int count = 0; true; count++)
    {
        buffer[ 0] = count % 256;

        neopixel_send_buffer( uBit.io.pin[0], buffer, bufferSize);

        //target_wait_us( 200);           // stops working quickly

        target_wait_us( 250);             // dataReady < 0

        //target_wait_us( 300);           // keeps going for longer
    }
}

int main() {
    uBit.init();
    create_fiber( forever);
    release_fiber();
    return 0;
}

    "config":{
        "DEVICE_DMESG_BUFFER_SIZE":0,
        "DMESG_SERIAL_DEBUG":0,
        "DMESG_ENABLE":0,

        "MICROBIT_BLE_ENABLED": 0,
        "MICROBIT_BLE_PAIRING_MODE": 1,
        "MICROBIT_BLE_DFU_SERVICE": 1,
        "MICROBIT_BLE_DEVICE_INFORMATION_SERVICE": 1,
        "MICROBIT_BLE_EVENT_SERVICE": 1,
        "MICROBIT_BLE_PARTIAL_FLASHING": 0,
        "MICROBIT_BLE_OPEN": 0,
        "MICROBIT_BLE_SECURITY_LEVEL": "SECURITY_MODE_ENCRYPTION_NO_MITM"
    }

```
</details>

martinwork, 2025-04-30:
When it locks up with target_wait_us(200), only the neopixel fiber has stopped. Another fiber keeps going.

[neopix-200us-2fibers.zip](https://github.com/user-attachments/files/19979073/neopix-200us-2fibers.zip)

<details>
<summary>source</summary>

```
#include "MicroBit.h"

MicroBit uBit;

int bufferSize = 30;

uint8_t *buffer = NULL;

void forever()
{
    uBit.sleep(1000);

    buffer = (uint8_t *) malloc( bufferSize);
    memset( buffer, 0, bufferSize);
    neopixel_send_buffer( uBit.io.pin[0], buffer, bufferSize);

    uBit.sleep(500);

    for ( int count = 0; true; count++)
    {
        buffer[ 0] = count % 256;

        neopixel_send_buffer( uBit.io.pin[0], buffer, bufferSize);

        target_wait_us( 200);           // stops working quickly

        //target_wait_us( 250);             // dataReady < 0

        //target_wait_us( 300);           // keeps going for longer
    }
}

void blinking_forever()
{
    while (true)
    {
        uBit.display.image.setPixelValue( 0, 0, uBit.display.image.getPixelValue( 0, 0) ? 0 : 255);
        uBit.sleep(200);
    }
}

int main() {
    uBit.init();
    create_fiber( forever);
    create_fiber( blinking_forever);
    release_fiber();
    return 0;
}

```

</details>

martinwork, 2025-04-30:
@finneyj Possible fixes...

If I am understanding the purpose, I think this test should be greater than, since a notify increases `locked`.  
https://github.com/lancaster-university/codal-core/blob/22aa198888097a3103ee13c5db5f21b06a6aea60/source/core/CodalFiber.cpp#L859
Like this... 
https://github.com/martinwork/codal-core/commit/2c67462eaebb55b83c5d8122db55d287cd982abe
That seems to make the 200us case behave like the 250us one.

I don't think there should be a notify during wait, so it seems like confirmation that there is a spurious IRQ.

The dataReady < 0 problem seems to go away after wrapping the tryPull calls in pullRequest with with a flag that makes the IRQs do nothing except reset the event flags - not tested for very long yet, and I'm not sure this is the right way, or the whole story...
https://github.com/martinwork/codal-nrf52/commit/7230f1fd7e4d1e874ca91d06f38901a80b96665f

Hopefully these C++ programs will keep running!
[neopix-200us-fixed.zip](https://github.com/user-attachments/files/19983418/neopix-200us-fixed.zip)
[neopix-250us-fixed.zip](https://github.com/user-attachments/files/19983417/neopix-250us-fixed.zip)

And here's a micropython runtime with the same fixes 
[test_neopix_mycropython.zip](https://github.com/user-attachments/files/19983436/test_neopix_mycropython.zip)
built from here 
https://github.com/martinwork/micropython-microbit-v2/tree/mw-test-python-v2-2

kevinjwalters, 2025-05-18:
How did your testing going for this? I'll be reunited with my micro:bits soon and can switch to this MicroPython version for some extensive testing. Is this likely to improve the `030` errors too?

kevinjwalters, 2025-05-27:
For my clock code which just sets three out of 60 neopixels to red, green and blue this new version of micropython from 
 https://github.com/lancaster-university/codal-microbit-v2/issues/475#issuecomment-2843150607 still glitches.

The micro:bit timing (using `micros()`) has improved a lot, is this using the crystal? This will fix https://github.com/microbit-foundation/micropython-microbit-v2/issues/225

kevinjwalters, 2025-05-29:
I'm just running some of bug reproduction code now on the new version of micropython from
https://github.com/lancaster-university/codal-microbit-v2/issues/475#issuecomment-2843150607

The third column is the number from the program.

```
ZHH B   v1.23.0   16
ZHH D   v1.23.0   21
ZHH E   v1.23.0   23
```

### [neopixel-thirty-bug-timedetector-16.py](https://github.com/kevinjwalters/micropython-examples/blob/master/microbit/neopixel-thirty-bug-timedetector-16.py)

Good so far but that as it's only been running for 30 mins that doesn't mean much.

### [neopixel-thirty-bug-timedetector-21.py](https://github.com/kevinjwalters/micropython-examples/blob/master/microbit/neopixel-thirty-bug-timedetector-21.py)

Fine with its first five minute test using 1 RGB LED then visually errorer very quickly
errored very quickly errors in the usual way within 5-15 minutes.

```
R21 MATH 456219
R21 SUMM 5 0 393757
R21 SUMM 10 249308 279802
```

### [neopixel-thirty-bug-timedetector-23.py](https://github.com/kevinjwalters/micropython-examples/blob/master/microbit/neopixel-thirty-bug-timedetector-23.py) 

This visually errors within 5-15 minutes.

```
R23 COUNTS (1, 6, 11, 17, 22, 27, 33, 60, 63, 64)
R23 MATH 455490
R23 SUMM 1 0 14564 sleep_ms
R23 SUMM 1 0 13140 sleep_us
R23 SUMM 6 0 14593 sleep_ms
R23 SUMM 6 0 13093 sleep_us
R23 SUMM 11 2 14162 sleep_ms
R23 SUMM 11 1 12239 sleep_us
R23 SUMM 17 0 13525 sleep_ms
R23 SUMM 17 33 11866 sleep_us
R23 SUMM 22 0 11742 sleep_ms
R23 SUMM 22 0 11561 sleep_us
R23 SUMM 27 0 11434 sleep_ms
R23 SUMM 27 0 11228 sleep_us
R23 SUMM 33 0 11273 sleep_ms
R23 SUMM 33 1 10918 sleep_us
R23 SUMM 60 0 9902 sleep_ms
R23 SUMM 60 0 9291 sleep_us
R23 SUMM 63 0 9885 sleep_ms
R23 SUMM 63 0 9290 sleep_us
R23 SUMM 64 0 9895 sleep_ms
R23 SUMM 64 0 9285 sleep_us
R23 SUMM 1 0 14565 sleep_ms
R23 SUMM 1 0 13140 sleep_us
R23 SUMM 6 0 14579 sleep_ms
R23 SUMM 6 0 13092 sleep_us
R23 SUMM 11 0 14132 sleep_ms
R23 SUMM 11 1 12240 sleep_us
R23 SUMM 17 0 13516 sleep_ms
R23 SUMM 17 27 11865 sleep_us
R23 SUMM 22 0 11715 sleep_ms
R23 SUMM 22 0 11558 sleep_us
```

There aren't any ultra-fast sub 200us `show()` executions so far.

kevinjwalters, 2025-06-01:
I've made a [neopixel-thirty-bug-timedetector-25.py](https://github.com/kevinjwalters/micropython-examples/blob/master/microbit/neopixel-thirty-bug-timedetector-25.py) to explore multiple executions of `show()` with unchanged data. It runs 2 `show()` for 31 to 35 pixels then repeats this with 1 `show()` and then 3 `show()`. There's a 100ms delay in the code to ensure `show()` can't over step each other which slows the execution rate down a fair bit.

For the SUMM lines the numerical values are 
 1. number of show()
 2. pixel count in use
 3. error count
 4. execution count over the two minute period
 5. threshold time in us based on measurements to decide its an error.

I've done one run over a few hours, see attached file for full output.

```
R25 SVMP 3.4.0; MicroPython v1.23.0 on 2025-04-30
R25 KZHH zhh-D.txt
R25 MATH 456539
R25 SUMM 2 31 1 1164 2438
R25 SUMM 2 32 5 1164 2450
R25 SUMM 2 33 0 1164 2464
R25 SUMM 2 34 0 1160 2766
R25 SUMM 2 35 0 1160 2754
R25 SUMM 1 31 0 1179 1142
R25 SUMM 1 32 0 1179 1139
R25 SUMM 1 33 0 1179 1142
R25 SUMM 1 34 0 1177 1282
R25 SUMM 1 35 0 1177 1282
R25 SUMM 3 31 1 1149 3752
R25 SUMM 3 32 2 1149 3773
R25 SUMM 3 33 1 1149 3783
R25 SUMM 3 34 0 1143 4182
R25 SUMM 3 35 0 1143 4249
```

I watched a few of the 1 show() tests and never saw any visual glitches. I can't absolutely guarantee it but I think they are free of errors.

A quick bit of graphing with awk shows this doesn't happen on one show() and only on 31 to 33 pixels with descending frequency.

```
$ awk '$2=="SUMM" { count[$3] += $5 } END { for (key in count) { print key, gensub(/ /, "x", "g", sprintf("%*s", count[key], "")) } }' r25-zhhd-run1.txt
1
2 xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
3 xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
```

```
$ awk '$2=="SUMM" { count[$4] += $5 } END { for (key in count) { print key, gensub(/ /, "x", "g", sprintf("%*s", count[key], "")) } }' r25-zhhd-run1.txt
31 xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
32 xxxxxxxxxxxxxxxxxxxxxxxxxx
33 xxxxxxxxx
34
35
```

kevinjwalters, 2025-07-21:
I'm still at the rubbing sticks together and not much happening stage of remote HW debugging but I just accidentally caught an `030` error on a recent compile of MicroPython master _with a few minor tweaks which are probably inconsequential_. It's running [neopixel-thirty-bug-timedetector-24.py](https://github.com/kevinjwalters/micropython-examples/blob/master/microbit/neopixel-thirty-bug-timedetector-24.py). I hit pause and this is the stack trace. Walking around it `dataReady` is 3, `tryPull(1)` is being executed and an interrupt has caught it in the assigment operator for `ManagedBuffer` which is very unfortunate as it calls `tryPull(1)` again. `samplesToSend` is 768 and `samplesSent` is 640.

The first call to `decr()` has `this = 0x20016730` with `refCount = 1`. The second `decr()` has `this = 0x5f1cf <codal::microbit_LEDMap_rowPins+2>` and `refCount = 4888` something is wrong here, `r0` is 8 which seems odd if `r0` is suppoesd to be `this`, `sp` is `0x2001fcc0` (in `main` it's 0x2001ffe8).

```
codal::microbit_panic@0x00020000 (/home/kevin/src/micropython-microbit-v2/lib/codal/libraries/codal-microbit-v2/source/MicroBitDevice.cpp:453)
target_panic@0x000200ce (/home/kevin/src/micropython-microbit-v2/lib/codal/libraries/codal-microbit-v2/source/MicroBitDevice.cpp:474)
isReadOnlyInline@0x00030c18 (/home/kevin/src/micropython-microbit-v2/lib/codal/libraries/codal-core/source/types/RefCounted.cpp:52)
codal::RefCounted::decr@0x00030c18 (/home/kevin/src/micropython-microbit-v2/lib/codal/libraries/codal-core/source/types/RefCounted.cpp:82)
codal::ManagedBuffer::operator=@0x0003079a (/home/kevin/src/micropython-microbit-v2/lib/codal/libraries/codal-core/source/types/ManagedBuffer.cpp:187)
codal::NRF52PWM::tryPull@0x0003207a (/home/kevin/src/micropython-microbit-v2/lib/codal/libraries/codal-nrf52/source/NRF52PWM.cpp:248)
codal::NRF52PWM::irq@0x00032182 (/home/kevin/src/micropython-microbit-v2/lib/codal/libraries/codal-nrf52/source/NRF52PWM.cpp:330)
codal::NRF52PWM::irq@0x00032182 (/home/kevin/src/micropython-microbit-v2/lib/codal/libraries/codal-nrf52/source/NRF52PWM.cpp:316)
nrf52_pwm2_irq@0x00032182 (/home/kevin/src/micropython-microbit-v2/lib/codal/libraries/codal-nrf52/source/NRF52PWM.cpp:28)
<signal handler called>@0xffffffe9 (Unknown Source:0)
codal::RefCounted::decr@0x00030c04 (/home/kevin/src/micropython-microbit-v2/lib/codal/libraries/codal-core/source/types/RefCounted.cpp:85)
codal::ManagedBuffer::operator=@0x0003079a (/home/kevin/src/micropython-microbit-v2/lib/codal/libraries/codal-core/source/types/ManagedBuffer.cpp:187)
codal::NRF52PWM::tryPull@0x0003207a (/home/kevin/src/micropython-microbit-v2/lib/codal/libraries/codal-nrf52/source/NRF52PWM.cpp:248)
codal::NRF52PWM::pullRequest@0x00032140 (/home/kevin/src/micropython-microbit-v2/lib/codal/libraries/codal-nrf52/source/NRF52PWM.cpp:299)
codal::NRF52PWM::pullRequest@0x00032140 (/home/kevin/src/micropython-microbit-v2/lib/codal/libraries/codal-nrf52/source/NRF52PWM.cpp:273)
codal::WS2812B::_play@0x000348c0 (/home/kevin/src/micropython-microbit-v2/lib/codal/libraries/codal-nrf52/source/WS2812B.cpp:185)
codal::WS2812B::play@0x000348c0 (/home/kevin/src/micropython-microbit-v2/lib/codal/libraries/codal-nrf52/source/WS2812B.cpp:160)
codal::neopixel_send_buffer@0x000343c6 (/home/kevin/src/micropython-microbit-v2/lib/codal/libraries/codal-nrf52/source/neopixel.cpp:27)
microbit_hal_pin_write_ws2812@0x0001ed4a (/home/kevin/src/micropython-microbit-v2/src/codal_app/microbithal.cpp:250)
microbit_ws2812_write@0x00054ed0 (/home/kevin/src/micropython-microbit-v2/src/codal_port/modmicrobit.c:94)
mp_execute_bytecode@0x00051c5a (/home/kevin/src/micropython-microbit-v2/lib/micropython/py/vm.c:957)
fun_bc_call@0x0004a8a4 (/home/kevin/src/micropython-microbit-v2/lib/micropython/py/objfun.c:267)
mp_execute_bytecode@0x00051be4 (/home/kevin/src/micropython-microbit-v2/lib/micropython/py/vm.c:1042)
fun_bc_call@0x0004a8a4 (/home/kevin/src/micropython-microbit-v2/lib/micropython/py/objfun.c:267)
microbit_pyexec_file@0x0003c1d6 (/home/kevin/src/micropython-microbit-v2/src/codal_port/main.c:147)
mp_main@0x0003c33a (/home/kevin/src/micropython-microbit-v2/src/codal_port/main.c:73)
main@0x0001e922 (/home/kevin/src/micropython-microbit-v2/src/codal_app/main.cpp:78)
```

kevinjwalters, 2025-07-23:
This is [neopixel-thirty-bug-timedetector-13.pp](https://github.com/kevinjwalters/micropython-examples/blob/master/microbit/neopixel-thirty-bug-timedetector-13.py) failing with dataReady exceeding 2, value of 3. This one is in the second called to `tryPull` in `pullRequest()` and the interrupt has fired in the middle of free operation.

```
codal::microbit_panic@0x00020008 (/home/kevin/src/micropython-microbit-v2/lib/codal/libraries/codal-microbit-v2/source/MicroBitDevice.cpp:456)
target_panic@0x000200ce (/home/kevin/src/micropython-microbit-v2/lib/codal/libraries/codal-microbit-v2/source/MicroBitDevice.cpp:474)
codal::NRF52PWM::pullRequest@0x00032142 (/home/kevin/src/micropython-microbit-v2/lib/codal/libraries/codal-nrf52/source/NRF52PWM.cpp:283)
codal::WS2812B::pull@0x00034880 (/home/kevin/src/micropython-microbit-v2/lib/codal/libraries/codal-nrf52/source/WS2812B.cpp:121)
codal::NRF52PWM::tryPull@0x0003205c (/home/kevin/src/micropython-microbit-v2/lib/codal/libraries/codal-nrf52/source/NRF52PWM.cpp:251)
codal::NRF52PWM::irq@0x000321b8 (/home/kevin/src/micropython-microbit-v2/lib/codal/libraries/codal-nrf52/source/NRF52PWM.cpp:331)
nrf52_pwm2_irq@0x000321b8 (/home/kevin/src/micropython-microbit-v2/lib/codal/libraries/codal-nrf52/source/NRF52PWM.cpp:28)
<signal handler called>@0xffffffe9 (Unknown Source:0)
device_free@0x0002aba0 (/home/kevin/src/micropython-microbit-v2/lib/codal/libraries/codal-core/source/core/CodalHeapAllocator.cpp:365)
codal::ManagedBuffer::operator=@0x0003079a (/home/kevin/src/micropython-microbit-v2/lib/codal/libraries/codal-core/source/types/ManagedBuffer.cpp:187)
codal::NRF52PWM::tryPull@0x00032064 (/home/kevin/src/micropython-microbit-v2/lib/codal/libraries/codal-nrf52/source/NRF52PWM.cpp:251)
codal::NRF52PWM::pullRequest@0x00032160 (/home/kevin/src/micropython-microbit-v2/lib/codal/libraries/codal-nrf52/source/NRF52PWM.cpp:308)
codal::WS2812B::_play@0x000348e0 (/home/kevin/src/micropython-microbit-v2/lib/codal/libraries/codal-nrf52/source/WS2812B.cpp:185)
codal::WS2812B::play@0x000348e0 (/home/kevin/src/micropython-microbit-v2/lib/codal/libraries/codal-nrf52/source/WS2812B.cpp:160)
codal::neopixel_send_buffer@0x000343e6 (/home/kevin/src/micropython-microbit-v2/lib/codal/libraries/codal-nrf52/source/neopixel.cpp:27)
microbit_hal_pin_write_ws2812@0x0001ed4a (/home/kevin/src/micropython-microbit-v2/src/codal_app/microbithal.cpp:250)
microbit_ws2812_write@0x00054ef0 (/home/kevin/src/micropython-microbit-v2/src/codal_port/modmicrobit.c:94)
mp_execute_bytecode@0x00051c7a (/home/kevin/src/micropython-microbit-v2/lib/micropython/py/vm.c:957)
fun_bc_call@0x0004a8c4 (/home/kevin/src/micropython-microbit-v2/lib/micropython/py/objfun.c:267)
mp_execute_bytecode@0x00051c04 (/home/kevin/src/micropython-microbit-v2/lib/micropython/py/vm.c:1042)
fun_bc_call@0x0004a8c4 (/home/kevin/src/micropython-microbit-v2/lib/micropython/py/objfun.c:267)
microbit_pyexec_file@0x0003c1f6 (/home/kevin/src/micropython-microbit-v2/src/codal_port/main.c:147)
mp_main@0x0003c35a (/home/kevin/src/micropython-microbit-v2/src/codal_port/main.c:73)
main@0x0001e922 (/home/kevin/src/micropython-microbit-v2/src/codal_app/main.cpp:78)
```

I think this all implies that `active` was `false` by the time pullRequest() was executed but somehow the interrupt for end of SEQ[0] has fired despite the line `PWM.TASKS_SEQSTART[0] = 1;` not being reached.

kevinjwalters, 2025-07-23:
This is [neopixel-thirty-bug-timedetector-13.pp](https://github.com/kevinjwalters/micropython-examples/blob/master/microbit/neopixel-thirty-bug-timedetector-13.py) failing with `dataReady` exceeding 2, value of 3. 

This doesn't make sense to me. `codal::WS2812B::_play` is at the `if (this->blockingPlayout)` implying preceeding `pullRequest()`  has executed (if optimisation at `-02` isn't misleading us). The interrupt has ended up indirectly calling `pullRequest()` which can only happen if `active` was `false` at the time . `enabled` and `active` (and `streaming` are `true`), repeatOnEmpty is `false`. `bufferPlaying=1`  `stopStreamingAfterBuf=0`. And `samplesToSend=1440` `samplesSent=512` `totalSamples=1540`. The `samplesSent` is equivalent to 4*128 value arrays worth of PWM data.

```
codal::microbit_panic@0x0001fffc (/home/kevin/src/micropython-microbit-v2/lib/codal/libraries/codal-microbit-v2/source/MicroBitDevice.cpp:453)
target_panic@0x000200ce (/home/kevin/src/micropython-microbit-v2/lib/codal/libraries/codal-microbit-v2/source/MicroBitDevice.cpp:474)
codal::NRF52PWM::pullRequest@0x00032142 (/home/kevin/src/micropython-microbit-v2/lib/codal/libraries/codal-nrf52/source/NRF52PWM.cpp:283)
codal::WS2812B::pull@0x00034880 (/home/kevin/src/micropython-microbit-v2/lib/codal/libraries/codal-nrf52/source/WS2812B.cpp:121)
codal::NRF52PWM::tryPull@0x0003205c (/home/kevin/src/micropython-microbit-v2/lib/codal/libraries/codal-nrf52/source/NRF52PWM.cpp:251)
codal::NRF52PWM::pullRequest@0x00032124 (/home/kevin/src/micropython-microbit-v2/lib/codal/libraries/codal-nrf52/source/NRF52PWM.cpp:301)
codal::WS2812B::pull@0x00034880 (/home/kevin/src/micropython-microbit-v2/lib/codal/libraries/codal-nrf52/source/WS2812B.cpp:121)
codal::NRF52PWM::tryPull@0x0003205c (/home/kevin/src/micropython-microbit-v2/lib/codal/libraries/codal-nrf52/source/NRF52PWM.cpp:251)
codal::NRF52PWM::irq@0x000321b8 (/home/kevin/src/micropython-microbit-v2/lib/codal/libraries/codal-nrf52/source/NRF52PWM.cpp:331)
nrf52_pwm2_irq@0x000321b8 (/home/kevin/src/micropython-microbit-v2/lib/codal/libraries/codal-nrf52/source/NRF52PWM.cpp:28)
<signal handler called>@0xffffffe9 (Unknown Source:0)
codal::WS2812B::_play@0x000348e0 (/home/kevin/src/micropython-microbit-v2/lib/codal/libraries/codal-nrf52/source/WS2812B.cpp:187)
codal::WS2812B::play@0x000348e0 (/home/kevin/src/micropython-microbit-v2/lib/codal/libraries/codal-nrf52/source/WS2812B.cpp:160)
codal::neopixel_send_buffer@0x000343e6 (/home/kevin/src/micropython-microbit-v2/lib/codal/libraries/codal-nrf52/source/neopixel.cpp:27)
microbit_hal_pin_write_ws2812@0x0001ed4a (/home/kevin/src/micropython-microbit-v2/src/codal_app/microbithal.cpp:250)
microbit_ws2812_write@0x00054ef0 (/home/kevin/src/micropython-microbit-v2/src/codal_port/modmicrobit.c:94)
mp_execute_bytecode@0x00051c7a (/home/kevin/src/micropython-microbit-v2/lib/micropython/py/vm.c:957)
fun_bc_call@0x0004a8c4 (/home/kevin/src/micropython-microbit-v2/lib/micropython/py/objfun.c:267)
mp_execute_bytecode@0x00051c04 (/home/kevin/src/micropython-microbit-v2/lib/micropython/py/vm.c:1042)
fun_bc_call@0x0004a8c4 (/home/kevin/src/micropython-microbit-v2/lib/micropython/py/objfun.c:267)
microbit_pyexec_file@0x0003c1f6 (/home/kevin/src/micropython-microbit-v2/src/codal_port/main.c:147)
mp_main@0x0003c35a (/home/kevin/src/micropython-microbit-v2/src/codal_port/main.c:73)
main@0x0001e922 (/home/kevin/src/micropython-microbit-v2/src/codal_app/main.cpp:78)
```

kevinjwalters, 2025-07-24:
This one is puzzling, I've got an 030 error from a basic heap free operation in a destructor using `RefCounted`.

```
codal::microbit_panic@0x00020004 (/home/kevin/src/micropython-microbit-v2/lib/codal/libraries/codal-microbit-v2/source/MicroBitDevice.cpp:453)
target_panic@0x000200d6 (/home/kevin/src/micropython-microbit-v2/lib/codal/libraries/codal-microbit-v2/source/MicroBitDevice.cpp:474)
device_free@0x0002abdc (/home/kevin/src/micropython-microbit-v2/lib/codal/libraries/codal-core/source/core/CodalHeapAllocator.cpp:370)
device_free@0x0002abdc (/home/kevin/src/micropython-microbit-v2/lib/codal/libraries/codal-core/source/core/CodalHeapAllocator.cpp:345)
codal::ManagedBuffer::~ManagedBuffer@0x0003078a (/home/kevin/src/micropython-microbit-v2/lib/codal/libraries/codal-core/source/types/ManagedBuffer.cpp:160)
codal::NRF52PWM::tryPull@0x00032106 (/home/kevin/src/micropython-microbit-v2/lib/codal/libraries/codal-nrf52/source/NRF52PWM.cpp:246)
codal::NRF52PWM::pullRequest@0x00032196 (/home/kevin/src/micropython-microbit-v2/lib/codal/libraries/codal-nrf52/source/NRF52PWM.cpp:290)
codal::NRF52PWM::pullRequest@0x00032196 (/home/kevin/src/micropython-microbit-v2/lib/codal/libraries/codal-nrf52/source/NRF52PWM.cpp:271)
codal::WS2812B::_play@0x00034960 (/home/kevin/src/micropython-microbit-v2/lib/codal/libraries/codal-nrf52/source/WS2812B.cpp:176)
codal::WS2812B::play@0x00034960 (/home/kevin/src/micropython-microbit-v2/lib/codal/libraries/codal-nrf52/source/WS2812B.cpp:153)
codal::neopixel_send_buffer@0x0003444e (/home/kevin/src/micropython-microbit-v2/lib/codal/libraries/codal-nrf52/source/neopixel.cpp:27)
microbit_hal_pin_write_ws2812@0x0001ed52 (/home/kevin/src/micropython-microbit-v2/src/codal_app/microbithal.cpp:250)
microbit_ws2812_write@0x00054f70 (/home/kevin/src/micropython-microbit-v2/src/codal_port/modmicrobit.c:94)
mp_execute_bytecode@0x00051cfa (/home/kevin/src/micropython-microbit-v2/lib/micropython/py/vm.c:957)
fun_bc_call@0x0004a944 (/home/kevin/src/micropython-microbit-v2/lib/micropython/py/objfun.c:267)
mp_execute_bytecode@0x00051c84 (/home/kevin/src/micropython-microbit-v2/lib/micropython/py/vm.c:1042)
fun_bc_call@0x0004a944 (/home/kevin/src/micropython-microbit-v2/lib/micropython/py/objfun.c:267)
microbit_pyexec_file@0x0003c276 (/home/kevin/src/micropython-microbit-v2/src/codal_port/main.c:147)
mp_main@0x0003c3da (/home/kevin/src/micropython-microbit-v2/src/codal_port/main.c:73)
main@0x0001e92a (/home/kevin/src/micropython-microbit-v2/src/codal_app/main.cpp:78)
```

The code has a check for `if (*cb == 0 || *cb & DEVICE_HEAP_BLOCK_FREE)` whcih looks like some sort of (32bit) attribute before the block.

The `samplesSent` value is `256` which strongly suggests two buffers have been loaded despite the `tryPull()` execution in stack trace being the first one in `pullRequest()`. Perhaps a spurious interrupt has fired here and messed up the buffers in some way.

kevinjwalters, 2025-07-28:
This version definitely improves things https://github.com/lancaster-university/codal-nrf52/tree/494ad55296629df36fb7ed8e60e712ea1b532dfb - about five hours at 60 pixels without any errors detected by timing method. The [neopixel-thirty-bug-timedetector-13.py](https://github.com/kevinjwalters/micropython-examples/blob/master/microbit/neopixel-thirty-bug-timedetector-13.py) changes pixel count on logo press, at 64 it does produce errors and these are real.

```
R13 ZIPCOUNT 60
R13 PXCT 61
R13 PXCT 62
R13 PXCT 63
R13 PXCT 64
R13 SLOW 2102
R13 SLOW 2124
R13 SLOW 2123
R13 PXCT 65
R13 PXCT 66
R13 PXCT 67
R13 PXCT 60
R13 PXCT 61
R13 PXCT 62
R13 PXCT 63
R13 PXCT 64
R13 SLOW 2125
```

That last one I caught in debugger and stats are

> p pwm.stats[3]
> $1 = {irqcount = 13, irq0 = 7, irq1 = 6, irqboth = 0, irq_times = {625800932, 625811170, 625821411, 625831745, 625841891, 625852130, 625862371, 625872610, 625882866, 625893090, 625903330, 625913571, 625923811, 123456 <repeats 27 times>}, irq_seqend = {1, 2, 1, 2, 1, 2, 1, 2, 1, 2, 1, 2, 1, -1 <repeats 27 times>}, irqinactive = 0, irqpoststop = 1, pwmstarts = 1, pwmstart_time = {625790698, 123456 <repeats 39 times>}, pwmstops = 1, preloadfails = 0, dataReadyAtStop = 0, nodata = 1, zero_time = 625777768}

---

> p pwm.stats[2]
> $2 = {irqcount = 13, irq0 = 7, irq1 = 6, irqboth = 0, irq_times = {625973880, 625983946, 625994189, 626004426, 626014666, 626024942, 626035146, 626045386, 626055626, 626065866, 626076106, 626086347, 626096586, 123456 <repeats 27 times>}, irq_seqend = {1, 2, 1, 2, 1, 2, 1, 2, 1, 2, 1, 2, 1, -1 <repeats 27 times>}, irqinactive = 0, irqpoststop = 1, pwmstarts = 1, pwmstart_time = {625963474, 123456 <repeats 39 times>}, pwmstops = 1, preloadfails = 0, dataReadyAtStop = 0, nodata = 1, zero_time = 625950544}

```
625950544 0 625950544 zero
625963474 12930 12930 pwmstart
625973880 23336 10406 irq
625983946 33402 10066 irq
625994189 43645 10243 irq
626004426 53882 10237 irq
626014666 64122 10240 irq
626024942 74398 10276 irq
626035146 84602 10204 irq
626045386 94842 10240 irq
626055626 105082 10240 irq
626065866 115322 10240 irq
626076106 125562 10240 irq
626086347 135803 10241 irq
626096586 146042 10239 irq
```

----

> p pwm.stats[1]
> $3 = {irqcount = 13, irq0 = 7, irq1 = 7, irqboth = 0, irq_times = {626144848, 626155086, 626165327, 626175681, 626185806, 626196046, 626206286, 626216526, 626226766, 626237175, 626247251, 626257487, 626268168, 123456 <repeats 27 times>}, irq_seqend = {1, 2, 1, 2, 1, 2, 1, 2, 1, 2, 1, 2, 1, -1 <repeats 27 times>}, irqinactive = 0, irqpoststop = 1, pwmstarts = 1, pwmstart_time = {626134614, 123456 <repeats 39 times>}, pwmstops = 1, preloadfails = 0, dataReadyAtStop = 0, nodata = 1, zero_time = 626121684}

The third one has `irq1` of 7 rather than 6 compared to prior ones. The irq_seqend appears inconsistent with this value but it indicates that SEQ[0] had ended when the interrupt started but while it was running SEQ[1] ended too.

```
626121684 0 626121684 zero
626134614 12930 12930 pwmstart
626144848 23164 10234 irq
626155086 33402 10238 irq
626165327 43643 10241 irq
626175681 53997 10354 irq
626185806 64122 10125 irq
626196046 74362 10240 irq
626206286 84602 10240 irq
626216526 94842 10240 irq
626226766 105082 10240 irq
626237175 115491 10409 irq
626247251 125567 10076 irq
626257487 135803 10236 irq
626268168 146484 10681 irq
```

---

> p pwm.stats[0]
> $4 = {irqcount = 11, irq0 = 7, irq1 = 6, irqboth = 1, irq_times = {626305342, 626319652, 626336082, 626346323, 626356563, 626366803, 626377045, 626387283, 626397522, 626407763, 626418003, 123456 <repeats 29 times>}, irq_seqend = {3, 1, 1, 2, 1, 2, 1, 2, 1, 2, 1, -1 <repeats 29 times>}, irqinactive = 0, irqpoststop = 1, pwmstarts = 1, pwmstart_time = {626333787, 123456 <repeats 39 times>}, pwmstops = 1, preloadfails = 0, dataReadyAtStop = 0, nodata = 1, zero_time = 626292376}

Four fouth one the timings are put in order below with some relative colums added. For the fourth the interrupts are running before PWM is started. This one takes 41411 cycles to get to PWM startup whereas this normally takes 12930.

```
626292376 0 626292376 zero
626305342 12966 12966 irq (SEQ[0] + SEQ[1])
626319652 27276 14310 irq (SEQ[0])
626333787 41411 14135 pwmstart
626336082 43706 2295 irq (SEQ[0])
626346323 53947 10241 irq (SEQ[1])
626356563 64187 10240 irq
626366803 74427 10240 irq
626377045 84669 10242 irq
626387283 94907 10238 irq
626397522 105146 10239 irq
626407763 115387 10241 irq
626418003 125627 10240 irq
```

finneyj, 2025-07-29:
Cor - great catch @kevinjwalters!

Let us know if that fixes the problem longer term!!

kevinjwalters, 2025-07-31:
@finneyj I think it's getting better but I went back to [neopixel-thirty-bug-timedetector-24.py](https://github.com/kevinjwalters/micropython-examples/blob/master/microbit/neopixel-thirty-bug-timedetector-24.py) and that still picks up problems which looking at (cycle) timing from two examples below are an irq firing 260 cycles before pwmstart timestamp. I suspect the 260 cycles is just the code running and it means that as soon as interrupts are enabled a bogus one fires.

```
  4095962766 0 4095962766 zero
  4095969503 6737 6737 irq
  4095969758 6992 255 pwmstart
  4095979700 16934 9942 irq
```

```
  891387683 0 891387683 zero
  891394941 7258 7258 irq
  891395204 7521 263 pwmstart
  891405140 17457 9936 irq
```

The code performs better now as I've removed the pointless buffer initialisation and reuse the the buffers. The former can be seen in terms of 6992 and 7521 cycles vs previous 13000-14000.

kevinjwalters, 2025-08-01:
There's still an issue lurking here with one or two extra pulses appearing after a time period which appears to depend on the number of pixels. Here's 4 and 5 pixels with misbehaving output.

Note: the x scale varies bere.

#### 4 pixels

[image]

[image]

#### 5 pixels

[image]

[image]

kevinjwalters, 2025-08-04:
This is `DMESG` output which appears if there's an issue but with emptyBuffer down at 4 samples to provoke strange effects. The code is slightly modified [neopixel-thirty-bug-timedetector-24.py](https://github.com/kevinjwalters/micropython-examples/blob/master/microbit/neopixel-thirty-bug-timedetector-24.py) outputting 4 pixels.

```
R24 TOOS 117966 vs 1170
NRF52PWM anomalyCheckStart(2) issues=1
stats[3]
irqcount=2 irq0=1 irq1=1 irqboth=0
irqinactive=0 irqprestart=0 irqpoststop=1 pwmstarts= pwmstops=1
trypullcnt=4 preloadfails=0 dataReadyAtStop=0 nodata=1
TS -886836965 zero
TS -886830316 pwmstart
TS -886820079 irq 1
TS -886809843 irq 2
TPDUR 3211
TPDUR 3118
TPDUR 68
TPDUR 273

stats[2]
irqcount=2 irq0=1 irq1=1 irqboth=0
irqinactive=0 irqprestart=0 irqpoststop=1 pwmstarts= pwmstops=1
trypullcnt=4 preloadfails=0 dataReadyAtStop=0 nodata=1
TS -886769949 zero
TS -886763300 pwmstart
TS -886753063 irq 1
TS -886742827 irq 2
TPDUR 3211
TPDUR 3118
TPDUR 68
TPDUR 269

stats[1]
irqcount=2 irq0=1 irq1=1 irqboth=0
irqinactive=0 irqprestart=0 irqpoststop=1 pwmstarts= pwmstops=1
trypullcnt=4 preloadfails=0 dataReadyAtStop=0 nodata=1
TS -886700793 zero
TS -886694144 pwmstart
TS -886683907 irq 1
TS -886673673 irq 2
TPDUR 3211
TPDUR 3118
TPDUR 68
TPDUR 275

stats[0]
irqcount=2 irq0=2 irq1=1 irqboth=1
irqinactive=0 irqprestart=0 irqpoststop=1 pwmstarts= pwmstops=1
trypullcnt=5 preloadfails=0 dataReadyAtStop=0 nodata=1
TS -886634293 zero
TS -886627644 pwmstart
TS -886617407 irq 1
TS -886606534 irq 3
TPDUR 3211
TPDUR 3118
TPDUR 68
TPDUR 268
TPDUR 25
```

The stats[0] might be the one that set the fifth pixel to green. The negatve TS (timestamp) values are due to DMESG not implementing `%u` just `%d`. Here PWM start happens 6649 cycles after zero which is normal. The first interrupt is 10237 which is spot on. The second one is 10873 later which equates to 135.9125 samples, 128 + 7.91250. The TPDUR are cycles measurement of each tryPull, this is clearly atypical for making five calls rather than four.

kevinjwalters, 2025-08-05:
I went through some interrupt priority level changes. All tests are without the debugger attached as this sometimes affects results. This is using [neopixel-thirty-bug-timedetector-24.py](https://github.com/kevinjwalters/micropython-examples/blob/master/microbit/neopixel-thirty-bug-timedetector-24.py) set to start from 4 pixels and running for 2 minutes per pixel value rather than 1.

Code has emptyBuffer enlarged to 12 from 8 and my fixes so far. Without changing interrupt priorities I get this sort of output from my anomaly reporting code at least once a second indicating two interrupts are occuring but three events. (I don't understand the CYCCNT based interval numbers, they should all be > 10240, puzzling.)
 
```
stats[0]
irqcount=2 irq0=2 irq1=1 irqboth=1
irqinactive=0 irqprestart=0 irqpoststop=1 pwmstarts=1 pwmstops=1
trypullcnt=5 preloadfails=0 dataReadyAtStop=0 nodata=1
TS 1893282998 zero
TS 1893289569 pwmstart fromzero 6571
TS 1893297546 interval 7977 irq 1
TS 1893304444 interval 6898 irq 3
TPDUR 2933
TPDUR 3327
TPDUR 71
TPDUR 281
TPDUR 25
```

---

#### PWM at priority 0

Works fine.

```
MPY: soft reboot
R24 SVMP 3.4.0; MicroPython 096ff8b9e on 2025-08-05
R24 KZHH
R24 MATH 437817
R24 SUMM 4 0 115348 514
R24 SUMM 5 0 115047 517
R24 SUMM 6 0 114683 519
R24 SUMM 7 0 99547 668
```

#### PWM at priority 1

works fine

```
   20015080, sink: 0x20015170
relocate vtor to 0x0 -> 0x20002200 0x20002200
R24 SVMP 3.4.0; MicroPython 096ff8b9e on 2025-08-05
R24 KZHH
R24 MATH 437478
R24 SUMM 4 0 115333 514
R24 SUMM 5 0 115070 517
R24 SUMM 6 0 114693 519
R24 SUMM 7 0 99571 667
R24 SUMM 8 0 99247 670
```

#### PWM at priority 2

works fine.

```
   20015080, sink: 0x20015170
relocate vtor to 0x0 -> 0x20002200 0x20002200
R24 SVMP 3.4.0; MicroPython 096ff8b9e on 2025-08-05
R24 KZHH
R24 MATH 437478
R24 SUMM 4 0 115338 514
R24 SUMM 5 0 115080 526
R24 SUMM 6 0 114691 519
R24 SUMM 7 0 99573 667
R24 SUMM 8 0 99250 670
R24 SUMM 9 0 98981 671
R24 SUMM 10 0 98989 671
R24 SUMM 27 0 70717 1105
R24 SUMM 28 0 64733 1249
R24 SUMM 29 0 64733 1249
R24 SUMM 30 0 64731 1249
R24 SUMM 31 0 64734 1249
R24 SUMM 32 0 64729 1249
```

#### PWM back at priority 3

```
R24 SVMP 3.4.0; MicroPython 096ff8b9e on 2025-08-05
R24 KZHH
R24 MATH 437798
NRF52PWM anomalyCheckStart(2) issues=1
stats[3]
irqcount=2 irq0=1 irq1=1 irqboth=0
irqinactive=0 irqprestart=0 irqpoststop=1 pwmstarts=1 pwmstops=1
trypullcnt=4 preloadfails=0 dataReadyAtStop=0 nodata=1
TS 1346255436 zero
TS 1346262104 pwmstart fromzero 6668
TS 1346267228 irq 1
TS 1346273468 irq 2
TPDUR 3213
TPDUR 3124
TPDUR 73
TPDUR 283

stats[2]
irqcount=2 irq0=1 irq1=1 irqboth=0
irqinactive=0 irqprestart=0 irqpoststop=1 pwmstarts=1 pwmstops=1
trypullcnt=4 preloadfails=0 dataReadyAtStop=0 nodata=1
TS 1346314104 zero
TS 1346320772 pwmstart fromzero 6668
TS 1346328368 irq 1
TS 1346334608 irq 2
TPDUR 3213
TPDUR 3124
TPDUR 73
TPDUR 283

stats[1]
irqcount=2 irq0=1 irq1=1 irqboth=0
irqinactive=0 irqprestart=0 irqpoststop=1 pwmstarts=1 pwmstops=1
trypullcnt=4 preloadfails=0 dataReadyAtStop=0 nodata=1
TS 1346376892 zero
TS 1346383560 pwmstart fromzero 6668
TS 1346391288 irq 1
TS 1346398204 irq 2
TPDUR 3213
TPDUR 3124
TPDUR 73
TPDUR 280

stats[0]
irqcount=2 irq0=2 irq1=1 irqboth=1
irqinactive=0 irqprestart=0 irqpoststop=1 pwmstarts=1 pwmstops=1
trypullcnt=5 preloadfails=0 dataReadyAtStop=0 nodata=1
TS 1346435004 zero
TS 1346441672 pwmstart fromzero 6668
TS 1346449368 interval 7696 irq 1
TS 1346456257 interval 6889 irq 3
TPDUR 3213
TPDUR 3124
TPDUR 73
TPDUR 279
TPDUR 25
```

#### PWM priority 3 TIMER4_IRQn at 4

TIMER4_IRQn (Display and Light Sensing) set to 4, spews errors.

#### PWM priority 3 TIMER3_IRQn at 4

TIMER3_IRQn (capacitive touch) set to 4.

```
20015080, sink: 0x20015170
relocate vtor to 0x0 -> 0x20002200 0x20002200
R24 SVMP 3.4.0; MicroPython 096ff8b9e on 2025-08-05
R24 KZHH
R24 MATH 437822
R24 SUMM 4 0 115334 514
R24 SUMM 5 0 115077 517
R24 SUMM 6 0 114698 519
R24 SUMM 7 0 99571 667
R24 SUMM 8 0 99255 670
R24 SUMM 9 0 98998 671
R24 SUMM 10 0 99003 671
R24 SUMM 27 0 70713 1105
R24 SUMM 28 0 64728 1249
R24 SUMM 29 0 64728 1249
R24 SUMM 30 0 64728 1249
```

All is fine. Surprising, as I thought nRF52 has a capacitive sensing hardware feature and not much would be going on here. I had a quick glance at the code and I think it's got a frequent interrupt at 16kHz but doesn't do much. I note it _disables all interrupts briefly_ - why does it need to do that?