lancaster-university/codal-microbit-v2 #477: audio refactor: StreamRecording::play() into SerialStreamer dumps the data but then hangs

martinwork, 2025-04-08:
@finneyj @carlosperate 

Test program - check serial output
[MICROBIT.zip](https://github.com/user-attachments/files/19657452/MICROBIT.zip)

RESET then press B - recording works; playAsync works, except synchronously; play() works but hangs at the end

Everything happens inside the playAsync call to pullRequest, so StreamRecording notifies playLock, before playAsync exits and playLock.wait is called. The sample uses playAsync, which works like play should!

Trivial extra observation... StreamRecording::play() always sends an empty buffer after the recorded buffers.

Built as described in [playAsync()](https://github.com/lancaster-university/codal-microbit-v2/issues/446#issuecomment-2786529072)

With sample in
https://github.com/martinwork/microbit-v2-samples/blob/mw-test-audio-refactor/source/test-audio-refactor.cpp
```
#include "MicroBit.h"
#include "StreamRecording.h"
#include "SerialStreamer.h"

MicroBit uBit;

void forever()
{
    SplitterChannel *splitterChannel  = uBit.audio.splitter->createChannel();
    StreamRecording *recording        = new StreamRecording(*splitterChannel);
    SerialStreamer  *streamer         = new SerialStreamer(*recording, SERIAL_STREAM_MODE_DECIMAL);

    while (true)
    {
        uBit.display.image.setPixelValue( 0, 0, uBit.display.image.getPixelValue( 0, 0) ? 0 : 255);

        if ( uBit.buttonB.isPressed())
        {
            uBit.display.print("R");
            DMESG("PAUSE");
            uBit.sleep(1000);
            DMESG("RECORD");

            // on demand activateMic doesn't work after an analogue read of a pin
            // calling activateMic in advance stops it hanging
            //uBit.audio.activateMic();
            //uBit.audio.mic->setStartDelay(4);

            recording->recordAsync();
            while ( recording->isRecording() && recording->duration( recording->getSampleRate()) < 0.1)
            {
                uBit.display.image.setPixelValue( 4, 0, uBit.display.image.getPixelValue( 4, 0) ? 0 : 255);
                uBit.sleep(0);
            }
            recording->stop();

            DMESG("RECORD done");
            uBit.display.clear();

            DMESG("STREAM");
            uBit.sleep(100);
            recording->connect(*streamer);
            DMESG("playAsync");
            recording->playAsync();   // works like sync
            DMESG("playAsync done");
            uBit.sleep(500);
            DMESG("play");
            uBit.sleep(100);
            recording->play();         // play() hangs
            DMESG("STREAM done");
        }

        if ( uBit.buttonA.isPressed())
        {
            uBit.display.print("A");
            int analog0 = uBit.io.P2.getAnalogValue();
            DMESG("analog0 %d", (int) analog0);
            uBit.sleep(100);
            uBit.display.clear();
        }

        uBit.sleep(100);
    }
}

int main() {
    uBit.init();
    create_fiber( forever);
    release_fiber();
    return 0;
}

```

finneyj, 2025-05-12:
Thanks @martinwork. I think this is entirely because SerialStreamer never actually yields control of the CPU... We'd need a new implementation of the class to deal with bulk transmission... which I was trying to avoid. I don't think we can fix the behaviour that make sync look like async in this case, simply because that's a feature of the implementation...

The hang caused by the play() call is a good catch though! That's simply due to the order of the notifyAll() call occurring before the thread can wait() on the lock. I've put a workaround in here: https://github.com/lancaster-university/codal-core/commit/443ee1b1f0d41b9119d4f6a2de48ed22ede65812

martinwork, 2025-05-13:
Thanks @finneyj