lancaster-university/codal-microbit-v2 #478: audio refactor: StreamRecording may pull a buffer before activateMic is called

martinwork, 2025-04-13:
@finneyj @carlosperate I saw this before, but couldn't figure out how it happened!

Related to https://github.com/lancaster-university/codal-microbit-v2/issues/476

Test program - check serial output
[beforeActivate.zip](https://github.com/user-attachments/files/19725607/beforeActivate.zip)

RESET, then press B 

After initialisation, the ADC mic channel is flagged as not enabled, bit left active in the ADC, and the mic pin is switched off. 
 
Starting analogue pin reading after starting an async recording, generates a mic buffer as the ADC is stopped to add the pin, because the mic channel has been marked as enabled by dataWanted. StreamRecording pulls this buffer before activateMic is called.

It doesn't happen a second time, even if the pin channel is released, because releasing it restarts the ADC while the mic channel is marked disabled, so the mic is no longer active in the ADC for the second run.

```
disable                                                                              on start, mic channel is flagged disabled but still active in ADC
MicroBitAudio::periodicCallback: deactivateMic()...
deactivateMic                                                                    mic pin switched off
PAUSE                                                                                button B pressed
RECORD
enable                                                                                 mic channel flagged enabled
ADC: STOPPING                                                                  restart ADC to add pin
demux 5                                                                               mic channel buffer generated
ADC: STARTING
enabled 1                                                                             mic channel is enabled
analog 199                                                                           pin value returned
rec   5                                                                                   StreamRecording pulls the buffer 
MicroBitAudio::periodicCallback: activateMic()...
activateMic                                                                           mic is switched on
delay 1                                                                                  warm up delay
analog 204
demux 6
rec   6
```

Built as described in https://github.com/lancaster-university/codal-microbit-v2/issues/446#issuecomment-2786529072

With sample in
https://github.com/martinwork/microbit-v2-samples/blob/mw-test-audio-refactor/source/test-audio-refactor.cpp

```
#include "MicroBit.h"
#include "StreamRecording.h"
#include "SerialStreamer.h"

MicroBit uBit;

void forever()
{
    SplitterChannel *splitterChannel  = uBit.audio.splitter->createChannel();
    StreamRecording *recording        = new StreamRecording(*splitterChannel);
    SerialStreamer  *streamer         = new SerialStreamer(*recording, SERIAL_STREAM_MODE_DECIMAL);

    while (true)
    {
        uBit.display.image.setPixelValue( 0, 0, uBit.display.image.getPixelValue( 0, 0) ? 0 : 255);
        uBit.sleep(100);

        if ( uBit.buttonB.isPressed())
        {
            uBit.display.print("B");
            DMESG("PAUSE");
            uBit.sleep(1000);
            uBit.display.print("R");
            DMESG("RECORD");

            recording->recordAsync();
            while ( recording->isRecording() && recording->duration( recording->getSampleRate()) < 0.1)
            {
                int analog = uBit.io.P2.getAnalogValue();
                DMESG("analog %d", (int) analog);

                uBit.display.image.setPixelValue( 4, 0, uBit.display.image.getPixelValue( 4, 0) ? 0 : 255);
                uBit.sleep(10);
            }
            recording->stop();
            uBit.adc.releaseChannel( uBit.io.P2);

            DMESG("RECORD done");
            uBit.display.clear();
        }
    }
}

int main() {
    uBit.init();
    create_fiber( forever);
    release_fiber();
    return 0;
}

```

finneyj, 2025-05-12:
Thanks @martinwork. I changed the calls to activateMic() in the dataWanted() function.... So it's possible we've now already addressed this issue in this patch: https://github.com/lancaster-university/codal-nrf52/commit/b0d6b49413ff737e1bb915e8b846afc82a96a60a

Is it easy for you to test to see if this is still happening? Or is this bug a real corner case such that it only surfaces very infrequently and is hard to reproduce?

martinwork, 2025-05-16:
I think it's still happening with the code from 4 days ago. I'll update and test again. It's only one buffer once and unlikely to be a problem in practise.

finneyj, 2025-05-16:
Thanks. I think I fixed it yesterday.  :)

martinwork, 2025-05-16:
I think it's still happening.

Adding a new ADC channel in the few ms between starting recording (which enables the mic channel) and the audio periodicCallback being called to activate the mic pin, causes an ADC stop/start, and a mic buffer is demuxed. 

It can be avoided by reading the pin before starting to record.

finneyj, 2025-05-16:
OK - thanks @martinwork . I'll take a look.

finneyj, 2025-05-16:
OK - I've added a filter into the StreamSplitter class that should filter out unwanted data from entering the chain. It does seem to trigger at roughly the right point, so could be worth a test after I address #485.

microbit-carlos, 2025-05-21:
@martinwork as #485 has been resolved, have you had a chance to check if this one again?