lancaster-university/codal-microbit-v2 #479: audio refactor: `SerialStreamer` only pulls 768 bytes when pairing mode is disabled.

microbit-carlos, 2025-04-25:
Might be related to:
- https://github.com/lancaster-university/codal-microbit-v2/issues/477

A very simple `SerialStreamer` programme with the default `microbit-v2-samples` `codal.json` file, pointing to the `audio-refactor` branch, only sends 768 bytes to serial and hangs there.

When pairing mode is enabled it works as expected.

[MICROBIT-binary-hangs.hex.zip](https://github.com/user-attachments/files/19906046/MICROBIT-binary-hangs.hex.zip)

```cpp
#include "MicroBit.h"
#include "SerialStreamer.h"

MicroBit uBit;

int main() {
    uBit.init();

    static SplitterChannel *splitterChannel = uBit.audio.splitter->createChannel();
    SerialStreamer *streamer = new SerialStreamer(*splitterChannel, SERIAL_STREAM_MODE_BINARY);
    uBit.audio.requestActivation();
    uBit.audio.activateMic();

    while (true) {
        uBit.sleep(1000);
    }
}
```

```json
{
    "target": {
        "name": "codal-microbit-v2",
        "url": "https://github.com/lancaster-university/codal-microbit-v2",
        "branch": "audio-refactor",
        "type": "git"
    },
    "config":{
        "MICROBIT_BLE_ENABLED" : 0,
        "MICROBIT_BLE_PAIRING_MODE": 0
    }
}
```

Changing `MICROBIT_BLE_PAIRING_MODE` to `1` seems to fix it.

Also, in case its useful, changing `SERIAL_STREAM_MODE_BINARY` to `SERIAL_STREAM_MODE_DECIMAL` with pairing mode disabled, and it sends 1280 samples before hanging.

martinwork, 2025-04-25:
It seems to be the sleep that makes all the difference.
https://github.com/lancaster-university/codal-microbit-v2/blob/d159aba47e1797ace83ffbac81009909326f3c4f/model/MicroBit.cpp#L237

martinwork, 2025-04-25:
MicroBitAudio::periodicCallback deactivates the mic because the mic ADC is not marked enabled. The sleep determines whether that happens before or after the activateMic in the test program.

The ADC channel is disabled in NRF52ADCChannel::dataWanted called originally in LevelDetectorSPL::periodicCallback
https://github.com/lancaster-university/codal-core/blob/5c0d922e5cd85ff034d57c76e4f61dc0014b42b1/source/streams/LevelDetectorSPL.cpp#L84

Perhaps the test code needs to call dataWanted instead of activateMic, and let the streams do their magic
```
    //uBit.audio.activateMic();
    splitterChannel->dataWanted( DATASTREAM_WANTED);
```
Then the splitter knows we want data on our channel, regardless of what LevelDetectorSPL wants.

microbit-carlos, 2025-04-25:
Is that due to a change in the `audio-refactor` branch?

If that's the case, is that something done by design? Or is calling `dataWanted(DATASTREAM_WANTED)` more like a workaround?

If it's by design, should all DataSink classes this implemented internally somehow?

martinwork, 2025-04-25:
I just realised... calling activateMic directly ignores the audio-refactor dataWanted logic and falls foul of the timing issue. 

Maybe SerialStreamer could have start and stop methods that call upstream.dataWanted, but that would not work with StreamRecording. Perhaps calling splitterChannel->dataWanted is a workaround for that?

**dataWanted** appeared in the audio-refactor branch, in DataSource, DataSourceSink, SplitterChannel, StreamSplitter and NRF52ADCChannel.

I think a downstream (endpoint?) component now calls dataWanted to tell the upstream whether it wants data or not, which is somehow the reverse of the old system.

For example, StreamRecording tells its splitter channel it wants to start receiving data in recordAsync(), or doesn't want data any more in stop(), which gets passed upstream to rawsplitter, which possibly calls dataWanted on the mic ADC channel, based on whether any stream wants data.
```
mic - rawsplitter - channel - LevelDetectorSPL
mic - rawsplitter - normalizer - splitter - channel - StreamRecording
mic - rawsplitter - normalizer - splitter - channel - SerialStreamer 
```
MicroBitAudio::periodicCallback() eventually enables or disables the mic hardware depending on whether the mic ADC channel is enabled

microbit-carlos, 2025-04-29:
Thanks Martin!

It looks like by design `SerialStreamer` should call `dataWanted()` from its DataSource.
@finneyj I guess it makes sense as well to move `SerialStreamer` from microbit-v2-samples to codal-core or codal-microbit-v2? To ensure any future refactors or bug fixes in  this class also get picked up.
We can then add a simple example project to `microbit-v2-samples` using it.

finneyj, 2025-05-12:
Yes - agree folks. Classes should not really be calling activateMic(), but instead just start using it. The presence of a live DataSink in the audio pipeline alone should be sufficient to turn on the audio....

I agree, probably a good time to move SerialStreamer! I'll migrate it now.

finneyj, 2025-05-12:
Done here: https://github.com/lancaster-university/codal-core/commit/36d69818b43f6875083c5c9f87fcb65da79cabc2

finneyj, 2025-05-13:
ahh, good catch @microbit-carlos. Yes, I've now put the call to dataWanted() into SerialStreamer. For future reference, all you really need now is this:

```
int main() {
    uBit.init();

    static SplitterChannel *splitterChannel = uBit.audio.splitter->createChannel();
    SerialStreamer *streamer = new SerialStreamer(*splitterChannel, SERIAL_STREAM_MODE_BINARY);

    while (true) {
        uBit.sleep(1000);
    }
}
```

To answer your other question about defaultSerial and the Serial class, we're just keeping track of the first instance of a Serial port created, so we can use it as a default when needed. SerialStreamer uses this so that (by default) we don't need to change the API (it historically had a static reference to a uBit object). It now uses the defaultSerial port unless otherwise configured by an optional parameter in the constructor, or a mutator.

Keeping debug channels simple is always a good idea! :)

microbit-carlos, 2025-05-13:
Thanks Joe!
I've updated the microbit-v2-samples `audio-refactor` branch to remove the SerialStreamer there:
- https://github.com/lancaster-university/microbit-v2-samples/commit/35bae350ab413affc26544fd5ecfbcb5ed3256e9

closed 2025-05-13 by microbit-carlos