lancaster-university/codal-microbit-v2 #482: audio refactor: A way to set up the microphone sampling rate

microbit-carlos, 2025-05-13:
@finneyj I believe this still feature still needs to be implemented in the audio refactor branch. This would probably be higher priority than the other bug fixes and it's blocking further testing.

finneyj, 2025-05-13:
Absolutely - thanks @microbit-carlos

microbit-carlos, 2025-05-13:
Thanks Joe!
Once you get a chance to look into this it'd be great if you could update the OOB file in the samples repo `audio-refactor` branch as well:
https://github.com/lancaster-university/microbit-v2-samples/blob/audio-refactor/source/samples/OOB_v3.cpp#L215-L216

finneyj, 2025-05-15:
Applied in the commits cited here:
https://github.com/lancaster-university/codal-microbit-v2/issues/474

I'll take a look at the OOB

finneyj, 2025-05-15:
Hmmm... just looking at the OOB, its setting the sample rate to 11000. Which is now the default anyway. So possibly nothing to do there?

microbit-carlos, 2025-05-15:
Ah, yeah, that's okay for the OOB then. If you could update the other `requestSampleRate()` functions from https://github.com/lancaster-university/microbit-v2-samples/blob/audio-refactor/source/samples/StreamAPITest.cpp that would be great.

finneyj, 2025-05-16:
ok - done. The sample code in there seems to work ok.

I'm not quite sure of the wisdom of changing the sample rate in there samples though, rather than changing the playout rate? The latter is more precise, simpler on the hardware and possibly simpler for the student to understand?

microbit-carlos, 2025-05-16:
Part of the learning goals for the recording and playback feature was to be able change the microphone sampling rate and be able to tell the difference in sound quality and memory consumption.

finneyj, 2025-05-16:
OK!

microbit-carlos, 2025-05-16:
Great, I run a simple test and seems to work well, thanks Joe!

closed 2025-05-16 by microbit-carlos