lancaster-university/codal-microbit-v2 #484: audio refactor: Should there be a min and/or max mic sampling rate values

microbit-carlos, 2025-05-16:
Should a min and max values be enforced by CODAL?
The MakeCode block slider goes fro 1K to 22K

[image]
[image]

MicroPython lets you add an arbitrary number, but using very low or very high values like 1 or 100K samples per second does not make much sense. Do we just leave it as it is and give the user freedom to request any value?

🐞 For example, right now setting `uBit.audio.mic->setSampleRate(0)` halt the programme (display still runs, but programme stops running, e.g. `display.print("A"); mic->setSampleRate(0); display.print("B"); ` permanently displays "B").

We can protect against zero, but that raises the question if we should set a range.

finneyj, 2025-05-16:
Good question. 

I don't see a real reason to prevent "extremes" if they work correctly with the default configuration. Maybe we should test what is stable, then set the bounds accordingly. 

e.g. I'd be a bit miffed it I was prevent from using, say, 50k samples per second if that's what I needed to do to detect the same thing as a dog or similar?

On the flip side I would be at all bothered if I asked for 1 bit per second but actually got 8. :)

finneyj, 2025-05-16:
Hehe... I just set the sample rate to '1' and ran @martinwork's sample code.  It munched away on it for about 5 minutes then gave me a single 256 byte buffer. :D

At the other end of the spectrum, I set the samplerate to 100000 and it gave me back a set of samples pretty quickly. It technically also worked right up to 1000000 samples/sec.... but I imagine the data isn't very useful up there!

I'm thinking a bound between 1...1000000, which I've implemented here:
https://github.com/lancaster-university/codal-nrf52/commit/03370108dec7310eed9a2cf0b072d869bec3b353

microbit-carlos, 2025-05-21:
Awesome, thanks Joe!