lancaster-university/codal-microbit-v2 #485: audio refactor: Attaching new SerialStreamer to StreamRecording turns on the mic

martinwork, 2025-05-16:
SerialStreamer calls StreamRecording::dataWanted which I think is in DataSourceSink, which passes the message upstream.

<details>
<summary>code</summary>

```
#include "MicroBit.h"
#include "StreamRecording.h"
#include "SerialStreamer.h"

MicroBit uBit;

void forever()
{
    SplitterChannel *splitterChannel  = uBit.audio.splitter->createChannel();
    StreamRecording *recording        = new StreamRecording(*splitterChannel);
    SerialStreamer  *streamer         = new SerialStreamer(*recording, SERIAL_STREAM_MODE_DECIMAL);

    while (true)
    {
        uBit.display.image.setPixelValue( 0, 0, uBit.display.image.getPixelValue( 0, 0) ? 0 : 255);
        uBit.sleep(100);
    }
}

int main() {
    uBit.init();
    create_fiber( forever);
    release_fiber();
    return 0;
}

```
</details>

finneyj, 2025-05-16:
Hmm.. so what behaviour do we want here?

I guess the mic should stay disabled until Stream Recording actively starts recording?
In which case it will probably need an override of the inherited behaviour....

martinwork, 2025-05-16:
Yes, I'm not sure. The ADC mic channel is also sending buffers, which StreamRecording ignores. 

Perhaps the dataWanted call from SerialStreamer or any other downstream simply doesn't apply to StreamRecording?

It's not set up to record and play at the same time. 

Is the problem down to StreamRecording's triple role: record, store and play?

finneyj, 2025-05-16:
right - it's a tricky beast! Exhibits a kindof of temporal polymorphism! :D

cool, let me think on it...

finneyj, 2025-05-16:
OK - fixed here I think!  :  https://github.com/lancaster-university/codal-core/commit/e659bf3a19716453dc75a0090ad00f89bb2bcb16

finneyj, 2025-05-16:
ahh, brilliant!

finneyj, 2025-05-16:
p.s. I forgot to mention to you a slight change in the NRF52ADC API @martinwork. I took the opportunity yesterday to scratch an itch and make the releaseChannel() API to be symmetric to activateChannel(). 

You used the old releaseChannel(Pin) in your sample, which you now need to use releasePin(Pin) instead.

finneyj, 2025-05-16:
Heh. I've had a lot of issues on the go this last few days - they tend to blue together. 🙂

closed 2025-05-21 by microbit-carlos