lancaster-university/codal-microbit-v2 #487: After mic has been allowed to deactivate, the next sound level is wrong

martinwork, 2025-06-04:
MakeCode test: https://makecode.microbit.org/_gLVCauHqVc8C (created in v6)
- Use a steady sound source through a headphone resting on mic: e.g. https://www.youtube.com/watch?v=xGXYFJmvIvk
- Wave a torch to collect a batch of values
- Adjust volume through headphone
- In Makecode beta, the next batch starts with a memory of the previous level

martinwork, 2025-06-05:
It happens with latest master repos, including codal-core commit 40198ceac815ce33c71888615f1270a252068d83
It happens with or without BLE pairing mode

```
145,145,145,145,145,145,145,146,146,146,145,145,145,145,145,145,145,145,145,145,
145,145,145,145,145,145,107,107,107,107,107,107,107,107,107,108,108,108,106,106,
108,108,108,108,108,108,35,35,35,32,32,32,32,32,32,32,32,32,32,32,
28,28,28,28,28,28,149,149,149,149,149,149,149,149,149,149,149,149,148,148,
```

It's OK with v0.2.63
```
175,175,175,175,175,175,175,174,174,174,175,175,175,175,175,174,174,174,175,175,
111,111,111,111,108,108,108,109,109,108,108,108,109,109,109,111,111,111,110,110,
71,71,71,66,66,66,66,66,66,66,66,66,66,66,66,68,68,68,66,66,
174,174,174,174,174,174,174,174,174,174,174,174,174,174,174,175,175,175,174,174,
```

<details>
<summary>C++ code</summary>

```
#include "MicroBit.h"

MicroBit uBit;

int array[20];

void batch()
{
    for ( int i = 0; i < 20; i++)
    {
        array[i] = uBit.audio.levelSPL->getValue();
        uBit.sleep(5);
    }

    for ( int i = 0; i < 20; i++)
    {
      uBit.serial.printf( "%d,", (int) array[i]);
    }
    uBit.serial.printf( "\r\n");
}

void forever()
{
    uBit.audio.levelSPL->setUnit( LEVEL_DETECTOR_SPL_8BIT);
    batch();
    uBit.sleep(1000);

    while (true)
    {
        if ( uBit.display.readLightLevel() > 200)
        {
            batch();
            uBit.sleep(1000);
        }
        uBit.sleep(20);
    }
}

int main()
{
    uBit.init();
    create_fiber( forever);
    release_fiber();
}

```

</details>

finneyj, 2025-06-05:
Thanks. I've been looking at this this morning. I have a fix, and am currently tuning it.

microbit-carlos, 2025-06-13:
Thanks Martin and Joe! I'll close this one as fixed then 👍🎉