lancaster-university/codal-microbit-v2 #489: audio refactor: Sound Level in dB value jumps from 35 dB to 53 dB

microbit-carlos, 2025-06-13:
With a programme like this and carefully playing a sound closer and closer to the micro:bit microphone we can see that the Sound Level in dB value returned jumps straight from 35 to 53 dB:

```cpp
#include "MicroBit.h"

MicroBit uBit;

int main()
{
    uBit.init();

    while (true) {
        uBit.serial.printf("8bit: %d,\tdB: %d\n",
                           (int)uBit.audio.levelSPL->getValue(LEVEL_DETECTOR_SPL_8BIT),
                           (int)uBit.audio.levelSPL->getValue(LEVEL_DETECTOR_SPL_DB));
        uBit.sleep(100);
    }
}
```

[MICROBIT.hex.zip](https://github.com/user-attachments/files/20728508/MICROBIT.hex.zip)

To replicate the test:
- Run the programme on a micro:bit and connect a serial terminal
- In a relatively quiet room it should output 35 dB fairly constantly
- I used this [online tone generator](https://onlinetonegenerator.com) in my phone, playing 440Hz at the lowest volume setting
- From around 30 cms away I start bringing the phone closer to the micro:bit
- At around 5cm the dB values jump from 35 directly to 53
- Finding the right spot where it constantly flips between these two values and we can see in the serial output that it won't return a value between 35 and 53.

```
8bit: 0,    dB: 35
8bit: 0,    dB: 35
8bit: 13,   dB: 54
8bit: 0,    dB: 35
8bit: 0,    dB: 35
8bit: 0,    dB: 35
8bit: 0,    dB: 35
8bit: 9,    dB: 53
8bit: 0,    dB: 35
8bit: 0,    dB: 35
8bit: 9,    dB: 53
8bit: 0,    dB: 35
8bit: 0,    dB: 35
8bit: 9,    dB: 53
8bit: 0,    dB: 35
8bit: 0,    dB: 35
```

kevinjwalters, 2025-06-18:
I doubt it'll have any effect in this case but single frequency sine waves leaves you vulnerable to standing waves creating variations in SPL with position. Might be worth switching to white noise with its broad spectrum

finneyj, 2025-07-15:
Thanks @microbit-carlos.

This is largely due to the noise gate I added to the sample data, which counteracts the 1-bit noise we see on the ADC. At low volumes, this does tend to mask out certain frequencies.

I've added a smoothing function in the 35..56db range to balance this, amended in https://github.com/lancaster-university/codal-core/commit/8c8366f9a1e92da69f90fe816456c4b9c42ffd13

microbit-carlos, 2025-08-05:
Thanks @kevinjwalters, you are right, we should use white noise for this test!
Tested, and we now have multiple steps between 35 and 56 dB, thanks Joe!

closed 2025-08-05 by microbit-carlos