lancaster-university/codal-microbit-v2 #490: microphone: Different microphone types might have different warm up times

microbit-carlos, 2025-06-17:
I'll add more info soon, just wanted to raise this first for tracking purposes.

microbit-carlos, 2025-06-18:
Okay, so I've created this simple C++ test programme, which does the following in an infinite loop:
- Start measuring time for this loop iteration
- Repeat 10 times in a row:
    - Take a sound level measurement (8-bits)
    - Calculate how long the measurement took and how long since the loop iteration
    - Print to serial the results
    - Wait until the next tick (each tick being 4ms)
- Wait 5 seconds until the next loop iteration

```cpp
```

With CODAL v0.2.71 ([soundleve-0271.hex.zip](https://github.com/user-attachments/files/20799135/soundleve-0271.hex.zip)) and a V2.00 board we can see that the first sound level measurement takes 15-17ms and it already provides a valid value (within expected range in the room I was in).
Any measurements after that don't really take time, as the data is processed in the background and `uBit.audio.levelSPL->getValue()` fetches the latest value: 

<details><summary>Click here to view the CODAL v0.2.71 with V2.00 (default microphone) results</summary>
<p>

```
-----
(Total: 17, last 17 ms) 28
(Total: 23, last 0 ms) 24
(Total: 27, last 0 ms) 24
(Total: 31, last 0 ms) 28
(Total: 35, last 0 ms) 15
(Total: 40, last 1 ms) 15
(Total: 43, last 0 ms) 17
(Total: 47, last 0 ms) 23
(Total: 51, last 0 ms) 17
(Total: 55, last 0 ms) 17
-----
(Total: 15, last 15 ms) 21
(Total: 19, last 0 ms) 21
(Total: 23, last 0 ms) 21
(Total: 27, last 0 ms) 31
(Total: 32, last 0 ms) 31
(Total: 35, last 0 ms) 15
(Total: 39, last 0 ms) 26
(Total: 43, last 0 ms) 28
(Total: 47, last 0 ms) 28
(Total: 51, last 0 ms) 33
-----
(Total: 16, last 16 ms) 26
(Total: 19, last 0 ms) 26
(Total: 23, last 0 ms) 39
(Total: 28, last 0 ms) 28
(Total: 31, last 0 ms) 28
(Total: 36, last 0 ms) 26
(Total: 39, last 0 ms) 15
(Total: 43, last 0 ms) 15
(Total: 47, last 0 ms) 17
(Total: 51, last 0 ms) 24
```

</p>
</details> 

However, using the same CODAL v0.2.71 hex with a V2.21 micro:bit with an alternative microphone (HongSound), we can see that, while the first measurement takes the same amount of time (15-18 ms), the value returned is invalid and it takes around 24ms from pipe startup before it returns a valid value.

<details><summary>Click here to view the CODAL v0.2.71 with V2.21 (alternative HongSound microphone) results</summary>
<p>

```
-----
(Total: 18, last 18 ms) 246
(Total: 24, last 0 ms) 41
(Total: 28, last 0 ms) 41
(Total: 32, last 0 ms) 28
(Total: 36, last 0 ms) 24
(Total: 40, last 0 ms) 24
(Total: 44, last 0 ms) 26
(Total: 48, last 0 ms) 13
(Total: 52, last 0 ms) 32
(Total: 56, last 0 ms) 32
-----
(Total: 15, last 15 ms) 248
(Total: 20, last 0 ms) 248
(Total: 24, last 0 ms) 46
(Total: 28, last 0 ms) 35
(Total: 32, last 0 ms) 35
(Total: 36, last 0 ms) 19
(Total: 40, last 0 ms) 24
(Total: 44, last 0 ms) 33
(Total: 48, last 0 ms) 33
(Total: 52, last 0 ms) 29
-----
(Total: 17, last 17 ms) 249
(Total: 20, last 0 ms) 249
(Total: 24, last 0 ms) 43
(Total: 28, last 0 ms) 28
(Total: 32, last 0 ms) 28
(Total: 36, last 0 ms) 23
(Total: 40, last 0 ms) 23
(Total: 44, last 0 ms) 23
(Total: 48, last 0 ms) 13
(Total: 52, last 0 ms) 9
```
</p>
</details> 

So it looks like the HongSound microphone might need up to 24ms to provide valid data.

I'll include results with CODAL v0.3.0 in the next post, as that includes other questions and considerations.

microbit-carlos, 2025-06-18:
## CODAL v0.3.0 (+ latest codal-core)

Same code as shown in https://github.com/lancaster-university/codal-microbit-v2/issues/490#issuecomment-2984567003.

codal.json:
```json
{
    "target": {
        "name": "codal-microbit-v2",
        "url": "https://github.com/lancaster-university/codal-microbit-v2",
        "branch": "master",
        "dev": true,
        "type": "git"
    },
    "config":{
        "MICROBIT_BLE_ENABLED" : 0,
        "MICROBIT_BLE_PAIRING_MODE": 0,
        "LEVEL_DETECTOR_SPL_8BIT_000_POINT": 52.0,
        "LEVEL_DETECTOR_SPL_8BIT_255_POINT": 100.0
    }
}
```

Tested with the following commits:
- codal-microbit-v2: 8256835171af53d745391c709ca67eeb9ea455a9
- codal-core: a2f5051dc60f89276546517f1077e75659ae6741
- codal-nrf52: 1fbb7240290fe36a55c61378f5cdeb7640f3ec4a
- codal-microbit-nrf5sdk: 4b8abc690f6c9fca6132e6db5ee13a795a263f88

Hex file:  [soundleve-030-plus.hex.zip](https://github.com/user-attachments/files/20800797/soundleve-030-plus.hex.zip)

## CODAL v0.3.0+ default microphone

In this case we can see that:
- The **first measurement** in the first loop iteration takes **19 ms**
    - However, following first measurement for all other loops iterations take around 47 ms
    - The very first measurement time might be somewhat caused by the initial `uBit.sleep(1000)` after `uBit.init()`, removing that line increases the first measurement to 78 ms, and later "first measurements" still take around 47 ms.
- During a single loop iteration, each measurement after the first does take between 17 and 20 ms.
    - This is probably a different discussion, so will open a separate issue 
- All the data is valid, as expected (we know this is working fine with the default microphone)

<details><summary><i>Click here to view the CODAL v0.3.0+ with V2.00 (default microphone) results</i></summary>
<p>

```
-----
(Total: 19, last 19 ms) 0
(Total: 42, last 18 ms) 0
(Total: 65, last 17 ms) 0
(Total: 88, last 20 ms) 0
(Total: 111, last 19 ms) 0
(Total: 134, last 18 ms) 0
(Total: 157, last 17 ms) 0
(Total: 180, last 20 ms) 0
(Total: 203, last 19 ms) 0
(Total: 226, last 18 ms) 0
-----
(Total: 47, last 47 ms) 0
(Total: 70, last 18 ms) 0
(Total: 93, last 17 ms) 0
(Total: 116, last 20 ms) 0
(Total: 139, last 19 ms) 0
(Total: 162, last 18 ms) 0
(Total: 185, last 17 ms) 0
(Total: 208, last 16 ms) 0
(Total: 231, last 19 ms) 0
(Total: 254, last 18 ms) 0
-----
(Total: 47, last 47 ms) 0
(Total: 70, last 18 ms) 0
(Total: 93, last 17 ms) 0
(Total: 116, last 20 ms) 0
(Total: 139, last 19 ms) 0
(Total: 162, last 18 ms) 0
(Total: 185, last 17 ms) 0
(Total: 208, last 16 ms) 0
(Total: 231, last 19 ms) 0
(Total: 254, last 18 ms) 0
```

</p>
</details> 

## CODAL v0.3.0+  alternative HongSound microphone

- The first loop iteration measurement takes 22 ms
    - This seems to be enough to resolve the original issue, as testing with CODAL v0.2.71 showed the HongSound microphone required between 21 and 24 ms
    - There is another alternative mic I haven't been able to test yet, which could have a different start up time
- Like the V2.00 test, each subsequent sound level "first measurement" takes 46 ms
- Like the V2.00 test, each measurement after the first does take between 17 and 20 ms.
- In this CODAL version all the data is valid, so the extra time taken on the first measurement has resolved this already

So, with the HongSound microphone, it looks like the additional time taken for the first measurement when the pipeline wakes up is long enough to resolve the original issue.
However, there is also an alternative ACC microphone we need to test to figure out its startup time.

<details><summary><i>Click here to view the CODAL v0.3.0+ with with V2.21 (alternative HongSound microphone) results</i></summary>
<p>

```
-----
(Total: 22, last 22 ms) 0
(Total: 45, last 18 ms) 0
(Total: 68, last 17 ms) 0
(Total: 91, last 20 ms) 0
(Total: 114, last 19 ms) 0
(Total: 137, last 18 ms) 0
(Total: 160, last 17 ms) 0
(Total: 183, last 20 ms) 0
(Total: 206, last 19 ms) 0
(Total: 229, last 18 ms) 0
-----
(Total: 46, last 46 ms) 0
(Total: 69, last 18 ms) 0
(Total: 92, last 17 ms) 0
(Total: 115, last 16 ms) 0
(Total: 138, last 19 ms) 0
(Total: 161, last 18 ms) 0
(Total: 185, last 18 ms) 0
(Total: 208, last 17 ms) 0
(Total: 231, last 20 ms) 0
(Total: 254, last 19 ms) 0
-----
(Total: 46, last 46 ms) 0
(Total: 69, last 18 ms) 0
(Total: 92, last 17 ms) 0
(Total: 115, last 16 ms) 0
(Total: 138, last 19 ms) 0
(Total: 161, last 18 ms) 0
(Total: 185, last 18 ms) 0
(Total: 208, last 17 ms) 0
(Total: 231, last 20 ms) 0
(Total: 254, last 19 ms) 0
```

</p>
</details>

martinwork, 2025-07-18:
@microbit-carlos See test results on 
https://support.microbit.org/helpdesk/tickets/88225
https://support.microbit.org/helpdesk/tickets/90023

microbit-carlos, 2025-07-18:
Also, trying this programme on MakeCode v7, and in a quiet room and my finger covering the microphone whole and the values shown are around 170ish, which is definitely too high, so there was probably a bug in CODAL v0.2.6.8 too

microbit-carlos, 2026-08-27:
This is working now with the work Joe released in v0.3.1.

closed 2026-08-27 by microbit-carlos